RoboStack/ros-noetic #419: How to update whole environment?

peci1, 2023-10-23:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

How do I update whole installed environment? This part is not covered in the docs. Trying to `mamba update --all` does nothing, although I know there are packages that could be installed (e.g. in one of my older envs, there are still packages requiring ros-distro-mutex 0.1 noetic, while there is already 0.4 noetic).

### Installed packages

```shell
# packages in environment at /home/deck/mambaforge/envs/ros_env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
_sysroot_linux-64_curr_repodata_hack 3                   h69a702a_13    conda-forge
alabaster                 0.7.13             pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.7.2              h166bdaf_0    conda-forge
aom                       3.5.0                h27087fc_0    conda-forge
apr                       1.7.0                h7f98852_5    conda-forge
assimp                    5.2.4                h4c92fa2_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
babel                     2.13.0             pyhd8ed1ab_0    conda-forge
bcrypt                    4.0.1            py39h9fdd4d6_1    conda-forge
beautifulsoup4            4.12.2             pyha770c72_0    conda-forge
binutils                  2.40                 hdd6e379_0    conda-forge
binutils_impl_linux-64    2.40                 hf600244_0    conda-forge
binutils_linux-64         2.40                 hbdbef99_2    conda-forge
boost                     1.74.0           py39h5472131_5    conda-forge
boost-cpp                 1.74.0               h6cacc03_7    conda-forge
brotli                    1.0.9                h166bdaf_9    conda-forge
brotli-bin                1.0.9                h166bdaf_9    conda-forge
brotli-python             1.0.9            py39h5a03fae_9    conda-forge
brotlipy                  0.7.0           py39hd1e30aa_1006    conda-forge
bullet                    3.24                 hcca971b_0    conda-forge
bullet-cpp                3.24                 h4661b88_0    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.20.1               hd590300_1    conda-forge
c-compiler                1.6.0                hd590300_0    conda-forge
ca-certificates           2023.7.22            hbcca054_0    conda-forge
cairo                     1.16.0            ha12eb4b_1010    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cffi                      1.16.0           py39h7a31438_0    conda-forge
chainer                   7.8.1              pyhd8ed1ab_0    conda-forge
charset-normalizer        3.3.1              pyhd8ed1ab_0    conda-forge
cmake                     3.25.2               h077f3f9_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
compilers                 1.6.0                ha770c72_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
contourpy                 1.1.1            py39h7633fee_1    conda-forge
coreutils                 9.4                  hd590300_0    conda-forge
cryptography              39.0.0           py39hd598818_0    conda-forge
curl                      7.87.0               h6312ad2_0    conda-forge
cxx-compiler              1.6.0                h00ab1b0_0    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
cython                    3.0.4            py39h3d6467e_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
deprecated                1.2.14             pyh1a96a4e_0    conda-forge
distro                    1.8.0              pyhd8ed1ab_0    conda-forge
dlib                      19.24.0          py39h9b0c1b2_3    conda-forge
docutils                  0.20.1           py39hf3d152e_2    conda-forge
double-conversion         3.2.0                h27087fc_1    conda-forge
doxygen                   1.9.8                h661eb56_0    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
elfutils                  0.186                he364ef2_0    conda-forge
[... 616 lines cut ...]
                          /home/deck/mambaforge/envs/ros_env/.condarc
          conda version : 22.9.0
    conda-build version : not installed
         python version : 3.10.6.final.0
       virtual packages : __linux=5.13.0=0
                          __glibc=2.36=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /home/deck/mambaforge  (writable)
      conda av data dir : /home/deck/mambaforge/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/robostack-staging/linux-64
                          https://conda.anaconda.org/robostack-staging/noarch
          package cache : /home/deck/mambaforge/pkgs
                          /home/deck/.conda/pkgs
       envs directories : /home/deck/mambaforge/envs
                          /home/deck/.conda/envs
               platform : linux-64
             user-agent : conda/22.9.0 requests/2.31.0 CPython/3.10.6 Linux/5.13.0-valve36-1-neptune steamos/3.4.8 glibc/2.36
                UID:GID : 1000:1000
             netrc file : /home/deck/.netrc
           offline mode : False
```

Tobias-Fischer, 2023-10-23:
Hi @peci1 - we have changed channels, and are now using `robostack-staging`. We never got around moving the packages into the main `robostack` channel, and `robostack-staging` is now the channel that you should be using (apologies for the confusing name). Upgrading is probably not super straightforward unfortunately, I think it would be easiest to re-create the environment with the packages that you need.

peci1, 2023-10-24:
Okay. I thought manually editing .condarc channels would make the switch to staging. Apparently not, so I'll have to reset the environment.

Once having a proper env set for robostack-staging, is `mamba update --all` the right way?

closed 2023-10-24 by Tobias-Fischer