RoboStack/ros-noetic #420: How is this feedstock maintained???

peci1, 2023-10-23:
No, this was deliberate, all good :)

Thanks again for the contribution!

_Originally posted by @Tobias-Fischer in https://github.com/RoboStack/ros-noetic/issues/345#issuecomment-1464762427_

Tobias-Fischer, 2023-10-23:
Hi @peci1 - thanks for getting in touch!

Regarding the changes in the yaml files: The commented-out packages are "artefacts" on how the yaml files are being processed with vinca. We have a flag `full_rebuild`, if this is set to `false`, then only a select subset of packages are rebuilt. So commenting packages out does not mean that they are not supported/maintained anymore, just that they didn't need to be built at this point in time as they have already been built. You can check out our https://robostack.github.io/Contributing.html for more details.

@traversaro is currently pushing toward rebuilding all currently supported packages in #418. You can see currently supported packages at https://robostack.github.io/noetic.html. Which packages exactly are "gone"?

Please note that RoboStack's core contributors are less than a handful of people who have full-time jobs outside of RoboStack. We are always looking for more hands to help out, you would be more than welcome to contribute. It would be great to see RoboStack being used more for teaching - some unis are already using it in their units.

Best, Tobi

peci1, 2023-10-24:
Ah, right. Thanks for reminding me about this procedure. It's been some time since I last contributed so I totally forgot it works like this.

Maybe to decrease the chance other contributors are surprised, there could be some comments added to the vinca files saying that the commented-out packages in some parts are still supported?

What it the current procedure if a community-added package starts having build problems? Do you git-blame to find the author and contact him/her? Or do you try to fix the build yourself first? Can it happen that a package is removed from the distribution?

And how do you figure out a package needs to be rebuilt? Is that on request, plus a few full rebuilds yearly?

> Which packages exactly are "gone"?

None from the anaconda hosting. I was just wondering whether their commenting out from the vinca files means they are no longer supported.

> Please note that RoboStack's core contributors are less than a handful of people who have full-time jobs outside of RoboStack. We are always looking for more hands to help out, you would be more than welcome to contribute.

Hmm, I hoped that after the announcement of https://discourse.ros.org/t/new-way-of-installing-ros/33929 , there was some task force helping robostack a bit more...

> It would be great to see RoboStack being used more for teaching - some unis are already using it in their units.

We'll definitely have a look at this possibility before the next semester as we're still struggling to find a simple yet working solution for students with all OSs.

Tobias-Fischer, 2023-10-24:
Hi @peci1 - we welcome contributions to our documentation, this is something we should definitely improve!

We typically try to fix build issues ourselves first, if it is too tricky we will be reaching out to people who have requested/added the package in question. We try not to remove packages anymore that are built in robostack-staging; that said at some point we tried to build as many packages as possible, and that became too much of a maintenance overload.

Yes, packages are rebuild on request + sporadic full rebuilds.

I wish there was a bigger task force :)

Tobias-Fischer, 2024-02-06:
#427 has been merged

closed 2024-02-06 by Tobias-Fischer