RoboStack/ros-noetic #434: Installing ros-noetic-catkin adds .catkin file to env directory, which breaks sourcing logic

fishbotics, 2024-01-10:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

When I install `ros-noetic-catkin` in a 3.9 conda environment, there is a file called `.catkin` that is added to the env directory. This file causes all sorts of havoc because the env is now considered a catkin directory and is then removed from my path when I source devel/setup.sh. Would it be possible to remove this?

### Installed packages

```shell
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
c-ares                    1.25.0               hd590300_0    conda-forge
ca-certificates           2023.12.12           h06a4308_0  
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
cmake                     3.21.3               h8897547_0    conda-forge
coreutils                 9.4                  hd590300_0    conda-forge
distro                    1.9.0              pyhd8ed1ab_0    conda-forge
docutils                  0.20.1           py39hf3d152e_3    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.5.0                hcb278e6_1    conda-forge
gmock                     1.11.0               h924138e_0    conda-forge
gtest                     1.11.0               h924138e_0    conda-forge
keyutils                  1.6.1                h166bdaf_0    conda-forge
krb5                      1.19.3               h08a2579_0    conda-forge
ld_impl_linux-64          2.38                 h1181459_1  
libcurl                   7.79.1               h494985f_1    conda-forge
libedit                   3.1.20191231         he28a2e2_2    conda-forge
libev                     4.33                 hd590300_2    conda-forge
libexpat                  2.5.0                hcb278e6_1    conda-forge
libffi                    3.4.4                h6a678d5_0  
libgcc-ng                 13.2.0               h807b86a_3    conda-forge
libgomp                   13.2.0               h807b86a_3    conda-forge
libnghttp2                1.43.0               ha19adfc_1    conda-forge
libssh2                   1.10.0               ha35d2d1_2    conda-forge
libstdcxx-ng              11.2.0               h1234567_1  
libuv                     1.46.0               hd590300_0    conda-forge
lz4-c                     1.9.3                h9c3ff4c_1    conda-forge
make                      4.3                  hd18ef5c_1    conda-forge
ncurses                   6.4                  h6a678d5_0  
nose                      1.3.7                   py_1006    conda-forge
openssl                   3.2.0                hd590300_1    conda-forge
osrf_pycommon             0.2.1              pyhd8ed1ab_0    conda-forge
pip                       23.3.1           py39h06a4308_0  
psutil                    5.9.7            py39hd1e30aa_0    conda-forge
pyparsing                 3.1.1              pyhd8ed1ab_0    conda-forge
python                    3.9.18               h955ad1f_0  
python-dateutil           2.8.2              pyhd8ed1ab_0    conda-forge
python_abi                3.9                      2_cp39    conda-forge
pyyaml                    6.0.1            py39hd1e30aa_1    conda-forge
readline                  8.2                  h5eee18b_0  
rhash                     1.4.3                hd590300_2    conda-forge
ros-distro-mutex          0.3.0                    noetic    robostack
ros-noetic-catkin         0.8.10          py39h6fdeb60_14    robostack
rosdep                    0.22.2             pyhd8ed1ab_2    conda-forge
rosdistro                 0.9.0            py39hf3d152e_1    conda-forge
rospkg                    1.5.0              pyhd8ed1ab_0    conda-forge
setuptools                68.2.2           py39h06a4308_0  
six                       1.16.0             pyh6c4a22f_0    conda-forge
sqlite                    3.41.2               h5eee18b_0  
tk                        8.6.12               h1ccaba5_0  
tzdata                    2023d                h04d1e81_0  
wheel                     0.41.2           py39h06a4308_0  
xz                        5.4.5                h5eee18b_0  
yaml                      0.2.5                h7f98852_2    conda-forge
zlib                      1.2.13               h5eee18b_0  
zstd                      1.5.0                ha95c52a_0    conda-forge
```

### Environment info

```shell
__    __    __    __
                 /  \  /  \  /  \  /  \
                /    \/    \/    \/    \
███████████████/  /██/  /██/  /██/  /████████████████████████
              /  / \   / \   / \   / \  \____
             /  /   \_/   \_/   \_/   \    o \__,
            / _/                       \_____/  `
            |/
        ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗
        ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗
        ██╔████╔██║███████║██╔████╔██║██████╔╝███████║
        ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║
        ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║
        ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝

        mamba (1.3.1) supported by @QuantStack

        GitHub:  https://github.com/mamba-org/mamba
        Twitter: https://twitter.com/QuantStack

█████████████████████████████████████████████████████████████

     active environment : adamtest1
    active env location : /home/rselab/miniconda3/envs/adamtest1
            shell level : 1
       user config file : /home/rselab/.condarc
 populated config files : /home/rselab/.condarc
                          /home/rselab/miniconda3/envs/adamtest1/.condarc
          conda version : 23.1.0
    conda-build version : not installed
         python version : 3.10.9.final.0
       virtual packages : __archspec=1=x86_64
                          __glibc=2.31=0
                          __linux=5.9.1=0
                          __unix=0=0
       base environment : /home/rselab/miniconda3  (writable)
      conda av data dir : /home/rselab/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack/linux-64
                          https://conda.anaconda.org/robostack/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /home/rselab/miniconda3/pkgs
                          /home/rselab/.conda/pkgs
       envs directories : /home/rselab/miniconda3/envs
                          /home/rselab/.conda/envs
               platform : linux-64
             user-agent : conda/23.1.0 requests/2.28.1 CPython/3.10.9 Linux/5.9.1-rt20 ubuntu/20.04.6 glibc/2.31
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2024-01-28:
Hi @fishbotics - where exactly is this `.catkin` directory located, and why/how does it cause issues?

Tobias-Fischer, 2024-02-06:
Closing for now - feel free to reopen with more information.