RoboStack/ros-noetic #440: Version of ROS package dependencies

alaurenzi, 2024-01-23:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Dear maintainers,
ROS packages typically link against their dependencies using a soname that includes the whole version number down to the patch version.

Example (truncated):
```
(conda_test_env) alaurenzi@alaurenzi-iit-desktop:~/mambaforge/envs/conda_test_env $ readelf -d lib/libmoveit_collision_detection.so

Dynamic section at offset 0x63208 contains 38 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libgeometric_shapes.so.0.7.3]
 0x0000000000000001 (NEEDED)             Shared library: [libpthread.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libclass_loader.so]
 0x0000000000000001 (NEEDED)             Shared library: [libconsole_bridge.so.1.0]
 0x0000000000000001 (NEEDED)             Shared library: [librosconsole.so]
 0x0000000000000001 (NEEDED)             Shared library: [librostime.so]
 0x0000000000000001 (NEEDED)             Shared library: [libroslib.so]
 0x0000000000000001 (NEEDED)             Shared library: [libboost_filesystem.so.1.78.0]
 0x0000000000000001 (NEEDED)             Shared library: [libtinyxml2.so.9]
 0x0000000000000001 (NEEDED)             Shared library: [libstdc++.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libgcc_s.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x000000000000000e (SONAME)             Library soname: [libmoveit_collision_detection.so.1.1.11]
 0x000000000000000f (RPATH)              Library rpath: [$ORIGIN/.]
```

Notice here, for instance, the requirement for `libgeometric_shapes.so.0.7.3`. 
However, when listing the dependencies of `ros-noetic-moveit-core` (which provides the above library), I see no version requirements at all in any of the ros package dependencies.
```
(conda_test_env) alaurenzi@alaurenzi-iit-desktop:~/mambaforge/envs/conda_test_env $ cat conda-meta/ros-noetic-moveit-core-1.1.11-py39h3fca0de_15.json 

{
  "build": "py39h3fca0de_15",
  "build_number": 15,
  "channel": "https://conda.anaconda.org/robostack-staging/linux-64",
  "constrains": [],
  "depends": [
    "python",
    "ros-noetic-std-msgs",
    "ros-noetic-geometry-msgs",
    "ros-noetic-sensor-msgs",
    "eigen",
    "ros-noetic-pluginlib",
    "ros-noetic-rosconsole",
    "ros-noetic-roslib",
    "ros-noetic-visualization-msgs",
    "ros-noetic-urdf",
    "ros-noetic-tf2-geometry-msgs",
    "ros-noetic-tf2-eigen",
    "ros-noetic-rostime",
    "ros-noetic-trajectory-msgs",
    "ros-noetic-moveit-msgs",
    "ros-noetic-xmlrpcpp",
    "ros-noetic-octomap",
    "ros-noetic-shape-msgs",
    "ros-noetic-geometric-shapes",
    "ros-noetic-kdl-parser",
    "urdfdom_headers",
    "bullet",
    "ros-noetic-eigen-stl-containers",
    "ros-noetic-octomap-msgs",
    "ros-noetic-fcl",
    "ros-noetic-random-numbers",
    "ros-noetic-srdfdom",
[... 646 lines cut ...]
          conda version : 22.11.1
    conda-build version : not installed
         python version : 3.10.8.final.0
       virtual packages : __archspec=1=x86_64
                          __cuda=12.2=0
                          __glibc=2.31=0
                          __linux=5.15.0=0
                          __unix=0=0
       base environment : /home/alaurenzi/mambaforge  (writable)
      conda av data dir : /home/alaurenzi/mambaforge/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/robostack-staging/linux-64
                          https://conda.anaconda.org/robostack-staging/noarch
          package cache : /home/alaurenzi/mambaforge/pkgs
                          /home/alaurenzi/.conda/pkgs
       envs directories : /home/alaurenzi/mambaforge/envs
                          /home/alaurenzi/.conda/envs
               platform : linux-64
             user-agent : conda/22.11.1 requests/2.31.0 CPython/3.10.8 Linux/5.15.0-91-generic ubuntu/20.04.6 glibc/2.31
                UID:GID : 1000:1000
             netrc file : /home/alaurenzi/.netrc
           offline mode : False
```

alaurenzi, 2024-01-24:
Update: I have uploaded my custom packages to a public channel. If interested, replicate the issue with this env file
```yaml
name: conda_test_env

channels:
- conda-forge
- robostack-staging
- iit_hhcm

dependencies:
- ros-noetic-desktop-full
- cartesian-interface
- xbot2
- modelinterfacerbdl
- xbot2-gui
- iit-centauro-ros-pkg
```

alaurenzi, 2024-01-24:
Update: this environment can be solved and works fine (so far)
Of course it is not ideal having to specify explicit versions
```yaml
name: conda_test_env

channels:
- conda-forge
- robostack-staging
- iit_hhcm

dependencies:
- ros-noetic-desktop-full
- ros-noetic-rosmon
- cartesian-interface
- xbot2
- modelinterfacerbdl
- xbot2-gui
- iit-centauro-ros-pkg
- ros-noetic-geometric-shapes==0.7.3  # because the only available moveit build links against 0.7.3
- ros-noetic-gazebo-ros==2.9.2=py39h67aadc1_15  #  workaround for [issue](https://github.com/RoboStack/ros-noetic/issues/415) 
```

v4hn, 2024-01-30:
> ROS packages typically link against their dependencies using a soname that includes the whole version number down to the patch version.

This is not true. Please note that we introduced SONAMES in Moveit and related packages (later including geometric_shapes) [in 2016](https://github.com/ros-planning/moveit/pull/273). The MoveIt stack is one of few in ROS that use SONAMES, we use it very drastically (by binding the SONAME to the release version) and I explained why in [my talk at ROSCon back then](https://roscon.ros.org/2017/).
Still, it does not change the fact that the relevant conda packages need to require the exact version of those upstream libs they were built with.
Notice that the same problem was never explicitly solved in OR deb releases either. They simply rebuilt all downstream packages with each release anyway.

Tobias-Fischer, 2024-02-06:
Hi @alaurenzi - many thanks for reporting this! This is indeed a problem that we should tackle. I think we should add `run_exports` to all packages and set the max_version to the patch version, so that the exact version is required to run. However, for this to work, we would need some way to capture all versions of all packages at a given point in time (https://github.com/RoboStack/robostack.github.io/issues/9). Otherwise, if there any package releases a new version, we might have to rebuild all packages which is not feasible currently.