RoboStack/ros-noetic #443: cmake fails with 20.04 project

pauldeee, 2024-02-05:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Hi there,

Apologies if this doesn't belong here...

On ubuntu 22.04 I am able to get ROS Noetic working fine. However some of my packages I would like to build require projects/SDKs that are meant to be built on 20.04. I have a fresh install of 22.04 and all I've followed the steps listed here: [https://robostack.github.io/GettingStarted.html](https://robostack.github.io/GettingStarted.html). Roscore, rviz and catkin_make appear to all work as they should.

However when I try to clone a project need for a package and do the usual:
```
git clone some_project
cd some_project && mkdir build && cd build
cmake ..
make
```

The project fails to build. When I say `which cmake` it lists the one in the ros_env... Am I missing something? Can I not use robostack in this way?

### Installed packages

```shell
# packages in environment at /home/paul/miniforge3/envs/ros_env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
adwaita-icon-theme        43                       unix_0    conda-forge
alsa-lib                  1.2.10               hd590300_0    conda-forge
aom                       3.8.1                h59595ed_0    conda-forge
apr                       1.7.0                h7f98852_5    conda-forge
argcomplete               3.2.2              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                hfb0e8fe_2    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
bcrypt                    4.1.2           py311h46250e7_0    conda-forge
binutils                  2.40                 hdd6e379_0    conda-forge
binutils_impl_linux-64    2.40                 hf600244_0    conda-forge
binutils_linux-64         2.40                 hbdbef99_2    conda-forge
boost                     1.82.0               h781c19f_6    conda-forge
boost-cpp                 1.82.0               h44aadfe_6    conda-forge
brotli                    1.1.0                hd590300_1    conda-forge
brotli-bin                1.1.0                hd590300_1    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
c-ares                    1.26.0               hd590300_0    conda-forge
c-compiler                1.7.0                hd590300_0    conda-forge
ca-certificates           2024.2.2             hbcca054_0    conda-forge
cairo                     1.18.0               h3faef2a_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py311hb3a22ac_0    conda-forge
cmake                     3.28.2               hcfe8598_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py311h38be061_1    conda-forge
colcon-core               0.15.0             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.2.5              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
[... 515 lines cut ...]
    conda-build version : not installed
         python version : 3.10.13.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=x86_64_v4
                          __conda=23.11.0=0
                          __glibc=2.35=0
                          __linux=6.5.0=0
                          __unix=0=0
       base environment : /home/paul/miniforge3  (writable)
      conda av data dir : /home/paul/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/linux-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/paul/miniforge3/pkgs
                          /home/paul/.conda/pkgs
       envs directories : /home/paul/miniforge3/envs
                          /home/paul/.conda/envs
               platform : linux-64
             user-agent : conda/23.11.0 requests/2.31.0 CPython/3.10.13 Linux/6.5.0-15-generic ubuntu/22.04.3 glibc/2.35 solver/libmamba conda-libmamba-solver/23.12.0 libmambapy/1.5.6
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2024-02-06:
Hi @pauldeee - thanks for reporting this! Could you please provide a bit more information, i.e. 1) which packages you are trying to build and 2) how the build fails?

pauldeee, 2024-02-06:
Hi, @Tobias-Fischer,

I ran into the following with [https://github.com/Livox-SDK/Livox-SDK2](https://github.com/Livox-SDK/Livox-SDK2):

```
[  1%] Building CXX object sdk_core/CMakeFiles/livox_lidar_sdk_static.dir/command_handler/command_impl.cpp.o
/home/paul/libs/Livox-SDK2/sdk_core/command_handler/command_impl.cpp: In static member function 'static livox_status livox::lidar::CommandImpl::SetLivoxLidarDebugPointCloud(uint32_t, bool, LivoxLidarLoggerCallback, void*)':
/home/paul/libs/Livox-SDK2/sdk_core/command_handler/command_impl.cpp:738:33: error: expected ')' before 'SCNu8'
  738 |                              "%" SCNu8 ".%" SCNu8 ".%" SCNu8 ".%" SCNu8, &req_buff.host_ip_addr[0]
      |                                 ^~~~~~
      |                                 )
/home/paul/libs/Livox-SDK2/sdk_core/command_handler/command_impl.cpp:737:9: note: to match this '('
  737 |   sscanf(GeneralCommandHandler::GetInstance().GetLidarCfg(handle).host_net_info.host_ip.c_str(),
      |         ^
/home/paul/libs/Livox-SDK2/sdk_core/command_handler/command_impl.cpp:42:1: note: 'SCNu8' is defined in header '<cinttypes>'; did you forget to '#include <cinttypes>'?
   41 | #include <iomanip>
  +++ |+#include <cinttypes>
   42 | #include <chrono>
/home/paul/libs/Livox-SDK2/sdk_core/command_handler/command_impl.cpp:738:31: warning: spurious trailing '%' in format [-Wformat=]
  738 |                              "%" SCNu8 ".%" SCNu8 ".%" SCNu8 ".%" SCNu8, &req_buff.host_ip_addr[0]
      |                               ^
/home/paul/libs/Livox-SDK2/sdk_core/command_handler/command_impl.cpp:738:30: warning: too many arguments for format [-Wformat-extra-args]
  738 |                              "%" SCNu8 ".%" SCNu8 ".%" SCNu8 ".%" SCNu8, &req_buff.host_ip_addr[0]
      |                              ^~~
make[2]: *** [sdk_core/CMakeFiles/livox_lidar_sdk_static.dir/build.make:440: sdk_core/CMakeFiles/livox_lidar_sdk_static.dir/command_handler/command_impl.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:223: sdk_core/CMakeFiles/livox_lidar_sdk_static.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

Tobias-Fischer, 2024-02-06:
Have you tried the suggestion provided by the compiler?
```
/home/paul/libs/Livox-SDK2/sdk_core/command_handler/command_impl.cpp:42:1: note: 'SCNu8' is defined in header '<cinttypes>'; did you forget to '#include <cinttypes>'?
   41 | #include <iomanip>
  +++ |+#include <cinttypes>
   42 | #include <chrono>
```

pauldeee, 2024-02-06:
Interestingly, when I place it where it suggests it should be placed it fails again, and just suggests to place it following where I have already placed it.

```
/home/paul/libs/Livox-SDK2/sdk_core/command_handler/command_impl.cpp:43:1: note: 'SCNu8' is defined in header '<cinttypes>'; did you forget to '#include <cinttypes>'?
   42 | #include <cinttypes>
  +++ |+#include <cinttypes>
   43 | #include <chrono>
```

However, if I move it to be the first include in the file it builds successfully.

Tobias-Fischer, 2024-02-06:
Great!

closed 2024-02-06 by Tobias-Fischer