RoboStack/ros-noetic #446: Package request: rosmon on Python 3.11

alaurenzi, 2024-02-09:
### Package name

ros-noetic-rosmon

### Operating System(s)

- [X] linux-64
- [ ] linux-aarch64
- [ ] osx-64
- [ ] osx-arm64
- [ ] win-64

### Additional context (optional)

Hi maintainers, 
I saw the new rebuild with Python 3.11 (which is super). However the rosmon package is missing (used to be there for Python 3.9). I was wondering if this is somehow a deliberate choice, or maybe there are some difficulties porting it to the new dependencies (in which case maybe I can help)

Thanks

Tobias-Fischer, 2024-02-09:
Hi @alaurenzi! With the recent builds, I've been running into https://github.com/xqms/rosfmt/issues/12. We should unvendor fmt in this package and use the version provided by conda-forge.

/cc @traversaro

Tobias-Fischer, 2024-02-18:
All done, unvendored + packages pushed.