RoboStack/ros-noetic #448: Roscore fails with "Procedure entry point could not be located"

dariosortino, 2024-02-23:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Ros commands fails with

"The procedure entry point could not be located in the dynamic link library C:\\ ...... \bin\roscpp.dll"

From the dependencies tool I see

[image]

### Installed packages

```shell
# Name                    Version                   Build  Channel
ace                       7.1.3                h63175ca_2    conda-forge
adwaita-icon-theme        43                        win_0    conda-forge
aiohttp                   3.9.3           py311ha68e1ae_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.8.1                h63175ca_0    conda-forge
apr                       1.7.0                heea9912_6    conda-forge
asio                      1.29.0               h63175ca_0    conda-forge
assimp                    5.3.1                h81f0834_2    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
azure-core-cpp            1.10.3               h249a519_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h91493d7_0    conda-forge
azure-storage-common-cpp  12.5.0               h91493d7_2    conda-forge
bcrypt                    4.1.2           py311hc37eb10_0    conda-forge
blosc                     1.21.5               hdccc3a2_0    conda-forge
boost                     1.82.0               h3810d55_6    conda-forge
boost-cpp                 1.82.0               h6f18f0d_6    conda-forge
brotli                    1.1.0                hcfcfb64_1    conda-forge
brotli-bin                1.1.0                hcfcfb64_1    conda-forge
bullet-cpp                3.25                 hf63dbb6_2    conda-forge
bzip2                     1.0.8                hcfcfb64_5    conda-forge
c-ares                    1.26.0               hcfcfb64_0    conda-forge
c-compiler                1.7.0                hcfcfb64_0    conda-forge
ca-certificates           2024.2.2             h56e8100_0    conda-forge
cairo                     1.18.0               h1fef639_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py311ha68e1ae_0    conda-forge
cfitsio                   4.3.1                h9b0cee5_0    conda-forge
clangdev                  5.0.0                   flang_3    conda-forge
cmake                     3.28.3               hf0feee3_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
compilers                 1.7.0                h57928b3_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
contourpy                 1.2.0           py311h005e61a_0    conda-forge
cppzmq                    4.10.0               h42135b4_1    conda-forge
cryptography              42.0.4          py311h28e9c30_0    conda-forge
cxx-compiler              1.7.0                h91493d7_0    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
dartsim                   6.13.1               hf6eeb0b_2    conda-forge
dav1d                     1.2.1                hcfcfb64_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
dirent                    1.21                          0    conda-forge
distro                    1.9.0              pyhd8ed1ab_0    conda-forge
dlfcn-win32               1.4.1                h63175ca_0    conda-forge
docutils                  0.20.1          py311h1ea47a8_3    conda-forge
double-conversion         3.3.0                h63175ca_0    conda-forge
eigen                     3.4.0                h91493d7_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.5.0                h63175ca_1    conda-forge
fcl                       0.7.0                he22821c_4    conda-forge
ffmpeg                    6.1.1           gpl_hb766fab_104    conda-forge
flang                     5.0.0           he025d50_20180525    conda-forge
[... 487 lines cut ...]
          conda version : 23.11.0
    conda-build version : not installed
         python version : 3.9.18.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=x86_64
                          __conda=23.11.0=0
                          __cuda=12.1=0
                          __win=0=0
       base environment : C:\Users\dsortino\AppData\Local\mambaforge  (writable)
      conda av data dir : C:\Users\dsortino\AppData\Local\mambaforge\etc\conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : C:\Users\dsortino\AppData\Local\mambaforge\pkgs
                          C:\Users\dsortino\.conda\pkgs
                          C:\Users\dsortino\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\dsortino\AppData\Local\mambaforge\envs
                          C:\Users\dsortino\.conda\envs
                          C:\Users\dsortino\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/23.11.0 requests/2.31.0 CPython/3.9.18 Windows/10 Windows/10.0.19045 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.6
          administrator : False
             netrc file : None
           offline mode : False
```

traversaro, 2024-02-23:
I replicated the issue on a fresh environment created only with `mamba create -n rosenv -c conda-forge -c robostack-staging ros-noetic-desktop` :

~~~
(rosenv) C:\Users\straversaro>mamba list
# packages in environment at C:\Users\straversaro\AppData\Local\miniforge3\envs\rosenv:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        43                        win_0    conda-forge
aom                       3.8.1                h63175ca_0    conda-forge
apr                       1.7.0                heea9912_6    conda-forge
assimp                    5.3.1                h81f0834_2    conda-forge
bcrypt                    4.1.2           py311hc37eb10_0    conda-forge
boost                     1.82.0               h3810d55_6    conda-forge
boost-cpp                 1.82.0               h6f18f0d_6    conda-forge
brotli                    1.1.0                hcfcfb64_1    conda-forge
brotli-bin                1.1.0                hcfcfb64_1    conda-forge
bzip2                     1.0.8                hcfcfb64_5    conda-forge
ca-certificates           2024.2.2             h56e8100_0    conda-forge
cairo                     1.18.0               h1fef639_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py311ha68e1ae_0    conda-forge
cmake                     3.28.3               hf0feee3_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
contourpy                 1.2.0           py311h005e61a_0    conda-forge
cryptography              42.0.4          py311h28e9c30_0    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
dav1d                     1.2.1                hcfcfb64_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.9.0              pyhd8ed1ab_0    conda-forge
docutils                  0.20.1          py311h1ea47a8_3    conda-forge
eigen                     3.4.0                h91493d7_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.5.0                h63175ca_1    conda-forge
ffmpeg                    6.1.1           gpl_hb766fab_104    conda-forge
fmt                       10.2.1               h181d51b_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_1    conda-forge
fontconfig                2.14.2               hbde0cde_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.49.0          py311ha68e1ae_0    conda-forge
freeglut                  3.2.2                h63175ca_2    conda-forge
freeimage                 3.18.0              h609497f_18    conda-forge
freetype                  2.12.1               hdaf720e_2    conda-forge
fribidi                   1.0.10               h8d14728_0    conda-forge
gdk-pixbuf                2.42.10              h90a7034_4    conda-forge
getopt-win32              0.1                  hcfcfb64_1    conda-forge
gettext                   0.21.1               h5728263_0    conda-forge
glib                      2.78.4               h12be248_0    conda-forge
glib-tools                2.78.4               h12be248_0    conda-forge
gmock                     1.14.0               h57928b3_1    conda-forge
graphite2                 1.3.13                     1000    conda-forge
graphviz                  9.0.0                h51cb2cd_1    conda-forge
gst-plugins-base          1.22.9               h001b923_0    conda-forge
gstreamer                 1.22.9               hb4038d2_0    conda-forge
gtest                     1.14.0               h91493d7_1    conda-forge
gts                       0.7.6                h6b5321d_4    conda-forge
harfbuzz                  8.3.0                h7ab893a_0    conda-forge
hdf5                      1.14.3          nompi_h73e8ff5_100    conda-forge
hicolor-icon-theme        0.17                 h57928b3_2    conda-forge
icu                       73.2                 h63175ca_0    conda-forge
imath                     3.1.9                h12be248_0    conda-forge
intel-openmp              2024.0.0         h57928b3_49841    conda-forge
jasper                    4.2.1                h28f2b1a_0    conda-forge
jxrlib                    1.1                  hcfcfb64_3    conda-forge
khronos-opencl-icd-loader 2023.04.17           h64bf75a_0    conda-forge
kiwisolver                1.4.5           py311h005e61a_1    conda-forge
[... 321 lines cut ...]
urdfdom_headers           1.1.1                h91493d7_0    conda-forge
vc                        14.3                hcf57466_18    conda-forge
vc14_runtime              14.38.33130         h82b7239_18    conda-forge
vs2015_runtime            14.38.33130         hcb4865c_18    conda-forge
wheel                     0.42.0             pyhd8ed1ab_0    conda-forge
x264                      1!164.3095           h8ffe710_2    conda-forge
x265                      3.5                  h2d74725_3    conda-forge
[... cut ...]
xorg-xproto               7.0.31            hcd874cb_1007    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
yaml-cpp                  0.8.0                h63175ca_0    conda-forge
zlib                      1.2.13               hcfcfb64_5    conda-forge
zstd                      1.5.5                h12be248_0    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
~~~

traversaro, 2024-02-23:
(I am looking into it).

traversaro, 2024-02-23:
I tried to install all libboost builds from `1.82.0=*_3` to `1.82.0=*_6`, and the problem is still there. So better to approach the problem systematically: if Dependency helper is complaining that some imports are missing, let's see what imports `roscpp.dll` uses that are missing `boost_chrono.dll`. 

The imports required by `roscpp.dll` are (top box in  the following screenshot) are:

Screenshot:

[image]

~~~
N/A, 31 (0x0000001f), public: static class boost::chrono::time_point<class boost::chrono::steady_clock,class boost::chrono::duration<__int64,struct std::ratio<1,1000000000> > > __cdecl boost::chrono::steady_clock::now(void), C:\Users\straversaro\AppData\Local\miniforge3\envs\rosenv\Library\bin\boost_chrono.dll, False, Microsoft
N/A, 33 (0x00000021), public: static class boost::chrono::time_point<class boost::chrono::system_clock,class boost::chrono::duration<__int64,struct std::ratio<1,10000000> > > __cdecl boost::chrono::system_clock::now(void), C:\Users\straversaro\AppData\Local\miniforge3\envs\rosenv\Library\bin\boost_chrono.dll, False, Microsoft
~~~

while the corresponding methods found in the `boost_chrono.dll` import (by searching `steady_clock::now(void)` and `system_clock::now(void)`) are:
~~~
32 (0x0020),  (0x), public: static class boost::chrono::time_point<class boost::chrono::steady_clock,class boost::chrono::duration<__int64,class boost::ratio<1,1000000000> > > __cdecl boost::chrono::steady_clock::now(void), 0x00002a80, Microsoft
34 (0x0022),  (0x), public: static class boost::chrono::time_point<class boost::chrono::system_clock,class boost::chrono::duration<__int64,class boost::ratio<1,10000000> > > __cdecl boost::chrono::system_clock::now(void), 0x00002ba0, Microsoft
~~~

Beside the hints that I get are not so important, the diff is:

~~~diff
--- <unnamed>
+++ <unnamed>
@@ -1,2 +1,2 @@
-N/A, 31 (0x0000001f), public: static class boost::chrono::time_point<class boost::chrono::steady_clock,class boost::chrono::duration<__int64,struct std::ratio<1,1000000000> > > __cdecl boost::chrono::steady_clock::now(void), C:\Users\straversaro\AppData\Local\miniforge3\envs\rosenv\Library\bin\boost_chrono.dll, False, Microsoft
-N/A, 33 (0x00000021), public: static class boost::chrono::time_point<class boost::chrono::system_clock,class boost::chrono::duration<__int64,struct std::ratio<1,10000000> > > __cdecl boost::chrono::system_clock::now(void), C:\Users\straversaro\AppData\Local\miniforge3\envs\rosenv\Library\bin\boost_chrono.dll, False, Microsoft
+32 (0x0020),  (0x), public: static class boost::chrono::time_point<class boost::chrono::steady_clock,class boost::chrono::duration<__int64,class boost::ratio<1,1000000000> > > __cdecl boost::chrono::steady_clock::now(void), 0x00002a80, Microsoft
+34 (0x0022),  (0x), public: static class boost::chrono::time_point<class boost::chrono::system_clock,class boost::chrono::duration<__int64,class boost::ratio<1,10000000> > > __cdecl boost::chrono::system_clock::now(void), 0x00002ba0, Microsoft
~~~

`roscpp.dll` wants the template arguments to be `std::ratio`, while the `boost_chrono.dll` instead is using `boost::ratio`. I guess there is some polyfill that is changing this (and hence the ABI) in the boost chrono headers, depending on something like the C++ standard version used while compilation. This is a typical problem that occurs in abseil, but I never this occured in boost.

traversaro, 2024-02-23:
I think the root of all problems boils down to this snippet of code (https://github.com/boostorg/chrono/blob/boost-1.82.0/include/boost/chrono/system_clocks.hpp#L75-L80):

~~~
#ifdef BOOST_CHRONO_WINDOWS_API
// The system_clock tick is 100 nanoseconds
# define BOOST_SYSTEM_CLOCK_DURATION boost::chrono::duration<boost::int_least64_t, ratio<BOOST_RATIO_INTMAX_C(1), BOOST_RATIO_INTMAX_C(10000000)> >
#else
# define BOOST_SYSTEM_CLOCK_DURATION boost::chrono::nanoseconds
#endif
~~~

On Windows, the `ratio` is a template argument, and probably for some strange reason of include order or `using namespace std` or who knows what, it is resolved to `std::ratio`, instead of `boost::ratio` . 

Interestingly, the problem does not appear if I build roscpp from source via normal cmake invocation, but it happens if I tried to build the package via the following boa recipe: [ros-noetic-roscpp.zip](https://github.com/RoboStack/ros-noetic/files/14385365/ros-noetic-roscpp.zip).

So a workaround is to build roscpp from source and install it in place of the roscpp provided by the package, this is the working `roscpp.dll` : [roscpp.zip](https://github.com/RoboStack/ros-noetic/files/14385386/roscpp.zip) . 

@dariosortino can you try as a workaround to do the following, in a Command Prompt with rosenv active (note to anyone else: this will corrupt your conda environment, do only if you know what you are doing):
~~~
curl -L https://github.com/RoboStack/ros-noetic/files/14385386/roscpp.zip -o roscpp.zip
tar -xf roscpp.zip
copy .\roscpp.dll %CONDA_PREFIX%\Library\bin
~~~

traversaro, 2024-02-23:
> I think the root of all problems boils down to this snippet of code (https://github.com/boostorg/chrono/blob/boost-1.82.0/include/boost/chrono/system_clocks.hpp#L75-L80):
>
> ```
> #ifdef [...]

Actually, I was barking at the wrong tree. The problem is that roscpp and almost all packages that depended on boost-cpp were built against 1.84.0, while at runtime boost 1.82.0 is installed, and so we are experiencing a regular ABI breakage due to different versions. The main problem is that in the rosdep mapping we are still mapping boost keys to `boost-cpp`, but then in the pinnings we are only pinning `libboost_devel` and `libboost_python_devel`. So, when the dependencies are installed, boost-cpp 1.84.0 is installed, and the pinning of `libboost_devel` and `libboost_python_devel` is ignored. The solution is to change all rosdep file to avoid using `boost-cpp` that is a deprecated package.

traversaro, 2024-02-24:
I just tested creating a new environment with:
~~~
mamba create -n newnoetic -c conda-forge -c robostack-staging ros-noetic-desktop
~~~

and both roscore and rviz work fine, mamba list for reference.

~~~
(newnoetic) C:\Users\straversaro>mamba list
# packages in environment at C:\Users\straversaro\AppData\Local\miniforge3\envs\newnoetic:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        43                        win_0    conda-forge
aom                       3.8.1                h63175ca_0    conda-forge
apr                       1.7.0                heea9912_6    conda-forge
assimp                    5.3.1                h81f0834_2    conda-forge
bcrypt                    4.1.2           py311hc37eb10_0    conda-forge
brotli                    1.1.0                hcfcfb64_1    conda-forge
brotli-bin                1.1.0                hcfcfb64_1    conda-forge
bzip2                     1.0.8                hcfcfb64_5    conda-forge
ca-certificates           2024.2.2             h56e8100_0    conda-forge
cairo                     1.18.0               h1fef639_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py311ha68e1ae_0    conda-forge
cmake                     3.28.3               hf0feee3_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
contourpy                 1.2.0           py311h005e61a_0    conda-forge
cryptography              42.0.5          py311h28e9c30_0    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
dav1d                     1.2.1                hcfcfb64_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.9.0              pyhd8ed1ab_0    conda-forge
docutils                  0.20.1          py311h1ea47a8_3    conda-forge
eigen                     3.4.0                h91493d7_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.5.0                h63175ca_1    conda-forge
ffmpeg                    6.1.1           gpl_hb766fab_104    conda-forge
fmt                       10.2.1               h181d51b_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_1    conda-forge
fontconfig                2.14.2               hbde0cde_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.49.0          py311ha68e1ae_0    conda-forge
freeglut                  3.2.2                h63175ca_2    conda-forge
freeimage                 3.18.0              h609497f_18    conda-forge
freetype                  2.12.1               hdaf720e_2    conda-forge
fribidi                   1.0.10               h8d14728_0    conda-forge
gdk-pixbuf                2.42.10              h90a7034_4    conda-forge
getopt-win32              0.1                  hcfcfb64_1    conda-forge
gettext                   0.21.1               h5728263_0    conda-forge
glib                      2.78.4               h12be248_0    conda-forge
glib-tools                2.78.4               h12be248_0    conda-forge
gmock                     1.14.0               h57928b3_1    conda-forge
graphite2                 1.3.13                     1000    conda-forge
graphviz                  9.0.0                h51cb2cd_1    conda-forge
gst-plugins-base          1.22.9               h001b923_0    conda-forge
gstreamer                 1.22.9               hb4038d2_0    conda-forge
gtest                     1.14.0               h91493d7_1    conda-forge
gts                       0.7.6                h6b5321d_4    conda-forge
harfbuzz                  8.3.0                h7ab893a_0    conda-forge
hdf5                      1.14.3          nompi_h73e8ff5_100    conda-forge
hicolor-icon-theme        0.17                 h57928b3_2    conda-forge
icu                       73.2                 h63175ca_0    conda-forge
imath                     3.1.9                h12be248_0    conda-forge
intel-openmp              2024.0.0         h57928b3_49841    conda-forge
jasper                    4.2.1                h28f2b1a_0    conda-forge
[... 324 lines cut ...]
urdfdom_headers           1.1.1                h91493d7_0    conda-forge
vc                        14.3                hcf57466_18    conda-forge
vc14_runtime              14.38.33130         h82b7239_18    conda-forge
vs2015_runtime            14.38.33130         hcb4865c_18    conda-forge
wheel                     0.42.0             pyhd8ed1ab_0    conda-forge
x264                      1!164.3095           h8ffe710_2    conda-forge
x265                      3.5                  h2d74725_3    conda-forge
xorg-kbproto              1.0.7             hcd874cb_1002    conda-forge
xorg-libice               1.1.1                hcd874cb_0    conda-forge
xorg-libsm                1.2.4                hcd874cb_0    conda-forge
xorg-libx11               1.8.7                hefa74cf_0    conda-forge
[... cut ...]
xorg-xproto               7.0.31            hcd874cb_1007    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
yaml-cpp                  0.8.0                h63175ca_0    conda-forge
zlib                      1.2.13               hcfcfb64_5    conda-forge
zstd                      1.5.5                h12be248_0    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
~~~

traversaro, 2024-02-24:
I think we can close the issue now @Tobias-Fischer !

closed 2024-02-25 by Tobias-Fischer