RoboStack/ros-noetic #451: Package request: ros-noetic-webots-ros

askman92, 2024-02-26:
### Package name

ros-noetic-webots-ros

### Operating System(s)

- [ ] linux-64
- [ ] linux-aarch64
- [ ] osx-64
- [X] osx-arm64
- [ ] win-64

### Additional context (optional)

If this is something that I can do myself and push a PR, please can you give me some pointers on doing this? Desperate for this package for my university dissertation, so anything I can do to help, I'm happy to

traversaro, 2024-02-26:
> If this is something that I can do myself and push a PR, please can you give me some pointers on doing this? Desperate for this package for my university dissertation, so anything I can do to help, I'm happy to

The general contributing instructions are available at https://robostack.github.io/Contributing.html . In a nutshell, you need to open a PR adding the package to the `vinca_***.yaml` file for the platform you are interested in.

askman92, 2024-02-26:
> The general contributing instructions are available at https://robostack.github.io/Contributing.html . In a nutshell, you need to open a PR adding the package to the `vinca_***.yaml` file for the platform you are interested in.

Not sure how I missed that. Thanks I'll take a look!

Tobias-Fischer, 2024-02-26:
Done.

closed 2024-02-26 by Tobias-Fischer