RoboStack/ros-noetic #453: webots_ros uses incorrect path for launching webots on Mac

askman92, 2024-02-27:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

In the launch script for Webots, I believe that the path used to launch the program points to the location of the executable on linux. This needs to be patched to point to `/Applications/Webots.app`

I tried to patch this locally however had some issues getting the patch to build correctly.  I now cannot seem to build the package locally as it is skipped in the vinca file due to existing on the robostack-staging channel.

If I can get some advice on how to do the build locally, I am pretty sure that I can fix this.

```
❯ roslaunch webots_ros e_puck_line.launch
... logging to /Users/jamesaskew/.ros/log/c88c356c-d5c0-11ee-bafd-bab821fef17f/roslaunch-Jamess-MacBook-Pro.local-20726.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt

started roslaunch server http://Jamess-MacBook-Pro.local:50769/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    epuck_line (webots_ros/e_puck_line)
    webots (webots_ros/webots_launcher.py)

auto-starting new master
process[master]: started with pid [20734]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to c88c356c-d5c0-11ee-bafd-bab821fef17f
process[rosout-1]: started with pid [20739]
started core service [/rosout]
process[webots-2]: started with pid [20740]
process[epuck_line-3]: started with pid [20741]
Traceback (most recent call last):
  File "/opt/homebrew/Caskroom/miniforge/base/envs/ros_env/lib/webots_ros/webots_launcher.py", line 81, in <module>
    subprocess.call(command)
  File "/opt/homebrew/Caskroom/miniforge/base/envs/ros_env/lib/python3.11/subprocess.py", line 389, in call
    with Popen(*popenargs, **kwargs) as p:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Caskroom/miniforge/base/envs/ros_env/lib/python3.11/subprocess.py", line 1026, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/opt/homebrew/Caskroom/miniforge/base/envs/ros_env/lib/python3.11/subprocess.py", line 1953, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/Applications/Webots.app/webots'
================================================================================REQUIRED process [webots-2] has died!
process has died [pid 20740, exit code 1, cmd /opt/homebrew/Caskroom/miniforge/base/envs/ros_env/lib/webots_ros/webots_launcher.py --world=/opt/homebrew/Caskroom/miniforge/base/envs/ros_env/share/webots_ros/worlds/e-puck_line.wbt --mode=realtime --no-gui=false, --stream=false __name:=webots __log:=/Users/jamesaskew/.ros/log/c88c356c-d5c0-11ee-bafd-bab821fef17f/webots-2.log].
log file: /Users/jamesaskew/.ros/log/c88c356c-d5c0-11ee-bafd-bab821fef17f/webots-2*.log
Initiating shutdown!
================================================================================
[epuck_line-3] killing on exit
[webots-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

### Installed packages

```shell
# packages in environment at /opt/homebrew/Caskroom/miniforge/base/envs/ros_env:
[... 500 lines cut ...]
          conda version : 23.11.0
    conda-build version : not installed
         python version : 3.10.13.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=m2
                          __conda=23.11.0=0
                          __osx=14.3.1=0
                          __unix=0=0
       base environment : /opt/homebrew/Caskroom/miniforge/base  (writable)
      conda av data dir : /opt/homebrew/Caskroom/miniforge/base/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /opt/homebrew/Caskroom/miniforge/base/pkgs
                          /Users/jamesaskew/.conda/pkgs
       envs directories : /opt/homebrew/Caskroom/miniforge/base/envs
                          /Users/jamesaskew/.conda/envs
               platform : osx-arm64
             user-agent : conda/23.11.0 requests/2.31.0 CPython/3.10.13 Darwin/23.3.0 OSX/14.3.1 solver/libmamba conda-libmamba-solver/23.12.0 libmambapy/1.5.6
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

traversaro, 2024-02-27:
The relevant code is https://github.com/cyberbotics/webots_ros/blob/a4c93300e1528ac4f8fbf5cb7c3ecee63ff69784/scripts/webots_launcher.py#L64-L67 . Did you already set `WEBOTS_HOME` to point to `/Applications/Webots.app` ? Otherwise your error is quite surprising.

Note that it is a bit of an hack, but to test you can also modify the `webots_launcher.py` installed to check how make it work.

traversaro, 2024-02-27:
> I tried to patch this locally however had some issues getting the patch to build correctly. I now cannot seem to build the package locally as it is skipped in the vinca file due to existing on the robostack-staging channel.

Not sure if it is the cleanest possible solution, a possible trick is to remove the `skip_existing` section from the vinca config file https://github.com/RoboStack/ros-noetic/blob/main/vinca_osx_arm64.yaml#L31-L34 . This will generate a lot of recipes, but then you can just build the `webots_ros` one to debug.

askman92, 2024-02-28:
> The relevant code is https://github.com/cyberbotics/webots_ros/blob/a4c93300e1528ac4f8fbf5cb7c3ecee63ff69784/scripts/webots_launcher.py#L64-L67 . Did you already set `WEBOTS_HOME` to point to [...]

I have `WEBOTS_HOME` set to `/Applications/Webots.app`, however the executable lives at `/Applications/Webots.app/Contents/MacOS/webots`, so when that command is concatenated, it points to a non-existent location.

> Not sure if it is the cleanest possible solution, a possible trick is to remove the `skip_existing` section from the vinca config file https://github.com/RoboStack/ros-noetic/blob/main/vinca_osx_arm64.yaml#L31-L34 . This will generate a lot of recipes (if you run `vinca --multiple`), but then you can just build the `webots_ros` one to debug.

I had tried just building the webots package, but was having all sorts of issues, but I think I just messed up my environment so I reset all my environments and started again, and now it seems to be building / running successfully with the patch. There's another similar bug related to paths, so I am going to try to patch that too and get them in a PR.

askman92, 2024-02-28:
I've raised the issue on the webots_ros repo, so hopefully they can provide some advice. If they don't come back in a day or two, I'll just raise the PR as in its current state as I have it at least working somewhat

askman92, 2024-03-03:
Nothing back from webots yet so have just created a PR for the original patch for now - #455

Not sure how to link it myself, I'm not sure I have sufficient privileges to do so

Tobias-Fischer, 2025-01-10:
Should be fixed with the rebuild - feel free to reopen if not.

closed 2025-01-10 by Tobias-Fischer