RoboStack/ros-noetic #454: Gazebo Spawn Fails for Turtlebot3 in ROS Noetic on M1 Mac

barrowben, 2024-03-01:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Hi there, I am trying to run Gazebo simulation with tb3_waffle in ros-noetic on m1 mac. Gazebo will run but the spawn service will fail, here's the output:

[ INFO] [1708571047.936553000]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1708571047.937299000]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[ INFO] [1708571048.448214000]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1708571048.449182000]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1708571049.298771000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1708571049.309822000]: Physics dynamic reconfigure ready.
[ERROR] [1708571059.682933, 10.004000]: Spawn service failed. Exiting.
[spawn_urdf-4] process has died [pid 63493, exit code 1, cmd {MY_HOME_DIR}/mambaforge/envs/tst4/lib/gazebo_ros/spawn_model -urdf -model turtlebot3_waffle -x 0.0 -y 0.0 -z 0.0 -param robot_description __name:=spawn_urdf __log:={MY_HOME_DIR}/.ros/log/fc9c7164-d12e-11ee-b132-ae64feed2c87/spawn_urdf-4.log].
log file: {MY_HOME_DIR}/.ros/log/fc9c7164-d12e-11ee-b132-ae64feed2c87/spawn_urdf-4*.log

Within the log file, this line seems relevant:

[rosout][INFO] 2024-02-22 03:04:19,681: Spawn status: SpawnModel: Entity pushed to spawn queue, but spawn service timed out waiting for entity to appear in simulation under the name turtlebot3_waffle

Any pointers what the issue could be here?

Thank you so much! Love this project and how active you are! :)

### Installed packages

```shell
# Name                    Version                   Build  Channel
adwaita-icon-theme        43                       unix_0    conda-forge
aom                       3.5.0                h7ea286d_0    conda-forge
apr                       1.7.0                h4614cfb_6    conda-forge
assimp                    5.2.5                h276577b_0    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
bcrypt                    4.1.2            py39h8fec3ad_0    conda-forge
blosc                     1.21.5               hc338f07_0    conda-forge
boost                     1.78.0           py39h819cc4c_5    conda-forge
boost-cpp                 1.78.0               hf1d6563_2    conda-forge
brotli                    1.1.0                hb547adb_1    conda-forge
brotli-bin                1.1.0                hb547adb_1    conda-forge
bullet                    3.21                 h85045c0_4    conda-forge
bullet-cpp                3.21             py39hde7b980_4    conda-forge
bzip2                     1.0.8                h93a5062_5    conda-forge
c-ares                    1.26.0               h93a5062_0    conda-forge
ca-certificates           2024.2.2             hf0a4a13_0    conda-forge
cairo                     1.16.0            h73a0509_1014    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0           py39he153c15_0    conda-forge
cfitsio                   4.2.0                h2f961c4_0    conda-forge
cmake                     3.26.3               hf234bd0_0    conda-forge
console_bridge            1.0.2                h3e96240_1    conda-forge
contourpy                 1.2.0            py39he9de807_0    conda-forge
cppzmq                    4.10.0               h42a7a8d_1    conda-forge
cryptography              42.0.2           py39hbc7c26c_0    conda-forge
curl                      7.88.1               h9049daf_1    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
dartsim                   6.12.2               h65d5661_3    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.9.0              pyhd8ed1ab_0    conda-forge
docutils                  0.20.1           py39h2804cbe_3    conda-forge
double-conversion         3.2.0                hb7217d7_1    conda-forge
eigen                     3.4.0                h1995070_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.5.0                hb7217d7_1    conda-forge
fcl                       0.7.0                h32c9b23_3    conda-forge
ffmpeg                    4.4.2           gpl_hf318d42_112    conda-forge
flann                     1.9.1             hd3e9afc_1011    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
[... 552 lines cut ...]
         python version : 3.10.8.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=m1
                          __conda=23.11.0=0
                          __osx=13.0=0
                          __unix=0=0
       base environment : /Users/benedictbarrow/mambaforge  (writable)
      conda av data dir : /Users/benedictbarrow/mambaforge/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/robostack/osx-arm64
                          https://conda.anaconda.org/robostack/noarch
          package cache : /Users/benedictbarrow/mambaforge/pkgs
                          /Users/benedictbarrow/.conda/pkgs
       envs directories : /Users/benedictbarrow/mambaforge/envs
                          /Users/benedictbarrow/.conda/envs
               platform : osx-arm64
             user-agent : conda/23.11.0 requests/2.31.0 CPython/3.10.8 Darwin/22.1.0 OSX/13.0 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.6
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2024-03-02:
Hi @barrowben it’s great to hear you like RoboStack :). Could you please update your environment to the new Python 3.11 builds? If it still doesn’t work, please let us know what exact commands you run and we can take a look.

Tobi

barrowben, 2024-03-05:
Thanks for the reply, Tobias. I made a fresh install following the installation guide using miniforge. I installed ros-noetic-desktop-full The commands I ran:

(base) $ conda install mamba -c conda-forge
(base) $ mamba create -n ros_env
(base) $ mamba activate ros_env
(ros_env) $ conda config --env --add channels conda-forge
(ros_env) $ conda config --env --add channels robostack-staging
(ros_env) $ conda config --env --remove channels defaults
(ros_env) $ mamba install ros-noetic-desktop-full

At this stage I am getting a bunch of these errors during installation:

===
Preparing transaction: done
Verifying transaction: done
Executing transaction: / g_module_open() failed for /Users/benedictbarrow/miniforge3/envs/ros_env/lib/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-ani.so: dlopen(/Users/benedictbarrow/miniforge3/envs/ros_env/lib/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-ani.so, 0x000A): Symbol not found: _libintl_bind_textdomain_codeset
  Referenced from: <CCFC52F6-2B87-3118-98A8-38DF74F15687> /Users/benedictbarrow/miniforge3/envs/ros_env/lib/libgdk_pixbuf-2.0.0.dylib
===

Charging ahead anyway with the installation:

(ros_env) $ mamba install compilers cmake pkg-config make ninja colcon-common-extensions catkin_tools
(ros_env) $ mamba deactivate
(base) $ mamba activate ros_env
(ros_env) roscore

ROS Master runs no problem! In a second terminal:

(ros_env) $ rviz

RVIZ runs. Now I install some turtlebot stuff:

(ros_env) $ mamba install -c robostack-staging \
ros-noetic-joy \
ros-noetic-teleop-twist-joy \
ros-noetic-rosserial-python \
ros-noetic-rosserial-msgs \
ros-noetic-amcl \
ros-noetic-map-server \
ros-noetic-move-base \
ros-noetic-urdf \
ros-noetic-xacro \
ros-noetic-compressed-image-transport \
ros-noetic-rviz \
ros-noetic-gmapping \
ros-noetic-interactive-markers \
ros-noetic-teleop-twist-keyboard \
ros-noetic-laser-proc \
ros-noetic-rgbd-launch \
ros-noetic-rosserial-arduino \
ros-noetic-rosserial-client \
ros-noetic-navigation \
ros-noetic-dynamixel-sdk \
ros-noetic-turtlebot3-msgs \
ros-noetic-turtlebot3 \
ros-noetic-turtlebot3-simulations

Build packages with catkin build, then try to run a package with Gazebo. I ran the Turtlebot3 empty world launch file. Here's the output:

started roslaunch server http://localhost:57884/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /use_sim_time: True
[... 700 lines cut ...]
xorg-libx11               1.8.7                hfd9643e_0    conda-forge
xorg-libxau               1.0.11               hb547adb_0    conda-forge
xorg-libxaw               1.0.14               h3422bc3_1    conda-forge
xorg-libxdmcp             1.1.3                h27ca646_0    conda-forge
xorg-libxext              1.3.4                h1a8c8d9_2    conda-forge
xorg-libxfixes            5.0.3             h3422bc3_1004    conda-forge
xorg-libxmu               1.1.3                h3422bc3_0    conda-forge
xorg-libxpm               3.5.17               hb547adb_0    conda-forge
xorg-libxrender           0.9.11               hb547adb_0    conda-forge
xorg-libxt                1.3.0                hb547adb_1    conda-forge
xorg-renderproto          0.11.1            h27ca646_1002    conda-forge
xorg-xextproto            7.3.0             h1a8c8d9_1003    conda-forge
xorg-xproto               7.0.31            h27ca646_1007    conda-forge
xz                        5.2.6                h57fd34a_0    conda-forge
yaml                      0.2.5                h3422bc3_2    conda-forge
yaml-cpp                  0.8.0                h13dd4ca_0    conda-forge
yarl                      1.9.4           py311h05b510d_0    conda-forge
zeromq                    4.3.5                hebf3989_1    conda-forge
zipp                      3.17.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h53f4e23_5    conda-forge
zstd                      1.5.5                h4f39d0f_0    conda-forge
zziplib                   0.13.69              he1e0b03_1    conda-forge
========================================

Would you have any idea what the issue is here? Please let me know if you need any more information. Thank you :)

Tobias-Fischer, 2024-04-01:
Could be the same issue as #459 

/cc @traversaro

traversaro, 2024-04-07:
> Could be the same issue as #459
> 
> /cc @traversaro

A workaround was found and reported in https://github.com/RoboStack/ros-noetic/issues/459#issuecomment-2027155423 .

bchax, 2024-11-11:
Hi there. Has this been solved? I am quiet new at github. I am looking to configure Robostack for a university project with ROS and Gazebo in mi M2 Mac. ROS runs perfectly but Gazebo is still not spawning the robot. I have read everything regarding libprotobuf==4.25.1 and so on, but the more technical it gets the more unable I find myself to follow the thread.

Is there any solution to this spawning problem? (for dummies)

traversaro, 2024-11-11:
> Is there any solution to this spawning problem? (for dummies)

Did you tried the workaround in https://github.com/RoboStack/ros-noetic/issues/459#issuecomment-2027155423 ?

bchax, 2024-11-11:
Ciao! I am very grateful for your fast reply.

I tried said workaraound after the complete installation of all the packages but I get:

 --o--
$ (ros_env) conda create -n envname ros-noetic-turtlebot3 ros-noetic-turtlebot3-simulations ros-noetic-turtlebot3-msgs libprotobuf==4.24.4

Channels:
 - conda-forge
Platform: osx-arm64
Collecting package metadata (repodata.json): done
Solving environment: failed

_PackagesNotFoundError_: The following packages are not available from current channels:

  - ros-noetic-turtlebot3-simulations
  - ros-noetic-turtlebot3-msgs
  - ros-noetic-turtlebot3

Current channels:

  - https://conda.anaconda.org/conda-forge

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
--o--

Of course then running:
--o--
(ros_env) export TURTLEBOT3_MODEL=waffle_pi                        
(ros_env) roslaunch turtlebot3_gazebo turtlebot3_empty_world.launch
--o--

Still does not spawn the robot.

bchax, 2024-11-11:
Ciao! I am very grateful for your fast response.

I did not see the edit. It indeed solved it.

Thank you very much.

For the inexperienced user as me: the solution is to create the ros_env different from the usual, i.e:

(base) conda create -n envname ros-noetic-turtlebot3 ros-noetic-turtlebot3-simulations ros-noetic-turtlebot3-msgs 

Insted of:

(base) mamba create -n ros_env python=3.11

So the steps become:

(base) conda install mamba -c conda-forge
_(base) conda create -n envname ros-noetic-turtlebot3 ros-noetic-turtlebot3-simulations ros-noetic-turtlebot3-msgs libprotobuf==4.24.4_

(base) mamba activate ros_env

# this adds the conda-forge channel to the new created environment configuration 
(ros_env) conda config --env --add channels conda-forge
# and the robostack channel
(ros_env) conda config --env --add channels robostack-staging
# remove the defaults channel just in case, this might return an error if it is not in the list which is ok
(ros_env) conda config --env --remove channels defaults

# Install ros-noetic into the environment (ROS1)
(ros_env) mamba install ros-noetic-desktop

# Install ros-humble into the environment (ROS2)
mamba install ros-humble-desktop

(ros_env) mamba deactivate
(base) mamba activate ros_env

(ros_env) mamba install compilers cmake pkg-config make ninja colcon-common-extensions catkin_tools rosdep

With this the robot will finally spawn:

(ros_env) export TURTLEBOT3_MODEL=waffle_pi
(ros_env) roslaunch turtlebot3_gazebo turtlebot3_empty_world.launch

bchax, 2024-11-11:
Ciao! Thank you for your fast response.

I did not see the edit up there, but it is not working or at least I do not yet understand the mechanics of conda create (I have been using mamba until now as instructed in https://robostack.github.io/GettingStarted.html#__tabbed_1_1).

Using:

`conda create -n envname ros-noetic-turtlebot3 ros-noetic-turtlebot3-simulations ros-noetic-turtlebot3-msgs libprotobuf==4.24.4`

I get the following response:

```
Channels:
 - conda-forge
Platform: osx-arm64
Collecting package metadata (repodata.json): done
_Solving environment: failed_

PackagesNotFoundError: The following packages are not available from current channels:

  - ros-noetic-turtlebot3-simulations
  - ros-noetic-turtlebot3-msgs
  - ros-noetic-turtlebot3

Current channels:

  - https://conda.anaconda.org/conda-forge

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```

If instead I type:

`conda create -n envname robostack::ros-noetic-turtlebot3 robostack::ros-noetic-turtlebot3-simulations robostack::ros-noetic-turtlebot3-msgs libprotobuf==4.24.4`

I get incompatibility of packages:

```
Channels:
 - conda-forge
 - robostack
Platform: osx-arm64
Collecting package metadata (repodata.json): done
Solving environment: failed

LibMambaUnsatisfiableError: Encountered problems while solving:
  - package ros-noetic-turtlebot3-1.2.5-py39hd384bd2_2 requires ros-noetic-turtlebot3-description, but none of the providers can be installed

Could not solve for environment specs
The following packages are incompatible
├─ libprotobuf 4.24.4  is requested and can be installed;
├─ ros-noetic-turtlebot3 is installable and it requires
│  └─ ros-noetic-turtlebot3-description, which requires
│     └─ ros-noetic-urdf with the potential options
│        ├─ ros-noetic-urdf 1.13.2 would require
│        │  └─ urdfdom >=2.3.5,<2.4.0a0 , which can be installed;
│        ├─ ros-noetic-urdf 1.13.2 would require
│        │  └─ ros-distro-mutex 0.1.* , which can be installed;
│        └─ ros-noetic-urdf 1.13.2 would require
│           └─ ros-distro-mutex 0.2.* , which can be installed;
└─ ros-noetic-turtlebot3-simulations is not installable because it requires
   ├─ ros-distro-mutex 0.3.* , which requires
   │  ├─ boost-cpp 1.74.* , which can be installed;
   │  └─ ogre 1.10.12* , which can be installed;
   └─ ros-noetic-turtlebot3-gazebo, which requires
      └─ ros-noetic-gazebo-ros, which requires
         └─ ros-noetic-gazebo-dev, which requires
            └─ gazebo >=11.10.1,<12.0a0  but there are no viable options
               ├─ gazebo [11.10.1|11.10.2] would require
               │  └─ libprotobuf >=3.19.4,<3.20.0a0 , which conflicts with any installable versions previously reported;
               ├─ gazebo [11.10.1|11.10.2|11.11.0|11.12.0] would require
               │  └─ ogre >=1.12.13,<1.13.0a0 , which conflicts with any installable versions previously reported;
               ├─ gazebo 11.10.2 would require
               │  └─ libprotobuf >=3.20.0,<3.21.0a0 , which conflicts with any installable versions previously reported;
               ├─ gazebo 11.11.0 would require
               │  └─ urdfdom >=3.1.0,<3.2.0a0 , which conflicts with any installable versions previously reported;
               ├─ gazebo 11.11.0 would require
               │  └─ libsdformat >=9.8.0,<10.0a0  but there are no viable options
               │     ├─ libsdformat 9.8.0 would require
               │     │  └─ urdfdom >=3.1.0,<3.2.0a0 , which conflicts with any installable versions previously reported;
               │     ├─ libsdformat 9.8.0 would require
               │     │  └─ urdfdom >=3.0.2,<3.1.0a0 , which conflicts with any installable versions previously reported;
               │     ├─ libsdformat 9.8.0 would require
               │     │  └─ urdfdom >=4.0.0,<4.1.0a0 , which conflicts with any installable versions previously reported;
               │     ├─ libsdformat 9.8.0 would require
               │     │  └─ urdfdom >=3.1.1,<3.2.0a0 , which conflicts with any installable versions previously reported;
               │     └─ libsdformat 9.8.0 would require
               │        └─ urdfdom >=4.0.1,<4.1.0a0 , which conflicts with any installable versions previously reported;
               ├─ gazebo [11.11.0|11.12.0|11.13.0] would require
               │  └─ boost-cpp >=1.78.0,<1.78.1.0a0 , which conflicts with any installable versions previously reported;
               ├─ gazebo [11.13.0|11.14.0] would require
               │  └─ urdfdom >=3.1.1,<3.2.0a0 , which conflicts with any installable versions previously reported;
               ├─ gazebo 11.14.0 would require
               │  └─ urdfdom >=4.0.0,<4.1.0a0 , which conflicts with any installable versions previously reported;
               └─ gazebo 11.14.0 would require
                  └─ urdfdom >=4.0.1,<4.1.0a0 , which conflicts with any installable versions previously reported.
```

Tobias-Fischer, 2025-01-10:
Should be fixed with the rebuild - feel free to reopen if not.

closed 2025-01-10 by Tobias-Fischer