RoboStack/ros-noetic #457: Ensure .dll files in Windows gets installed in %CONDA_PREFIX%/Library/bin and not %CONDA_PREFIX%/Library/lib

traversaro, 2024-03-10:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

See https://github.com/RoboStack/ros-noetic/issues/374#issuecomment-1987329600 .

### Installed packages

```shell
.
```

### Environment info

```shell
.
```

traversaro, 2024-03-10:
A specific case of this was reported in https://github.com/RoboStack/ros-noetic/issues/381 .

traversaro, 2024-03-11:
> I wonder whether we should include a test section in vinca that looks for any dlls installed in the wrong location, to avoid issues in the future?

Unfortunately it seems that some packages install their `.dll` in the form of plugin that are expected to be found in `%CONDA_PREFIX%\Library\lib`, as such location is declared in a `.xml` file to locate the plugin, so those are ok, so we can't just check that there are no `.dll` in `%CONDA_PREFIX%\Library\lib`, unless we have a test that we skip for some packages, but that is a bit more complex.

traversaro, 2024-03-11:
> Also, do you think we have similar problems in ros-humble? It might be worth checking.

Good point, installing `ros-humble-desktop ros-humble-moveit` there are two problematic files:
~~~
 Directory of C:\Users\straversaro\AppData\Local\miniforge3\envs\ros-humble\Library\lib

09/17/2023  06:18 AM            84,992 backward.dll
02/07/2023  08:17 PM            13,312 tf2_eigen_kdl.dll
               2 File(s)         98,304 bytes
               0 Dir(s)  443,462,447,104 bytes free
~~~

I will open a ros-humble issue to tackle those.

Tobias-Fischer, 2024-03-13:
Reopening till packages are rebuilt

Tobias-Fischer, 2025-01-10:
Should be fixed with the rebuild - feel free to reopen if not.

closed 2025-01-10 by Tobias-Fischer