RoboStack/ros-noetic #459: Gazebo doesn't populate with URDF model on Mac (Apple M2)

GokulEdakkePuram, 2024-03-29:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

When launching turtlebot3 simulation on gazebo, the URDF model doesn't get spawned. When running the simulation with an empty world there is just an empty gazebo window, whereas when running with a house world, it gets populated with the house model but again there is no robot.

### Installed packages

```shell
# Name                    Version                   Build  Channel
adwaita-icon-theme        43                       unix_0    conda-forge
aom                       3.8.2                h078ce10_0    conda-forge
apr                       1.7.0                h4614cfb_6    conda-forge
argcomplete               3.2.3              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                he63ff86_2    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
aws-c-auth                0.7.16               h0d2f7a6_8    conda-forge
aws-c-cal                 0.6.10               h677d54c_2    conda-forge
aws-c-common              0.9.14               h93a5062_0    conda-forge
aws-c-compression         0.2.18               h677d54c_2    conda-forge
aws-c-event-stream        0.4.2                h59ac3ca_6    conda-forge
aws-c-http                0.8.1                hfe5d766_7    conda-forge
aws-c-io                  0.14.6               h9ac2cdb_2    conda-forge
aws-c-mqtt                0.10.3               hb8a1441_2    conda-forge
aws-c-s3                  0.5.4                h4398043_0    conda-forge
aws-c-sdkutils            0.1.15               h677d54c_2    conda-forge
aws-checksums             0.1.18               h677d54c_2    conda-forge
aws-crt-cpp               0.26.4               he5ad744_2    conda-forge
aws-sdk-cpp               1.11.267             ha761c4c_4    conda-forge
azure-core-cpp            1.11.1               he231e37_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h2ffa867_1    conda-forge
azure-storage-common-cpp  12.5.0               h09a5875_4    conda-forge
bcrypt                    4.1.2           py311h94f323b_0    conda-forge
blosc                     1.21.5               hc338f07_0    conda-forge
boost                     1.82.0               hf3b2ce4_6    conda-forge
boost-cpp                 1.82.0               hca5e981_6    conda-forge
bullet                    3.25                 hf3b2ce4_2    conda-forge
bullet-cpp                3.25            py311h6e08293_2    conda-forge
bzip2                     1.0.8                h93a5062_5    conda-forge
c-ares                    1.27.0               h93a5062_0    conda-forge
c-compiler                1.7.0                h6aa9301_0    conda-forge
ca-certificates           2024.2.2             hf0a4a13_0    conda-forge
cairo                     1.18.0               hd1e100b_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
cctools                   986                  h4faf515_0    conda-forge
cctools_osx-arm64         986                  h62378fb_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py311h4a08483_0    conda-forge
cfitsio                   4.3.1                h808cd33_0    conda-forge
clang                     16.0.6               h30cc82d_6    conda-forge
clang-16                  16.0.6          default_he012953_6    conda-forge
clang_impl_osx-arm64      16.0.6              hc421ffc_11    conda-forge
clang_osx-arm64           16.0.6              h54d7cd3_11    conda-forge
clangxx                   16.0.6          default_h4cf2255_6    conda-forge
clangxx_impl_osx-arm64    16.0.6              hcd7bac0_11    conda-forge
clangxx_osx-arm64         16.0.6              h54d7cd3_11    conda-forge
cmake                     3.28.3               h50fd54c_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py311h267d04e_1    conda-forge
colcon-core               0.16.0             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.3.0              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
[... 467 lines cut ...]
          conda version : 24.3.0
    conda-build version : not installed
         python version : 3.11.8.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=m2
                          __conda=24.3.0=0
                          __osx=13.5.1=0
                          __unix=0=0
       base environment : /Users/gokulmnambiar/miniconda3  (writable)
      conda av data dir : /Users/gokulmnambiar/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/gokulmnambiar/miniconda3/pkgs
                          /Users/gokulmnambiar/.conda/pkgs
       envs directories : /Users/gokulmnambiar/miniconda3/envs
                          /Users/gokulmnambiar/.conda/envs
               platform : osx-arm64
             user-agent : conda/24.3.0 requests/2.31.0 CPython/3.11.8 Darwin/22.6.0 OSX/13.5.1 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.8
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

traversaro, 2024-03-29:
> When launching turtlebot3 simulation on gazebo, the URDF model doesn't get spawned. When running the simulation with an empty world there is just an empty gazebo window, whereas when running with a house world, it gets populated with the house model but again there is no robot.

Can you report the commands that you launch for these three experiments?

GokulEdakkePuram, 2024-03-29:
> Can you report the commands that you launch for these three experiments?

After installing ROS Noetic in the Conda environment by `conda install ros-noetic-full`, I installed the turtlebot3 packages by `conda install ros-noetic-turtlebot3`, `conda install ros-noetic-turtlebot3-simulations` and `conda install ros-noetic-turtlebot3-msgs`.

Then I launched the simulation with an empty world by running:
`roslaunch turtlebot3_gazebo turtlebot3_empty_world.launch`
but it was just an empty gazebo window.

I even tried running:
` roslaunch turtlebot3_gazebo turtlebot3_house.launch`
it spawns the house world but there is no turtlebot3 robot.

traversaro, 2024-03-29:
I am trying to reproduce your problem (even if just on linux), but running your commands results in an error related to `TURTLEBOT3_MODEL` env variable: 

~~~
(turtlebot) traversaro@IITBMP014LW012:~$  roslaunch turtlebot3_gazebo turtlebot3_empty_world.launch
... logging to /home/traversaro/.ros/log/7d9d0696-ede7-11ee-af6b-00155d46d1b4/roslaunch-IITBMP014LW012-962289.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

RLException: Invalid <arg> tag: environment variable 'TURTLEBOT3_MODEL' is not set.

Arg xml is <arg name="model" default="$(env TURTLEBOT3_MODEL)" doc="model type [burger, waffle, waffle_pi]"/>
The traceback for the exception was written to the log file
~~~

Is that expected? Should be set to some value?

GokulEdakkePuram, 2024-03-29:
> I am trying to reproduce your problem (even if just on linux), but running your commands results in an error related to `TURTLEBOT3_MODEL` env variable:
>
> ```
> (turtlebot) [...]

That can be fixed with 
`echo 'export TURTLEBOT3_MODEL=burger' >> ~/.bashrc`
`source ~/.bashrc`

Or you can add `export TURTLEBOT3_MODEL=burger` in the `.bashrc` file in the root folder.

GokulEdakkePuram, 2024-03-29:
I should also add that I am running these commands from bash. If you are using the terminal in Mac that is if you are running it from .zsh then you will have to add them in `.zshrc` file. 

I tried running them in zsh too but I get the same problem of the URDF model not spawning in gazebo.

I even tried creating a new Conda environment and cloning the turtlebot3 packages from the GitHub repository into my catkin workspace (I had tested this out on a linux machine and it worked) and built them from that but again its the same issue.

traversaro, 2024-03-30:
I have the same problem in Linux. Do you have any idea on how to launch gazebo with the `--verbose` option? This should provide some more info on what is going wrong.

GokulEdakkePuram, 2024-03-30:
> I have the same problem in Linux. Do you have any idea on how to launch gazebo with the `--verbose` option? This should provide some more info on what is going wrong.

I did try running the commands with:
`GAZEBO_VERBOSE=1 roslaunch turtlebot3_gazebo turtlebot3_house.launch` but i got the same output. I guess it runs automatically on verbose, I am not sure.

I tried these too:
 `export GAZEBO_LOG=debug`
 `roslaunch turtlebot3_gazebo turtlebot3_house.launch`

but again the same result.

GokulEdakkePuram, 2024-03-30:
I just found out this same problem mentioned in #343. @mmattamala had a work around and I tried to replicate it but I couldn't. @traversaro would be cool if you could look into that observation.

sylarevan, 2024-04-05:
Here is what I get when setting verbose flag:

```
>> roslaunch turtlebot3_gazebo turtlebot3_autorace_2020.launch
... logging to /Users/sylar/.ros/log/f96f2460-f351-11ee-9fc6-9aa119b2a19d/roslaunch-Air-de-Sylvain-44299.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://127.0.0.1:58738/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    spawn_urdf (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [44303]
ROS_MASTER_URI=http://127.0.0.1:11311

setting /run_id to f96f2460-f351-11ee-9fc6-9aa119b2a19d
process[rosout-1]: started with pid [44304]
started core service [/rosout]
process[gazebo-2]: started with pid [44305]
process[gazebo_gui-3]: started with pid [44306]
process[spawn_urdf-4]: started with pid [44309]
Gazebo multi-robot simulator, version 11.14.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Wrn] [GuiIface.cc:194] g/gui-plugin is really loading a SystemPlugin. To load a GUI plugin please use --gui-client-plugin
Gazebo multi-robot simulator, version 11.14.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Err] [Plugin.hh:212] Failed to load plugin /opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/libgazebo_ros_api_plugin.dylib: dlopen(/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/libgazebo_ros_api_plugin.dylib, 0x0009): Library not loaded: @rpath/libprotobuf.24.4.0.dylib
  Referenced from: <B6359CCF-A6FA-39B4-B2CF-93385FE1144E> /opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/libgazebo_ros_api_plugin.dylib
  Reason: tried: '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/gazebo-11/plugins/libprotobuf.24.4.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/gazebo-11/plugins/libprotobuf.24.4.0.dylib' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/libprotobuf.24.4.0.dylib' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/gazebo-11/plugins/libprotobuf.24.4.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/gazebo-11/plugins/libprotobuf.24.4.0.dylib' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/libprotobuf.24.4.0.dylib' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/libprotobuf.24.4.0.dylib' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/bin/../lib/libprotobuf.24.4.0.dylib' (no such file)
Failed to load plugin /opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/libgazebo_ros_api_plugin.dylib: dlopen(/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/libgazebo_ros_api_plugin.dylib, 0x0009): Library not loaded: @rpath/libprotobuf.24.4.0.dylib
  Referenced from: <B6359CCF-A6FA-39B4-B2CF-93385FE1144E> /opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/libgazebo_ros_api_plugin.dylib
  Reason: tried: '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/gazebo-11/plugins/libprotobuf.24.4.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/gazebo-11/plugins/libprotobuf.24.4.0.dylib' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/libprotobuf.24.4.0.dylib' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/gazebo-11/plugins/libprotobuf.24.4.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/gazebo-11/plugins/libprotobuf.24.4.0.dylib' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/libprotobuf.24.4.0.dylib' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/libprotobuf.24.4.0.dylib' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/bin/../lib/libprotobuf.24.4.0.dylib' (no such file)
[Msg] Waiting for master.
[Err] [Plugin.hh:212] Failed to load plugin /opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/libgazebo_ros_api_plugin.dylib: dlopen(/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/libgazebo_ros_api_plugin.dylib, 0x0009): Library not loaded: @rpath/libprotobuf.24.4.0.dylib
  Referenced from: <B6359CCF-A6FA-39B4-B2CF-93385FE1144E> /opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/libgazebo_ros_api_plugin.dylib
  Reason: tried: '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/gazebo-11/plugins/libprotobuf.24.4.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/gazebo-11/plugins/libprotobuf.24.4.0.dylib' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/libprotobuf.24.4.0.dylib' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/gazebo-11/plugins/libprotobuf.24.4.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/gazebo-11/plugins/libprotobuf.24.4.0.dylib' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/libproto [... cut ...]
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/share/turtlebot3_gazebo/models/turtlebot3_autorace"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/share/turtlebot3_gazebo/models/turtlebot3_autorace_2020"
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/share/turtlebot3_gazebo/models/turtlebot3_common_meshes
```

traversaro, 2024-04-05:
Thanks, this is really helpful. Can you share who did you set the `--verbose` flag, and your `mamba list`? I guess a possible workaround is just to manually pin libprotobuf to 24.4.0, i.e. `mamba install libprotobuf==24.4.0`. 

I think that the problem is that plugins such as libgazebo_ros_api_plugin.dylib are transitively depending on libprotobuf via gazebo public headers and CMake config, but as they do not declare an explicit dependency on libprotobuf, they do not get libprotobuf's run_export correctly set, and then they break as soon as a new libprotobuf is released. 

We do not get this on conda-forge packages as the CI fails if DSO library dependencies are not correctly declared (and you get errors such as `ERROR (libgz-sim-yarp-plugins,lib/libgz-sim-yarp-imu-system.so): Needed DSO lib/libprotobuf.so.25.3.0 found in ['conda-forge/linux-64::libprotobuf==4.25.3=h08a7969_0']`, see https://github.com/conda-forge/staged-recipes/pull/25910#issuecomment-2034209218). However, somehow this is not happening in our package generation, if possible we may consider enabling that in rattler-build migration fyi @wolfv @Tobias-Fischer . 

However, having these checks would only make the CI fail, it would not fix the issue. Possible solutions for the issue (that has been tracked for a long time at https://github.com/conda-forge/gazebo-feedstock/issues/107) are:
* Manually add run_exports for libprotobuf (and I guess also libabseil, as it is used in public headers of libprotobuf) in gazebo package
* Inject the libprotobuf and libabseil dependency at the rosdep level, i.e. replacing in https://github.com/conda-forge/gazebo-feedstock/issues/107 `[gazebo]` with all its public dependencies i.e. `[gazebo, libprotobuf]`  and if necessary more

sylarevan, 2024-04-05:
> Thanks, this is really helpful. Can you share who did you set the `--verbose` flag, and your `mamba list`? I guess a possible workaround is just to manually pin libprotobuf to 24.4.0, i.e. `mamba install libprotobuf==24.4.0`.

The verbose flag has been added to both `gzclient` and `gzserver` in the `empty_world.launch` from the gazebo_ros package. FYI there is no 24.4.0 version of libprotobuf, the only available versions extends from 3.13.0 to 5.26.1 on conda-forge.

Here is my `mamba list`:

```
> mamba list
# packages in environment at /opt/homebrew/Caskroom/mambaforge/base/envs/ros_env:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        43                       unix_0    conda-forge
aiohttp                   3.9.3           py311h05b510d_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.8.1                h078ce10_0    conda-forge
apr                       1.7.0                h9b22ae9_5    conda-forge
argcomplete               3.2.2              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                he63ff86_2    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
azure-core-cpp            1.10.3               he231e37_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h6aa02a4_0    conda-forge
azure-storage-common-cpp  12.5.0               h607ffeb_2    conda-forge
bcrypt                    4.1.2           py311h94f323b_0    conda-forge
blosc                     1.21.5               hc338f07_0    conda-forge
boost                     1.82.0               hf3b2ce4_6    conda-forge
boost-cpp                 1.82.0               hca5e981_6    conda-forge
brotli                    1.1.0                hb547adb_1    conda-forge
brotli-bin                1.1.0                hb547adb_1    conda-forge
bullet                    3.25                 hf3b2ce4_2    conda-forge
bullet-cpp                3.25            py311h6e08293_2    conda-forge
bzip2                     1.0.8                h93a5062_5    conda-forge
c-ares                    1.26.0               h93a5062_0    conda-forge
c-compiler                1.7.0                h6aa9301_0    conda-forge
ca-certificates           2024.2.2             hf0a4a13_0    conda-forge
cairo                     1.18.0               hd1e100b_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             h4faf515_16    conda-forge
cctools_osx-arm64         973.0.1             h62378fb_16    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py311h4a08483_0    conda-forge
cfitsio                   4.3.1                h808cd33_0    conda-forge
clang                     16.0.6               h30cc82d_5    conda-forge
clang-16                  16.0.6          default_he012953_5    conda-forge
clang_impl_osx-arm64      16.0.6               hc421ffc_9    conda-forge
clang_osx-arm64           16.0.6               h54d7cd3_9    conda-forge
clangxx                   16.0.6          default_h4cf2255_5    conda-forge
clangxx_impl_osx-arm64    16.0.6               hcd7bac0_9    conda-forge
clangxx_osx-arm64         16.0.6               h54d7cd3_9    conda-forge
click                     8.1.7           unix_pyh707e725_0    conda-forge
cmake                     3.28.3               h50fd54c_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py311h267d04e_1    conda-forge
colcon-core               0.15.0             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.2.5              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.4.0              pyhd8ed1ab_0    conda-forge
colcon-python-setup-py    0.2.7                      py_0    conda-forge
colcon-recursive-crawl    0.2.3              pyhd8ed1ab_0    conda-forge
[... 581 lines cut ...]
xorg-libice               1.1.1                hb547adb_0    conda-forge
xorg-libsm                1.2.4                hb547adb_0    conda-forge
xorg-libx11               1.8.7                hfd9643e_0    conda-forge
xorg-libxau               1.0.11               hb547adb_0    conda-forge
xorg-libxaw               1.0.14               h3422bc3_1    conda-forge
xorg-libxdmcp             1.1.3                h27ca646_0    conda-forge
[... cut ...]
yaml-cpp                  0.8.0                h13dd4ca_0    conda-forge
yarl                      1.9.4           py311h05b510d_0    conda-forge
zeromq                    4.3.5                h965bd2d_0    conda-forge
zipp                      3.17.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h53f4e23_5    conda-forge
zstd                      1.5.5                h4f39d0f_0    conda-forge
zziplib                   0.13.69              he1e0b03_1    conda-forge
```

traversaro, 2024-04-05:
> The verbose flag has been added to both `gzclient` and `gzserver` in the `empty_world.launch` from the `gazebo_ros` package. FYI there is no 24.4.0 version of libprotobuf, the only available [...]

My bad, protobuf versioning scheme is a bit confusing (https://protobuf.dev/support/version-support/). The correct version is `4.24.4`.

sylarevan, 2024-04-05:
OK. So I can confirm that installing v4.24.4 of libprotobuf makes the spawn working again.
FYI, one still needs to use #228 solution to make the camera from gazebo to publish the image on ros with `ln -s $CONDA_PREFIX/lib/gazebo-11/plugins/libCameraPlugin.dylib $CONDA_PREFIX/lib/libCameraPlugin.dylib`.

With these 2 steps, everything seems to work fine, thanks!

GokulEdakkePuram, 2024-04-05:
> > > I have the same problem in Linux. Do you have any idea on how to launch gazebo with the `--verbose` option? This should provide some more info on what is going wrong.
> > 
> > 
> > I did try running the commands with: `GAZEBO_VERBOSE=1 roslaunch turtlebot3_gazebo turtlebot3_house.launch` but i got the same output. I guess it runs automatically on verbose, I am not sure.
> > I tried these too: `export GAZEBO_LOG=debug` `roslaunch turtlebot3_gazebo turtlebot3_house.launch`
> > but again the same result.
> 
> Why did you set `GAZEBO_VERBOSE` or `GAZEBO_LOG` environment variables? Gazebo does not checks for those env variables, and I could not find any docs that mentione those variables. What you need to do is to pass `--verbose` command line option to the gazebo invocation (or `gzserver`/`gzclient`), for example modifying the corresponding launch file.

I found those commands online looking at answers posted by people who were trying to run gazebo in verbose, I am not an expert. I am still learning. Apologies if I have given wrong information.

GokulEdakkePuram, 2024-04-05:
> My bad, protobuf versioning scheme is a bit confusing (https://protobuf.dev/support/version-support/). The correct version is `4.24.4`.

This fixed the issue of the URDF model spawn. Thank you!

GokulEdakkePuram, 2024-04-05:
> OK. So I can confirm that installing v4.24.4 of libprotobuf makes the spawn working again. FYI, one still needs to use #228 solution to make the camera from gazebo to publish the image on ros with [...]

Does doing this for the laser fix the issue of it not publishing LaserScan too? I just tried to run a Subscriber that I wrote to access the LaserScan, but there was no scan topic.

traversaro, 2024-04-07:
> 
> I found those commands online looking at answers posted by people who were trying to run gazebo in verbose, I am not an expert. I am still learning. Apologies if I have given wrong information.

No problem, no need for apologies. I was asking this as if the resources that you used contain incorrect information, other users may find them, so I wanted to understand if we can fix them. Can you share the link to them? Thanks!

sylarevan, 2024-04-07:
> Does doing this for the laser fix the issue of it not publishing LaserScan too? I just tried to run a Subscriber that I wrote to access the LaserScan, but there was no scan topic.

Yes, I also had to add the link for the LDS to work.

sylarevan, 2024-04-07:
> OK. So I can confirm that installing v4.24.4 of libprotobuf makes the spawn working again..

I'm now on another computer, and I'm unable to install this specific version of libprotobuf:

```
(ros_env) sylar@m3book ~> mamba install libprotobuf==4.24.4

Looking for: ['libprotobuf==4.24.4']

conda-forge/osx-arm64                                       Using cache
conda-forge/noarch                                          Using cache
robostack-staging/noarch                                      No change
robostack-staging/osx-arm64                                   No change

Pinned packages:
  - python 3.9.*

Could not solve for environment specs
The following package could not be installed
└─ libprotobuf 4.24.4  is requested and can be installed.
```
Any idea?

Tobias-Fischer, 2024-04-07:
It seems like your are using Python 3.9 - you should be using 3.11 with the latest builds of RoboStack.

traversaro, 2024-04-07:
You have a pin to python 3.9, so probably you are using an older version of robostack packages, that was also built with an older version of protobuf. You can either create a fresh environment, or check via the error which version of protobuf those version of the plugin require.

GokulEdakkePuram, 2024-04-07:
> > Does doing this for the laser fix the issue of it not publishing LaserScan too? I just tried to run a Subscriber that I wrote to access the LaserScan, but there was no scan topic.
> 
> Yes, I also had to add the link for the LDS to work.

I tried to do that but I'm still not getting the topic running. Could you please tell me the command that you used?

sylarevan, 2024-04-07:
> I tried to do that but I'm still not getting the topic running. Could you please tell me the command that you used?

I have just entirely reinstall a new env to switch to libprotobuf v4.24.4, and I used these two commands:
- `ln -s $CONDA_PREFIX/lib/gazebo-11/plugins/libCameraPlugin.dylib $CONDA_PREFIX/lib/libCameraPlugin.dylib` 
- `ln -s $CONDA_PREFIX/lib/gazebo-11/plugins/libRayPlugin.dylib $CONDA_PREFIX/lib/libRayPlugin.dylib`
So far, both camera and lidar are working correctly.

Tobias-Fischer, 2024-12-26:
Do you think this would be resolved by a simple rebuild @traversaro thanks to https://github.com/conda-forge/gazebo-feedstock/pull/218?

traversaro, 2024-12-26:
> Do you think this would be resolved by a simple rebuild @traversaro thanks to https://github.com/conda-forge/gazebo-feedstock/pull/218?

That is my guess.

Tobias-Fischer, 2025-01-10:
Should be fixed with the rebuild - feel free to reopen if not.

closed 2025-01-10 by Tobias-Fischer