RoboStack/ros-noetic #462: libgazebo_ros_laser and libgazebo_ros_camera plugins do not work in macOS (April 2024)

traversaro, 2024-04-07:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

See https://github.com/RoboStack/ros-noetic/issues/228 .

### Installed packages

```shell
.
```

### Environment info

```shell
.
```

traversaro, 2024-04-07:
@sylarevan @GokulEdakkePuram I do not have access to mac to test, can you report the output of `env` in your environment to understand why the solution to https://github.com/RoboStack/ros-noetic/issues/228 is not working? Thanks!

traversaro, 2024-04-07:
In particular, we need to understand if `DYLD_LIBRARY_PATH` is correctly set.

sylarevan, 2024-04-07:
Here is what I have on my current environment:

```
(ros_env) sylar@m3book ~> env
SHELL=/opt/homebrew/bin/fish
ROS_VERSION=1
ITERM_PROFILE=Bash
GAZEBO_MASTER_URI=http://localhost:11345
COLORTERM=truecolor
XPC_FLAGS=0x0
TERM_PROGRAM_VERSION=3.4.23
CONDA_EXE=/opt/homebrew/Caskroom/mambaforge/base/bin/conda
_CE_M=
PKG_CONFIG_PATH=/Users/sylar/catkin_ws/devel/lib/pkgconfig:/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/pkgconfig
ROS_PYTHON_VERSION=3.9
__CFBundleIdentifier=com.googlecode.iterm2
SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.0dtmCA73vj/Listeners
ROS_OS_OVERRIDE=conda:osx
TERM_SESSION_ID=w0t0p0:27D72C5A-1EAE-4060-983F-2B0D20D0267D
ROS_PACKAGE_PATH=/Users/sylar/catkin_ws/src:/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/share
XML_CATALOG_FILES=file:///opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/etc/xml/catalog file:///etc/xml/catalog
ROSLISP_PACKAGE_DIRECTORIES=/Users/sylar/catkin_ws/devel/share/common-lisp
PWD=/Users/sylar
GSETTINGS_SCHEMA_DIR=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/share/glib-2.0/schemas
LOGNAME=sylar
CONDA_PREFIX=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env
GSETTINGS_SCHEMA_DIR_CONDA_BACKUP=
LaunchInstanceID=78402B8A-B3C3-4907-9CCE-813ADA180EFB
COMMAND_MODE=unix2003
ITERM_SESSION_ID=w0t0p0:27D72C5A-1EAE-4060-983F-2B0D20D0267D
HOME=/Users/sylar
GAZEBO_PLUGIN_PATH=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/gazebo-11/plugins:
LANG=fr_FR.UTF-8
ROS_ETC_DIR=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/etc/ros
GAZEBO_MODEL_DATABASE_URI=http://models.gazebosim.org
SECURITYSESSIONID=186ab
CONDA_PROMPT_MODIFIER=(ros_env)
AMENT_PREFIX_PATH=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env
TMPDIR=/var/folders/96/b9fy3r7s3nqb975nnsqmqc3h0000gn/T/
CMAKE_PREFIX_PATH=/Users/sylar/catkin_ws/devel:/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env
LC_TERMINAL=iTerm2
CPL_ZIP_ENCODING=UTF-8
GAZEBO_RESOURCE_PATH=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/share/gazebo-11:
GEM_HOME=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/share/rubygems/
PYTHONPATH=/Users/sylar/catkin_ws/devel/lib/python3.9/site-packages:/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/python3.9/site-packages
TERM=xterm-256color
_CE_CONDA=
USER=sylar
COLORFGBG=0;15
CONDA_SHLVL=2
LC_TERMINAL_VERSION=3.4.23
SHLVL=1
GDAL_DRIVER_PATH=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/gdalplugins
TURTLEBOT3_MODEL=burger
ROS_MASTER_URI=http://127.0.0.1:11311
SBCL_HOME=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/sbcl
PROJ_DATA=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/share/proj
ROS_HOSTNAME=127.0.0.1
XPC_SERVICE_NAME=0
CONDA_PYTHON_EXE=/opt/homebrew/Caskroom/mambaforge/base/bin/python
LD_LIBRARY_PATH=:/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/gazebo-11/plugins:
CONDA_DEFAULT_ENV=ros_env
GAZEBO_MODEL_PATH=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/share/gazebo-11/models:/Users/sylar/catkin_ws/src/projet2024/worlds/model:
GDAL_DATA=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/share/gdal
PATH=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/bin:/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/share/rubygems/bin:/opt/homebrew/Caskroom/mambaforge/base/condabin:/opt/homebrew/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin
CONDA_PREFIX_1=/opt/homebrew/Caskroom/mambaforge/base
ROS_ROOT=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/share/ros
ROS_DISTRO=noetic
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
PROJ_NETWORK=ON
OGRE_RESOURCE_PATH=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env/lib/OGRE
__CF_USER_TEXT_ENCODING=0x1F5:0x0:0x1
TERM_PROGRAM=iTerm.app
_=/usr/bin/env
```

Tobias-Fischer, 2024-04-07:
Could you please provide the output of conda list?

GokulEdakkePuram, 2024-04-07:
> @sylarevan @GokulEdakkePuram I do not have access to mac to test, can you report the output of `env` in your environment to understand why the solution to #228 is not working? Thanks!

Here is mine:

```(ros_noetic_full) gokulmnambiar@Gokuls-MacBook-Pro ~ % env
__CFBundleIdentifier=com.apple.Terminal
TMPDIR=/var/folders/_s/0cdfdzw518l_b33zt2wr2ch40000gn/T/
XPC_FLAGS=0x0
TERM=xterm-256color
DISPLAY=/private/tmp/com.apple.launchd.0RGb8WQZyM/org.xquartz:0
SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.6wkcubo4Ej/Listeners
XPC_SERVICE_NAME=0
TERM_PROGRAM=Apple_Terminal
TERM_PROGRAM_VERSION=447
TERM_SESSION_ID=8619FBFA-5E6B-4883-850C-D495C58AB5BD
SHELL=/bin/zsh
HOME=/Users/gokulmnambiar
LOGNAME=gokulmnambiar
USER=gokulmnambiar
PATH=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/share/rubygems/bin:/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/bin:/Users/gokulmnambiar/miniconda3/condabin:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin:/Library/Apple/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin
SHLVL=1
PWD=/Users/gokulmnambiar
OLDPWD=/Users/gokulmnambiar
HOMEBREW_PREFIX=/opt/homebrew
HOMEBREW_CELLAR=/opt/homebrew/Cellar
HOMEBREW_REPOSITORY=/opt/homebrew
MANPATH=/opt/homebrew/share/man::
INFOPATH=/opt/homebrew/share/info:
CONDA_EXE=/Users/gokulmnambiar/miniconda3/bin/conda
_CE_M=
_CE_CONDA=
CONDA_PYTHON_EXE=/Users/gokulmnambiar/miniconda3/bin/python
CONDA_SHLVL=2
CONDA_PREFIX=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full
CONDA_DEFAULT_ENV=ros_noetic_full
CONDA_PROMPT_MODIFIER=(ros_noetic_full) 
TURTLEBOT3_MODEL=burger
CMAKE_PREFIX_PATH=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full
PKG_CONFIG_PATH=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/lib/pkgconfig
PYTHONPATH=/Users/gokulmnambiar/gaz_new/devel/lib/python3.11/site-packages:/Users/gokulmnambiar/catkin_ws_new_ws/devel/lib/python3.11/site-packages:/Users/gokulmnambiar/catkin_ws/devel/lib/python3.11/site-packages
DYLD_LIBRARY_PATH=:/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/lib/gazebo-11/plugins:
CONDA_PREFIX_1=/Users/gokulmnambiar/miniconda3
GFORTRAN=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/bin/arm64-apple-darwin20.0.0-gfortran
FC_FOR_BUILD=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/bin/arm64-apple-darwin20.0.0-gfortran
FFLAGS=-march=armv8.3-a -ftree-vectorize -fPIC -fno-stack-protector -O2 -pipe -isystem /Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/include
FORTRANFLAGS=-march=armv8.3-a -ftree-vectorize -fPIC -fno-stack-protector -O2 -pipe -isystem /Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/include
DEBUG_FFLAGS=-march=armv8.3-a -ftree-vectorize -fPIC -fno-stack-protector -O2 -pipe -isystem /Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/include -march=armv8.3-a -ftree-vectorize -fPIC -fno-stack-protector -O2 -pipe -Og -g -Wall -Wextra -fcheck=all -fbacktrace -fimplicit-none -fvar-tracking-assignments
DEBUG_FORTRANFLAGS=-march=armv8.3-a -ftree-vectorize -fPIC -fno-stack-protector -O2 -pipe -isystem /Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/include -march=armv8.3-a -ftree-vectorize -fPIC -fno-stack-protector -O2 -pipe -Og -g -Wall -Wextra -fcheck=all -fbacktrace -fimplicit-none -fvar-tracking-assignments
FC=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/bin/arm64-apple-darwin20.0.0-gfortran
F77=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/bin/arm64-apple-darwin20.0.0-gfortran
F90=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/bin/arm64-apple-darwin20.0.0-gfortran
F95=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/bin/arm64-apple-darwin20.0.0-gfortran
CONDA_BACKUP_HOST=Gokuls-MacBook-Pro.local
HOST=arm64-apple-darwin20.0.0
CONDA_TOOLCHAIN_HOST=arm64-apple-darwin20.0.0
CONDA_TOOLCHAIN_BUILD=arm64-apple-darwin20.0.0
AR=arm64-apple-darwin20.0.0-ar
AS=arm64-apple-darwin20.0.0-as
CHECKSYMS=arm64-apple-darwin20.0.0-checksyms
INSTALL_NAME_TOOL=arm64-apple-darwin20.0.0-install_name_tool
LIBTOOL=arm64-apple-darwin20.0.0-libtool
LIPO=arm64-apple-darwin20.0.0-lipo
NM=arm64-apple-darwin20.0.0-nm
NMEDIT=arm64-apple-darwin20.0.0-nmedit
OTOOL=arm64-apple-darwin20.0.0-otool
PAGESTUFF=arm64-apple-darwin20.0.0-pagestuff
RANLIB=arm64-apple-darwin20.0.0-ranlib
REDO_PREBINDING=arm64-apple-darwin20.0.0-redo_prebinding
SEG_ADDR_TABLE=arm64-apple-darwin20.0.0-seg_addr_table
SEG_HACK=arm64-apple-darwin20.0.0-seg_hack
[... 35 lines cut ...]
GAZEBO_MODEL_PATH=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/share/gazebo-11/models:
LD_LIBRARY_PATH=:/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/lib/gazebo-11/plugins:
OGRE_RESOURCE_PATH=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/lib/OGRE
GDAL_DATA=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/share/gdal
GDAL_DRIVER_PATH=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/lib/gdalplugins
CPL_ZIP_ENCODING=UTF-8
GSETTINGS_SCHEMA_DIR_CONDA_BACKUP=
GSETTINGS_SCHEMA_DIR=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/share/glib-2.0/schemas
XML_CATALOG_FILES=file:///Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/etc/xml/catalog file:///etc/xml/catalog
PROJ_DATA=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/share/proj
PROJ_NETWORK=ON
ROS_DISTRO=noetic
[... cut ...]
ROS_PACKAGE_PATH=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/share
ROS_PYTHON_VERSION=3.11
ROS_VERSION=1
ROS_ROOT=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/share/ros
ROS_MASTER_URI=http://localhost:11311
ROSLISP_PACKAGE_DIRECTORIES=
ROS_OS_OVERRIDE=conda:osx
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
AMENT_PREFIX_PATH=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full
GEM_HOME=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/share/rubygems/
LC_CTYPE=UTF-8
_=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/bin/env```

GokulEdakkePuram, 2024-04-07:
> @sylarevan @GokulEdakkePuram I do not have access to mac to test, can you report the output of `env` in your environment to understand why the solution to #228 is not working? Thanks!

Here is mine

```
(ros_noetic_full) gokulmnambiar@Gokuls-MacBook-Pro ~ % env
__CFBundleIdentifier=com.apple.Terminal
TMPDIR=/var/folders/_s/0cdfdzw518l_b33zt2wr2ch40000gn/T/
XPC_FLAGS=0x0
TERM=xterm-256color
DISPLAY=/private/tmp/com.apple.launchd.0RGb8WQZyM/org.xquartz:0
SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.6wkcubo4Ej/Listeners
XPC_SERVICE_NAME=0
TERM_PROGRAM=Apple_Terminal
TERM_PROGRAM_VERSION=447
TERM_SESSION_ID=8619FBFA-5E6B-4883-850C-D495C58AB5BD
SHELL=/bin/zsh
HOME=/Users/gokulmnambiar
LOGNAME=gokulmnambiar
USER=gokulmnambiar
PATH=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/share/rubygems/bin:/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/bin:/Users/gokulmnambiar/miniconda3/condabin:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin:/Library/Apple/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin
SHLVL=1
PWD=/Users/gokulmnambiar
OLDPWD=/Users/gokulmnambiar
HOMEBREW_PREFIX=/opt/homebrew
HOMEBREW_CELLAR=/opt/homebrew/Cellar
HOMEBREW_REPOSITORY=/opt/homebrew
MANPATH=/opt/homebrew/share/man::
INFOPATH=/opt/homebrew/share/info:
CONDA_EXE=/Users/gokulmnambiar/miniconda3/bin/conda
_CE_M=
_CE_CONDA=
CONDA_PYTHON_EXE=/Users/gokulmnambiar/miniconda3/bin/python
CONDA_SHLVL=2
CONDA_PREFIX=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full
CONDA_DEFAULT_ENV=ros_noetic_full
CONDA_PROMPT_MODIFIER=(ros_noetic_full) 
TURTLEBOT3_MODEL=burger
CMAKE_PREFIX_PATH=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full
PKG_CONFIG_PATH=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/lib/pkgconfig
PYTHONPATH=/Users/gokulmnambiar/gaz_new/devel/lib/python3.11/site-packages:/Users/gokulmnambiar/catkin_ws_new_ws/devel/lib/python3.11/site-packages:/Users/gokulmnambiar/catkin_ws/devel/lib/python3.11/site-packages
DYLD_LIBRARY_PATH=:/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/lib/gazebo-11/plugins:
CONDA_PREFIX_1=/Users/gokulmnambiar/miniconda3
GFORTRAN=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/bin/arm64-apple-darwin20.0.0-gfortran
FC_FOR_BUILD=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/bin/arm64-apple-darwin20.0.0-gfortran
FFLAGS=-march=armv8.3-a -ftree-vectorize -fPIC -fno-stack-protector -O2 -pipe -isystem /Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/include
FORTRANFLAGS=-march=armv8.3-a -ftree-vectorize -fPIC -fno-stack-protector -O2 -pipe -isystem /Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/include
DEBUG_FFLAGS=-march=armv8.3-a -ftree-vectorize -fPIC -fno-stack-protector -O2 -pipe -isystem /Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/include -march=armv8.3-a -ftree-vectorize -fPIC -fno-stack-protector -O2 -pipe -Og -g -Wall -Wextra -fcheck=all -fbacktrace -fimplicit-none -fvar-tracking-assignments
DEBUG_FORTRANFLAGS=-march=armv8.3-a -ftree-vectorize -fPIC -fno-stack-protector -O2 -pipe -isystem /Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/include -march=armv8.3-a -ftree-vectorize -fPIC -fno-stack-protector -O2 -pipe -Og -g -Wall -Wextra -fcheck=all -fbacktrace -fimplicit-none -fvar-tracking-assignments
FC=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/bin/arm64-apple-darwin20.0.0-gfortran
F77=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/bin/arm64-apple-darwin20.0.0-gfortran
F90=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/bin/arm64-apple-darwin20.0.0-gfortran
F95=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/bin/arm64-apple-darwin20.0.0-gfortran
CONDA_BACKUP_HOST=Gokuls-MacBook-Pro.local
HOST=arm64-apple-darwin20.0.0
CONDA_TOOLCHAIN_HOST=arm64-apple-darwin20.0.0
CONDA_TOOLCHAIN_BUILD=arm64-apple-darwin20.0.0
AR=arm64-apple-darwin20.0.0-ar
AS=arm64-apple-darwin20.0.0-as
CHECKSYMS=arm64-apple-darwin20.0.0-checksyms
INSTALL_NAME_TOOL=arm64-apple-darwin20.0.0-install_name_tool
LIBTOOL=arm64-apple-darwin20.0.0-libtool
LIPO=arm64-apple-darwin20.0.0-lipo
NM=arm64-apple-darwin20.0.0-nm
NMEDIT=arm64-apple-darwin20.0.0-nmedit
OTOOL=arm64-apple-darwin20.0.0-otool
PAGESTUFF=arm64-apple-darwin20.0.0-pagestuff
RANLIB=arm64-apple-darwin20.0.0-ranlib
REDO_PREBINDING=arm64-apple-darwin20.0.0-redo_prebinding
SEG_ADDR_TABLE=arm64-apple-darwin20.0.0-seg_addr_table
[... 38 lines cut ...]
OGRE_RESOURCE_PATH=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/lib/OGRE
GDAL_DATA=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/share/gdal
GDAL_DRIVER_PATH=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/lib/gdalplugins
CPL_ZIP_ENCODING=UTF-8
GSETTINGS_SCHEMA_DIR_CONDA_BACKUP=
GSETTINGS_SCHEMA_DIR=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/share/glib-2.0/schemas
XML_CATALOG_FILES=file:///Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/etc/xml/catalog file:///etc/xml/catalog
PROJ_DATA=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/share/proj
PROJ_NETWORK=ON
ROS_DISTRO=noetic
ROS_ETC_DIR=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/etc/ros
ROS_PACKAGE_PATH=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/share
ROS_PYTHON_VERSION=3.11
ROS_VERSION=1
ROS_ROOT=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/share/ros
ROS_MASTER_URI=http://localhost:11311
ROSLISP_PACKAGE_DIRECTORIES=
ROS_OS_OVERRIDE=conda:osx
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
AMENT_PREFIX_PATH=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full
GEM_HOME=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/share/rubygems/
LC_CTYPE=UTF-8
_=/Users/gokulmnambiar/miniconda3/envs/ros_noetic_full/bin/env

```

GokulEdakkePuram, 2024-04-07:
> Could you please provide the output of conda list?

```

(ros_noetic_full) gokulmnambiar@Gokuls-MacBook-Pro ~ % conda list
# packages in environment at /Users/gokulmnambiar/miniconda3/envs/ros_noetic_full:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        43                       unix_0    conda-forge
aiohttp                   3.9.3           py311h05b510d_1    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.7.1                h463b476_0    conda-forge
apr                       1.7.0                h4614cfb_6    conda-forge
argcomplete               3.2.3              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                he63ff86_2    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
aws-c-auth                0.7.16               h0d2f7a6_8    conda-forge
aws-c-cal                 0.6.10               h677d54c_2    conda-forge
aws-c-common              0.9.14               h93a5062_0    conda-forge
aws-c-compression         0.2.18               h677d54c_2    conda-forge
aws-c-event-stream        0.4.2                h59ac3ca_6    conda-forge
aws-c-http                0.8.1                hfe5d766_7    conda-forge
aws-c-io                  0.14.6               h9ac2cdb_2    conda-forge
aws-c-mqtt                0.10.3               hb8a1441_2    conda-forge
aws-c-s3                  0.5.5                h4398043_0    conda-forge
aws-c-sdkutils            0.1.15               h677d54c_2    conda-forge
aws-checksums             0.1.18               h677d54c_2    conda-forge
aws-crt-cpp               0.26.4               ha2646ad_3    conda-forge
aws-sdk-cpp               1.11.267             ha761c4c_4    conda-forge
azure-core-cpp            1.10.3               he231e37_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h6aa02a4_0    conda-forge
azure-storage-common-cpp  12.5.0               h607ffeb_2    conda-forge
bcrypt                    4.1.2           py311h94f323b_0    conda-forge
blosc                     1.21.5               hc338f07_0    conda-forge
boost                     1.82.0               hf3b2ce4_6    conda-forge
boost-cpp                 1.82.0               hca5e981_6    conda-forge
brotli                    1.1.0                hb547adb_1    conda-forge
brotli-bin                1.1.0                hb547adb_1    conda-forge
bullet-cpp                3.25            py311h6e08293_2    conda-forge
bzip2                     1.0.8                h93a5062_5    conda-forge
c-ares                    1.28.1               h93a5062_0    conda-forge
c-compiler                1.7.0                h6aa9301_0    conda-forge
ca-certificates           2024.2.2             hf0a4a13_0    conda-forge
cairo                     1.18.0               hd1e100b_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
cctools                   986                  h4faf515_0    conda-forge
cctools_osx-arm64         986                  h62378fb_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py311h4a08483_0    conda-forge
cfitsio                   4.3.1                h808cd33_0    conda-forge
clang                     16.0.6               h30cc82d_6    conda-forge
clang-16                  16.0.6          default_he012953_6    conda-forge
clang_impl_osx-arm64      16.0.6              hc421ffc_11    conda-forge
clang_osx-arm64           16.0.6              h54d7cd3_11    conda-forge
clangxx                   16.0.6          default_h4cf2255_6    conda-forge
clangxx_impl_osx-arm64    16.0.6              hcd7bac0_11    conda-forge
clangxx_osx-arm64         16.0.6              h54d7cd3_11    conda-forge
cmake                     3.28.3               h50fd54c_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py311h267d04e_1    conda-forge
colcon-core               0.16.0             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.3.0              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
[... 564 lines cut ...]
xorg-libsm                1.2.4                hb547adb_0    conda-forge
xorg-libx11               1.8.9                h570a39f_0    conda-forge
xorg-libxau               1.0.11               hb547adb_0    conda-forge
xorg-libxaw               1.0.14               h3422bc3_1    conda-forge
xorg-libxdmcp             1.1.3                h27ca646_0    conda-forge
xorg-libxext              1.3.4                h1a8c8d9_2    conda-forge
xorg-libxfixes            5.0.3             h3422bc3_1004    conda-forge
xorg-libxmu               1.1.3                h3422bc3_0    conda-forge
xorg-libxpm               3.5.17               hb547adb_0    conda-forge
[... cut ...]
yaml-cpp                  0.8.0                h13dd4ca_0    conda-forge
yarl                      1.9.4           py311h05b510d_0    conda-forge
zeromq                    4.3.5                hebf3989_1    conda-forge
zipp                      3.17.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h53f4e23_5    conda-forge
zstd                      1.5.5                h4f39d0f_0    conda-forge
zziplib                   0.13.69              he1e0b03_1    conda-forge

```

GokulEdakkePuram, 2024-04-07:
> In particular, we need to understand if `DYLD_LIBRARY_PATH` is correctly set after [conda-forge/gazebo-feedstock#163](https://github.com/conda-forge/gazebo-feedstock/pull/163) . It would be helpful if you report the exact version of gazebo you are using, thanks!

```
(ros_noetic_full) gokulmnambiar@Gokuls-MacBook-Pro ~ % conda list gazebo
# packages in environment at /Users/gokulmnambiar/miniconda3/envs/ros_noetic_full:
#
# Name                    Version                   Build  Channel
gazebo                    11.14.0              h9ba9e67_6    conda-forge
ros-noetic-gazebo-dev     2.9.2           py311hffdb928_8    robostack-staging
ros-noetic-gazebo-msgs    2.9.2           py311hbfa132e_8    robostack-staging
ros-noetic-gazebo-plugins 2.9.2           py311h7f7f440_8    robostack-staging
ros-noetic-gazebo-ros     2.9.2           py311h7f7f440_8    robostack-staging
ros-noetic-gazebo-ros-control 2.9.2           py311h7f7f440_8    robostack-staging
ros-noetic-gazebo-ros-pkgs 2.9.2           py311hbfa132e_8    robostack-staging
```

sylarevan, 2024-04-07:
After a new installation in a new environment, I get:

<details>
<summary>env</summary>

```
(ros_env2) sylar@m3book ~> env
SHELL=/opt/homebrew/bin/fish
ROS_VERSION=1
ITERM_PROFILE=Bash
GAZEBO_MASTER_URI=http://localhost:11345
COLORTERM=truecolor
XPC_FLAGS=0x0
TERM_PROGRAM_VERSION=3.4.23
CONDA_EXE=/opt/homebrew/Caskroom/mambaforge/base/bin/conda
_CE_M=
PKG_CONFIG_PATH=/Users/sylar/catkin_ws/devel/lib/pkgconfig:/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/lib/pkgconfig
ROS_PYTHON_VERSION=3.11
__CFBundleIdentifier=com.googlecode.iterm2
SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.0dtmCA73vj/Listeners
ROS_OS_OVERRIDE=conda:osx
TERM_SESSION_ID=w0t0p0:C46ECBA4-9E98-4114-A06C-B80D1145AEB7
ROS_PACKAGE_PATH=/Users/sylar/catkin_ws/src:/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/share
XML_CATALOG_FILES=file:///opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/etc/xml/catalog file:///etc/xml/catalog
ROSLISP_PACKAGE_DIRECTORIES=/Users/sylar/catkin_ws/devel/share/common-lisp
PWD=/Users/sylar
GSETTINGS_SCHEMA_DIR=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/share/glib-2.0/schemas
LOGNAME=sylar
CONDA_PREFIX=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2
GSETTINGS_SCHEMA_DIR_CONDA_BACKUP=
LaunchInstanceID=78402B8A-B3C3-4907-9CCE-813ADA180EFB
COMMAND_MODE=unix2003
ITERM_SESSION_ID=w0t0p0:C46ECBA4-9E98-4114-A06C-B80D1145AEB7
HOME=/Users/sylar
GAZEBO_PLUGIN_PATH=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/lib/gazebo-11/plugins:
LANG=fr_FR.UTF-8
ROS_ETC_DIR=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/etc/ros
GAZEBO_MODEL_DATABASE_URI=http://models.gazebosim.org
SECURITYSESSIONID=186ab
CONDA_PROMPT_MODIFIER=(ros_env2)
AMENT_PREFIX_PATH=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2
TMPDIR=/var/folders/96/b9fy3r7s3nqb975nnsqmqc3h0000gn/T/
CMAKE_PREFIX_PATH=/Users/sylar/catkin_ws/devel:/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2
LC_TERMINAL=iTerm2
CPL_ZIP_ENCODING=UTF-8
GAZEBO_RESOURCE_PATH=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/share/gazebo-11:
GEM_HOME=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/share/rubygems/
PYTHONPATH=/Users/sylar/catkin_ws/devel/lib/python3.11/site-packages:/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/lib/python3.11/site-packages
TERM=xterm-256color
_CE_CONDA=
USER=sylar
COLORFGBG=0;15
CONDA_SHLVL=2
LC_TERMINAL_VERSION=3.4.23
SHLVL=1
GDAL_DRIVER_PATH=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/lib/gdalplugins
TURTLEBOT3_MODEL=burger
ROS_MASTER_URI=http://127.0.0.1:11311
SBCL_HOME=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/lib/sbcl
PROJ_DATA=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/share/proj
ROS_HOSTNAME=127.0.0.1
XPC_SERVICE_NAME=0
CONDA_PYTHON_EXE=/opt/homebrew/Caskroom/mambaforge/base/bin/python
LD_LIBRARY_PATH=:/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/lib/gazebo-11/plugins:
CONDA_DEFAULT_ENV=ros_env2
GAZEBO_MODEL_PATH=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/share/gazebo-11/models:/Users/sylar/catkin_ws/src/projet2024/worlds/model:
GDAL_DATA=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/share/gdal
PATH=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/bin:/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/share/rubygems/bin:/opt/homebrew/Caskroom/mambaforge/base/condabin:/opt/homebrew/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin
CONDA_PREFIX_1=/opt/homebrew/Caskroom/mambaforge/base
ROS_ROOT=/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/share/ros
[... 606 lines cut ...]
xorg-libice               1.1.1                hb547adb_0    conda-forge
xorg-libsm                1.2.4                hb547adb_0    conda-forge
xorg-libx11               1.8.9                h570a39f_0    conda-forge
xorg-libxau               1.0.11               hb547adb_0    conda-forge
xorg-libxaw               1.0.14               h3422bc3_1    conda-forge
xorg-libxdmcp             1.1.3                h27ca646_0    conda-forge
xorg-libxext              1.3.4                h1a8c8d9_2    conda-forge
xorg-libxfixes            5.0.3             h3422bc3_1004    conda-forge
xorg-libxmu               1.1.3                h3422bc3_0    conda-forge
xorg-libxpm               3.5.17               hb547adb_0    conda-forge
xorg-libxrender           0.9.11               hb547adb_0    conda-forge
xorg-libxt                1.3.0                hb547adb_1    conda-forge
xorg-renderproto          0.11.1            h27ca646_1002    conda-forge
xorg-xextproto            7.3.0             h1a8c8d9_1003    conda-forge
xorg-xproto               7.0.31            h27ca646_1007    conda-forge
xz                        5.2.6                h57fd34a_0    conda-forge
yaml                      0.2.5                h3422bc3_2    conda-forge
yaml-cpp                  0.8.0                h13dd4ca_0    conda-forge
yarl                      1.9.4           py311h05b510d_0    conda-forge
zeromq                    4.3.5                hebf3989_1    conda-forge
zlib                      1.2.13               h53f4e23_5    conda-forge
zstd                      1.5.5                h4f39d0f_0    conda-forge
zziplib                   0.13.69              he1e0b03_1    conda-forge
```
</details>

traversaro, 2024-04-07:
Interestingly, in the last case `DYLD_LIBRARY_PATH` is correctly set. Do you still need to add manually the symbolic links to get the plugins to work?

sylarevan, 2024-04-07:
Yes, I had to. Otherwise no camera nor laser on ROS.

traversaro, 2024-04-07:
Thanks. I suspect that the solution in https://github.com/ros-simulation/gazebo_ros_pkgs/pull/974 is not working properly in macOS . 

If you are interesting in getting a solution, could you please:
*  run the gazebo with verbose to get an error similar to the one reported in https://github.com/RoboStack/ros-noetic/issues/228#issuecomment-1035689132
* Report the output of `otool -L $CONDA_PREFIX/lib/libgazebo_ros_laser.dylib`

sylarevan, 2024-04-07:
Here is what i get:
- verbose mode of gzserver:
```
[Err] [Plugin.hh:212] Failed to load plugin libgazebo_ros_camera.so: dlopen(libgazebo_ros_camera.so, 0x0009): tried: 'libgazebo_ros_camera.so' (no such file), '/System/Volumes/Preboot/Cryptexes/OSlibgazebo_ros_camera.so' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/lib/libgazebo_ros_camera.so' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/bin/../lib/libgazebo_ros_camera.so' (no such file), '/usr/lib/libgazebo_ros_camera.so' (no such file, not in dyld cache), 'libgazebo_ros_camera.so' (no such file), '/usr/local/lib/libgazebo_ros_camera.so' (no such file), '/usr/lib/libgazebo_ros_camera.so' (no such file, not in dyld cache)
Failed to load plugin /opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/share/gazebo_plugins/../../lib/libgazebo_ros_camera.dylib: dlopen(/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/share/gazebo_plugins/../../lib/libgazebo_ros_camera.dylib, 0x0009): Library not loaded: @rpath/libCameraPlugin.dylib
  Referenced from: <3ABED57A-B693-33C5-92CB-71E429167FF7> /opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/lib/libgazebo_ros_camera.dylib
  Reason: tried: '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/lib/libCameraPlugin.dylib' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/lib/libCameraPlugin.dylib' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/lib/libCameraPlugin.dylib' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/bin/../lib/libCameraPlugin.dylib' (no such file)
[Err] [Plugin.hh:212] Failed to load plugin libgazebo_ros_laser.so: dlopen(libgazebo_ros_laser.so, 0x0009): tried: 'libgazebo_ros_laser.so' (no such file), '/System/Volumes/Preboot/Cryptexes/OSlibgazebo_ros_laser.so' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/lib/libgazebo_ros_laser.so' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/bin/../lib/libgazebo_ros_laser.so' (no such file), '/usr/lib/libgazebo_ros_laser.so' (no such file, not in dyld cache), 'libgazebo_ros_laser.so' (no such file), '/usr/local/lib/libgazebo_ros_laser.so' (no such file), '/usr/lib/libgazebo_ros_laser.so' (no such file, not in dyld cache)
Failed to load plugin /opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/share/gazebo_plugins/../../lib/libgazebo_ros_laser.dylib: dlopen(/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/share/gazebo_plugins/../../lib/libgazebo_ros_laser.dylib, 0x0009): Library not loaded: @rpath/libRayPlugin.dylib
  Referenced from: <8FA92339-E219-38DC-B42C-C35EEF0047EC> /opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/lib/libgazebo_ros_laser.dylib
  Reason: tried: '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/lib/libRayPlugin.dylib' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/lib/libRayPlugin.dylib' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/lib/libRayPlugin.dylib' (no such file), '/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/bin/../lib/libRayPlugin.dylib' (no such file)
```

- otool:
```
(ros_env2) sylar@m3book ~> otool -L $CONDA_PREFIX/lib/libgazebo_ros_laser.dylib
/opt/homebrew/Caskroom/mambaforge/base/envs/ros_env2/lib/libgazebo_ros_laser.dylib:
	@rpath/libgazebo_ros_laser.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libRayPlugin.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libgazebo_sensors.11.dylib (compatibility version 11.0.0, current version 11.14.0)
	@rpath/libgazebo_physics.11.dylib (compatibility version 11.0.0, current version 11.14.0)
	@rpath/libgazebo_transport.11.dylib (compatibility version 11.0.0, current version 11.14.0)
	@rpath/libgazebo_msgs.11.dylib (compatibility version 11.0.0, current version 11.14.0)
	@rpath/libgazebo_common.11.dylib (compatibility version 11.0.0, current version 11.14.0)
	@rpath/libboost_thread.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libsdformat9.9.dylib (compatibility version 9.0.0, current version 9.8.0)
	@rpath/libtf.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1336.0.0)
	@rpath/libroscpp.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librosconsole.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libroscpp_serialization.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/librostime.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libprotobuf.24.4.0.dylib (compatibility version 0.0.0, current version 24.4.0)
	@rpath/libignition-math6.6.dylib (compatibility version 6.0.0, current version 6.15.1)
	@rpath/libc++.1.dylib (compatibility version 1.0.0, current version 1.0.0)
```

traversaro, 2024-04-07:
If I understood correctly from https://briandfoy.github.io/macos-s-system-integrity-protection-sanitizes-your-environment/, it seems that `DYLD_LIBRARY_PATH` is sanitized. At this point, we have two options:
* Patch gazebo_ros_plugins to add a rpath pointing to $CONDA_PREFIX/lib/gazebo-11/plugins.
* Patch gazebo to symlink all the plugins also in `$CONDA_PREFIX/lib`

The second solution seems simpler.

Tobias-Fischer, 2024-07-30:
This should be done in https://github.com/conda-forge/gazebo-feedstock/pull/218

Tobias-Fischer, 2024-07-30:
Keeping it open as it probably needs a rebuild.

Tobias-Fischer, 2025-01-10:
Should be fixed with the rebuild - feel free to reopen if not.

closed 2025-01-10 by Tobias-Fischer