RoboStack/ros-noetic #466: ros-noetic-velodyne-pointcloud listed as available in noetic for linux-64 but requires ros-distro-mutex 0.5.*

mantkiew, 2024-04-17:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

```
$ mamba install ros-noetic-velodyne-pointcloud
Looking for: ['ros-noetic-velodyne-pointcloud']
conda-forge/linux-64                                        Using cache
conda-forge/noarch                                          Using cache
robostack-staging/noarch                                      No change
robostack-staging/linux-64                                    No change

Pinned packages:
  - python 3.9.*

Could not solve for environment specs
The following packages are incompatible
├─ libprotobuf is installable with the potential options
│  ├─ libprotobuf [3.10.0|3.10.1|...|5.26.1], which can be installed;
│  └─ libprotobuf 4.24.3 would require
│     └─ libabseil >=20230802.1,<20230803.0a0 , which can be installed;
└─ ros-noetic-velodyne-pointcloud is not installable because it requires
   └─ ros-distro-mutex 0.5.* , which requires
      ├─ libabseil 20230802 , which conflicts with any installable versions previously reported;
      └─ libprotobuf 4.24.3 , which conflicts with any installable versions previously reported.
```

### Installed packages

```shell
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
adwaita-icon-theme        43                       unix_0    conda-forge
aiohttp                   3.9.3            py39hd1e30aa_1    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.8                h166bdaf_0    conda-forge
aom                       3.5.0                h27087fc_0    conda-forge
apr                       1.7.0                hd590300_6    conda-forge
argcomplete               3.3.0              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                hf40c2ba_0    conda-forge
async-timeout             4.0.3              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
bcrypt                    4.1.2            py39h9fdd4d6_0    conda-forge
binutils                  2.40                 hdd6e379_0    conda-forge
binutils_impl_linux-64    2.40                 hf600244_0    conda-forge
binutils_linux-64         2.40                 hdade7a5_3    conda-forge
blosc                     1.21.5               hc2324a3_1    conda-forge
boost                     1.78.0           py39hda80f44_5    conda-forge
boost-cpp                 1.78.0               h5adbc97_2    conda-forge
brotli                    1.1.0                hd590300_1    conda-forge
brotli-bin                1.1.0                hd590300_1    conda-forge
bullet-cpp                3.21                 h4661b88_4    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
c-ares                    1.28.1               hd590300_0    conda-forge
c-compiler                1.7.0                hd590300_0    conda-forge
ca-certificates           2024.2.2             hbcca054_0    conda-forge
cairo                     1.16.0            ha61ee94_1014    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
ceres-solver              2.2.0                h30ec75d_2    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0           py39h7a31438_0    conda-forge
cfitsio                   4.2.0                hd9d235c_0    conda-forge
cmake                     3.26.3               h077f3f9_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
[... 639 lines cut ...]
         python version : 3.10.14.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=skylake
                          __conda=24.3.0=0
                          __cuda=12.4=0
                          __glibc=2.39=0
                          __linux=6.8.0=0
                          __unix=0=0
       base environment : /home/ma/miniforge3  (writable)
      conda av data dir : /home/ma/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/linux-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/ma/miniforge3/pkgs
                          /home/ma/.conda/pkgs
       envs directories : /home/ma/miniforge3/envs
                          /home/ma/.conda/envs
               platform : linux-64
             user-agent : conda/24.3.0 requests/2.31.0 CPython/3.10.14 Linux/6.8.0-22-generic ubuntu/24.04 glibc/2.39 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.8
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

mantkiew, 2024-04-17:
It seems that the `velodyne` packages cannot be installed together with `ros-noetic-desktop` dues to conflict between `ros-distro-mutex 0.4.* noetic` and `ros-distro-mutex 0.5.* noetic`.
I was able to install these packages in an empty environment.

mantkiew, 2024-04-17:
I was able to narrow down the conflict:
```
Could not solve for environment specs
The following packages are incompatible
├─ libprotobuf 4.24.3  is installable and it requires
│  └─ libabseil >=20230802.1,<20230803.0a0 , which can be installed;
└─ ros-noetic-velodyne-pointcloud is not installable because it requires
   └─ ros-distro-mutex 0.5.* , which requires
      └─ libabseil 20230802 , which conflicts with any installable versions previously reported.
```
the root cause is the constraint `libabseil >=20230802.1` conflicting with `libabseil 20230802`.

Tobias-Fischer, 2024-04-17:
Hi @mantkiew - thanks for reporting this! It seems like you’re still using old packages that were built for Python 3.9. If you create a new environment with Python 3.11 you should be able to install the packages without issues.

mantkiew, 2024-04-17:
I have only followed the instructions from https://robostack.github.io/GettingStarted.html and ended up with pinned Python 3.9. I am not sure what else I should have done. Just installing `ros-noetic-desktop` does it, I think.

Tobias-Fischer, 2024-04-17:
I’ve updated instructions to explicitly install Python 3.11.

mantkiew, 2024-04-17:
Thanks. Unfortunately, the `ros-distro-mutex` is still conflicting. I did as follows
```
$ mamba create -n ros-env python=3.11
$ mamba activate ros-env
$ conda config --env --add channels robostack-staging
$ mamba install ros-noetic-desktop ros-noetic-velodyne-pointcloud
...
│  └─ ros-noetic-desktop 1.5.0 would require
│     └─ ros-distro-mutex 0.4.* , which can be installed;
└─ ros-noetic-velodyne-pointcloud is not installable because it requires
   └─ ros-distro-mutex 0.5.* , which requires
      └─ libprotobuf 4.24.3 , which conflicts with any installable versions previously reported.
```

Tobias-Fischer, 2024-04-17:
/cc @traversaro - I think this is because of https://github.com/RoboStack/ros-noetic/pull/464

Do you think we need to rebuild?

olivier-stasse, 2024-04-18:
For what is worth I am having the same problem with a fresh new environment using python 3.11.

traversaro, 2024-04-18:
I reduced the problem to:

~~~
mamba create -n test -c conda-forge -c robostack-staging ros-noetic-cv-bridge ros-distro-mutex==0.5.*
~~~

That fails with:
~~~
traversaro@IITBMP014LW012:~$ mamba create -n test -c conda-forge -c robostack-staging ros-noetic-cv-bridge ros-distro-mutex==0.5.*

Looking for: ['ros-noetic-cv-bridge', 'ros-distro-mutex=0.5']

conda-forge/linux-64                                        Using cache
conda-forge/noarch                                          Using cache
robostack-staging/linux-64                                  Using cache
robostack-staging/noarch                                    Using cache
Could not solve for environment specs
The following packages are incompatible
├─ ros-distro-mutex 0.5**  is installable and it requires
│  └─ libprotobuf 4.24.3 , which can be installed;
└─ ros-noetic-cv-bridge is not installable because there are no viable options
   ├─ ros-noetic-cv-bridge 1.16.2 would require
   │  └─ py-opencv >=4.9.0,<5.0a0  but there are no viable options
   │     ├─ py-opencv 4.9.0 would require
   │     │  └─ libprotobuf >=4.25.3,<4.25.4.0a0 , which conflicts with any installable versions previously reported;
   │     ├─ py-opencv 4.9.0 would require
   │     │  └─ libprotobuf >=4.25.2,<4.25.3.0a0 , which conflicts with any installable versions previously reported;
   │     ├─ py-opencv 4.9.0 would require
   │     │  └─ libprotobuf >=4.24.4,<4.24.5.0a0 , which conflicts with any installable versions previously reported;
   │     ├─ py-opencv 4.9.0 would require
   │     │  └─ libprotobuf >=4.25.1,<4.25.2.0a0 , which conflicts with any installable versions previously reported;
   │     └─ py-opencv 4.9.0 would require
   │        └─ libprotobuf >=3.21.12,<3.22.0a0 , which conflicts with any installable versions previously reported;
   └─ ros-noetic-cv-bridge 1.16.2 would require
      └─ ros-distro-mutex 0.4.* noetic, which conflicts with any installable versions previously reported.
~~~

What it happened with the last rebuild is that we built some packages with a given version of libprotobuf (`4.24.3`), and others with another (something newer, as we used `py-opencv` 4.9.0 that does not have a `4.24.3` build. I can't think of a solution without going for a rebuild, probably with the latest pinned version of libprotobuf. The main problem here is that libprotobuf is also a dependency of most machine learning packages that frequently are not migrated quickly.

traversaro, 2024-04-21:
Probably a easier fix in the short term may be to revert https://github.com/RoboStack/ros-noetic/pull/464, we will have again the gazebo plugin problems, but that seems less problematic that the impossibility of installing `ros-noetic-desktop`.

Tobias-Fischer, 2024-04-21:
Agreed! Will aim for a rebuild in the not too distant future.

closed 2024-04-21 by Tobias-Fischer

mantkiew, 2024-04-22:
Thanks a lot! I was able to install this together with `ros-noetic-desktop`. Just did `mamba update --all` before and installation went through.

commit 7f743314 (2024-04-21): Revert #464 to fix #466