RoboStack/ros-noetic #469: Package request: PACKAGE_NAME

kneave, 2024-04-30:
### Package name

depthai-ros

### Operating System(s)

- [X] linux-64
- [X] linux-aarch64
- [ ] osx-64
- [ ] osx-arm64
- [X] win-64

### Additional context (optional)

This is the ROS package for the DepthAI range of cameras and would be incredible handy to have available in RoboStack please to enable much easier use of this, especially on SBC like the Raspberry Pi where building packages can be a bit of a nightmare.

### Links and references (optional)

https://github.com/luxonis/depthai-ros

traversaro, 2024-05-13:
Just by chance, I started working with depthai-core (not ROS) at my job, and I found that it uses `hunter` to deal with C++ dependencies, and it is not obvious to me how that interacts with apt and/or conda. Do you have any pointer on how depthai-ros is packaged in the ROS buildfarm?

See also https://github.com/RoboStack/ros-noetic/issues/151 for a related old discussion on this.

traversaro, 2024-05-13:
> See also #151 for a related old discussion on this.

Actually that discussion refer to https://github.com/rapyuta-robotics/depthai_ros/, while this to discussion to https://github.com/luxonis/depthai-ros, not sure how the two are related.

kneave, 2024-05-13:
Hi Traversaro, I'm going to be completely honest and say I've no idea how the buildfarm handles these things. I started to look at setting one up so I could get packages built for the Raspberry Pi easier but then discovered robostack 😅

I'd be happy to learn if it could help but I'm just not sure where to start.

traversaro, 2024-05-13:
Cool, thanks! 

Just some pointer I found:
* `depthai-ros` depends on `depthai-core`, that is also packaged in the ROS build farm (see https://github.com/ros/rosdistro/pull/40111)
* `depthai-core` has a branch for ros releases (https://github.com/luxonis/depthai-core/tree/ros-release), that is different from the default branch of the repo (https://github.com/luxonis/depthai-core/tree/master).

Given that, I guess we could simply try to add `depthai-ros` as a package, and check if the CI is able to do built it. You can follow the instructions on how to prepare a PR to add `depthai-ros` in https://robostack.github.io/Contributing.html, then we will se in the PR CI if the package builds fine or not.

kneave, 2024-05-13:
Done :)
https://github.com/RoboStack/ros-noetic/pull/471

kneave, 2024-05-13:
Looks like Tobias may have beaten us to it a few weeks ago though!
https://github.com/RoboStack/ros-noetic/actions/runs/8902332643

traversaro, 2024-05-13:
> Looks like Tobias may have beaten us to it a few weeks ago though! https://github.com/RoboStack/ros-noetic/actions/runs/8902332643

Good point, let's cross link to avoid forgetting about it: https://github.com/RoboStack/ros-noetic/pull/470 .

traversaro, 2024-05-13:
Also https://github.com/RoboStack/ros-noetic/pull/441#issuecomment-1913420942 is related.