RoboStack/ros-noetic #472: conflict with mutex 0.5.* and mutex 0.4.* when trying to install packages

hsparkastro, 2024-05-19:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I did a clean install on a new env. Trying to install ros-noetic-jsk-visualization causes a conflict with mutex 0.4.* and mutex 0.5.*. The ros-noetic hector-gazebo-plugins causes a simular conflict.

```
mamba install --dry-run ros-noetic-jsk-visualization

conda-forge/linux-64                                        Using cache
conda-forge/noarch                                          Using cache

Looking for: ['ros-noetic-jsk-visualization']

Pinned packages:
  - python 3.11.*

Could not solve for environment specs
The following packages are incompatible

...
├─ ros-noetic-hector-gazebo-plugins is not installable because there are no viable options
│  ├─ ros-noetic-hector-gazebo-plugins 0.5.4 would require
│  │  ├─ gazebo with the potential options
 ...
│  │  ├─ ros-distro-mutex 0.5.* , which can be installed (as previously explained);
│  │  └─ ros-noetic-gazebo-ros but there are no viable options
│  │     ├─ ros-noetic-gazebo-ros 2.9.2 would require
│  │     │  ├─ ros-noetic-gazebo-dev but there are no viable options
│  │     │  │  ├─ ros-noetic-gazebo-dev 2.9.2 would require
│  │     │  │  │  └─ gazebo >=11.14.0,<12.0a0  but there are no viable options
│  │     │  │  │     ├─ gazebo 11.14.0, which cannot be installed (as previously explained);
│  │     │  │  │     └─ gazebo 11.14.0 conflicts with any installable versions previously reported;
│  │     │  │  └─ ros-noetic-gazebo-dev 2.9.2 would require
│  │     │  │     └─ ros-distro-mutex 0.4.* noetic, which conflicts with any installable versions previously reported;
│  │     │  └─ ros-noetic-gazebo-msgs with the potential options
│  │     │     ├─ ros-noetic-gazebo-msgs 2.9.2 would require
│  │     │     │  └─ ros-distro-mutex 0.5.* , which can be installed (as previously explained);
│  │     │     └─ ros-noetic-gazebo-msgs 2.9.2 would require
│  │     │        └─ ros-distro-mutex 0.4.* noetic, which conflicts with any installable versions previously reported;
│  │     └─ ros-noetic-gazebo-ros 2.9.2 would require
│  │        └─ ros-distro-mutex 0.4.* noetic, which conflicts with any installable versions previously reported;

``` 

### Installed packages

```shell
packages in environment at /home/hspark/miniforge3/envs/ros:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
_sysroot_linux-64_curr_repodata_hack 3                   h69a702a_14    conda-forge
adwaita-icon-theme        46.0                     unix_0    conda-forge
alsa-lib                  1.2.11               hd590300_1    conda-forge
aom                       3.9.0                hac33072_0    conda-forge
apr                       1.7.0                hd590300_6    conda-forge
argcomplete               3.3.0              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                hfb0e8fe_2    conda-forge
atk-1.0                   2.38.0               h04ea711_2    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
bcrypt                    4.1.3           py311h5ecf98a_0    conda-forge
binutils                  2.40                 h4852527_0    conda-forge
binutils_impl_linux-64    2.40                 ha885e6a_0    conda-forge
binutils_linux-64         2.40                 hdade7a5_3    conda-forge
boost                     1.82.0               h781c19f_6    conda-forge
boost-cpp                 1.82.0               h44aadfe_6    conda-forge
[... 552 lines cut ...]
         python version : 3.10.14.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=zen
                          __conda=24.3.0=0
                          __cuda=11.4=0
                          __glibc=2.31=0
                          __linux=5.15.0=0
                          __unix=0=0
       base environment : /home/hspark/miniforge3  (writable)
      conda av data dir : /home/hspark/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/linux-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/hspark/miniforge3/pkgs
                          /home/hspark/.conda/pkgs
       envs directories : /home/hspark/miniforge3/envs
                          /home/hspark/.conda/envs
               platform : linux-64
             user-agent : conda/24.3.0 requests/2.31.0 CPython/3.10.14 Linux/5.15.0-105-generic ubuntu/20.04.6 glibc/2.31 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.8
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2024-05-23:
Hi @hsparkastro - could you please try creating a new environment? This works without issues for me: `micromamba create -n testenv --dry-run ros-noetic-jsk-visualization ros-noetic-desktop-full ros-noetic-hector-gazebo-plugins -c robostack-staging`.

hsparkastro, 2024-05-23:
Hi @Tobias-Fischer - your solution installs without error, but uses python=3.9 instead of python=3.11. Your env with python=3.9 won't properly populate my robot in Gazebo.

traversaro, 2024-05-30:
@hsparkastro can you try if something changed after https://github.com/RoboStack/ros-noetic/issues/475 ? Thanks!

Tobias-Fischer, 2024-07-21:
`micromamba create -n testenv --dry-run ros-noetic-jsk-visualization ros-noetic-desktop-full ros-noetic-hector-gazebo-plugins python=3.11 -c robostack-staging` works without issue for me now, closing.

closed 2024-07-21 by Tobias-Fischer

edoardo-cioffi-1997, 2024-09-19:
> `micromamba create -n testenv --dry-run ros-noetic-jsk-visualization ros-noetic-desktop-full ros-noetic-hector-gazebo-plugins python=3.11 -c robostack-staging` works without issue for me now, closing.

I havee tried to follow this suggestion, but i get the error:

 py-opencv 4.2.0 would require
│  │  └─ libopencv 4.2.0 py36_5, which does not exist (perhaps a missing channel);

What can I do?

Tobias-Fischer, 2024-09-19:
We need #495 ..