RoboStack/ros-noetic #475: Can not install ros-noetic-desktop-full

SJoJoK, 2024-05-29:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

After following the instruction in https://robostack.github.io/GettingStarted.html, I cant use `mamba install ros-noetic-desktop-full` to install it

### Installed packages

```shell
❯ mamba list
# packages in environment at /home/me/Softwares/miniconda3/envs/ros_env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main    defaults
_openmp_mutex             5.1                       1_gnu    defaults
bzip2                     1.0.8                h5eee18b_6    defaults
ca-certificates           2024.3.11            h06a4308_0    defaults
ld_impl_linux-64          2.38                 h1181459_1    defaults
libffi                    3.4.4                h6a678d5_1    defaults
libgcc-ng                 11.2.0               h1234567_1    defaults
libgomp                   11.2.0               h1234567_1    defaults
libstdcxx-ng              11.2.0               h1234567_1    defaults
libuuid                   1.41.5               h5eee18b_0    defaults
ncurses                   6.4                  h6a678d5_0    defaults
openssl                   3.0.13               h7f8727e_2    defaults
pip                       24.0            py311h06a4308_0    defaults
python                    3.11.9               h955ad1f_0    defaults
readline                  8.2                  h5eee18b_0    defaults
setuptools                69.5.1          py311h06a4308_0    defaults
sqlite                    3.45.3               h5eee18b_0    defaults
tk                        8.6.14               h39e8969_0    defaults
tzdata                    2024a                h04d1e81_0    defaults
wheel                     0.43.0          py311h06a4308_0    defaults
xz                        5.4.6                h5eee18b_1    defaults
zlib                      1.2.13               h5eee18b_1    defaults
```

### Environment info

```shell
mamba version : 1.5.8
     active environment : ros_env
    active env location : /home/me/Softwares/miniconda3/envs/ros_env
            shell level : 2
       user config file : /home/me/.condarc
 populated config files : /home/me/.condarc
                          /home/me/Softwares/miniconda3/envs/ros_env/.condarc
          conda version : 24.5.0
    conda-build version : not installed
         python version : 3.12.2.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=skylake
                          __conda=24.5.0=0
                          __cuda=12.2=0
                          __glibc=2.35=0
                          __linux=6.2.0=0
                          __unix=0=0
       base environment : /home/me/Softwares/miniconda3  (writable)
      conda av data dir : /home/me/Softwares/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/linux-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://mirrors.zju.edu.cn/anaconda/cloud/conda-forge/linux-64
                          https://mirrors.zju.edu.cn/anaconda/cloud/conda-forge/noarch
          package cache : /home/me/Softwares/miniconda3/pkgs
                          /home/me/.conda/pkgs
       envs directories : /home/me/Softwares/miniconda3/envs
                          /home/me/.conda/envs
               platform : linux-64
             user-agent : conda/24.5.0 requests/2.31.0 CPython/3.12.2 Linux/6.2.0-26-generic ubuntu/22.04.2 glibc/2.35 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.8
                UID:GID : 1000:1000
             netrc file : /home/me/.netrc
           offline mode : False
```

SJoJoK, 2024-05-29:
Here's the out put of the `mamba install`

```
The following packages are incompatible
├─ python is installable with the potential options
│  ├─ python 3.11.9 would require
│  │  └─ libuuid >=1.41.5,<2.0a0 , which can be installed;
│  ├─ python [1.0.1|1.2|...|3.8.8], which can be installed;
│  ├─ python 3.11.0 would require
│  │  └─ openssl >=1.1.1q,<1.1.2a , which can be installed;
│  ├─ python [3.9.0|3.9.1|...|3.9.9], which can be installed;
│  ├─ python 3.11.0 would require
│  │  └─ openssl >=1.1.1s,<1.1.2a , which can be installed;
│  └─ python 3.12.0rc3 would require
│     └─ _python_rc, which does not exist (perhaps a missing channel);
└─ ros-noetic-desktop-full is not installable because there are no viable options
   ├─ ros-noetic-desktop-full 1.5.0 would require
   │  ├─ ros-distro-mutex 0.4.* , which can be installed;
   │  ├─ ros-noetic-perception but there are no viable options
   │  │  ├─ ros-noetic-perception 1.5.0 would require
   │  │  │  └─ ros-noetic-ros-base but there are no viable options
   │  │  │     ├─ ros-noetic-ros-base 1.5.0 would require
   │  │  │     │  ├─ ros-noetic-bond-core but there are no viable options
   │  │  │     │  │  ├─ ros-noetic-bond-core 1.8.6 would require
   │  │  │     │  │  │  └─ ros-noetic-bondcpp but there are no viable options
   │  │  │     │  │  │     ├─ ros-noetic-bondcpp 1.8.6 would require
   │  │  │     │  │  │     │  └─ libuuid >=2.32.1,<3.0a0 , which conflicts with any installable versions previously reported;
   │  │  │     │  │  │     └─ ros-noetic-bondcpp 1.8.6 would require
   │  │  │     │  │  │        └─ ros-distro-mutex 0.5.* , which requires
   │  │  │     │  │  │           └─ gazebo 11 , which can be installed;
   │  │  │     │  │  └─ ros-noetic-bond-core 1.8.6 would require
   │  │  │     │  │     └─ ros-distro-mutex 0.5.* , which cannot be installed (as previously explained);
   │  │  │     │  └─ ros-noetic-ros-core with the potential options
   │  │  │     │     ├─ ros-noetic-ros-core 1.5.0 would require
   │  │  │     │     │  └─ ros-noetic-roslisp with the potential options
   │  │  │     │     │     ├─ ros-noetic-roslisp 1.9.24 would require
   │  │  │     │     │     │  └─ ros-noetic-ros-environment with the potential options
   │  │  │     │     │     │     ├─ ros-noetic-ros-environment 1.3.2 would require
   │  │  │     │     │     │     │  └─ python_abi 3.9.* *_cp39 with the potential options
   │  │  │     │     │     │     │     ├─ python_abi 3.9 would require
   │  │  │     │     │     │     │     │  └─ python 3.9.* , which can be installed;
   │  │  │     │     │     │     │     └─ python_abi 3.9 would require
   │  │  │     │     │     │     │        └─ python 3.9.* *_cpython, which conflicts with any installable versions previously reported;
   │  │  │     │     │     │     └─ ros-noetic-ros-environment 1.3.2 would require
   │  │  │     │     │     │        └─ ros-distro-mutex 0.5.* , which cannot be installed (as previously explained);
   │  │  │     │     │     └─ ros-noetic-roslisp 1.9.25 would require
   │  │  │     │     │        └─ ros-distro-mutex 0.5.* , which cannot be installed (as previously explained);
   │  │  │     │     └─ ros-noetic-ros-core 1.5.0 would require
   │  │  │     │        └─ ros-distro-mutex 0.5.* , which cannot be installed (as previously explained);
   │  │  │     └─ ros-noetic-ros-base 1.5.0 would require
   │  │  │        └─ ros-distro-mutex 0.5.* , which cannot be installed (as previously explained);
   │  │  └─ ros-noetic-perception 1.5.0 would require
   │  │     └─ ros-distro-mutex 0.5.* , which cannot be installed (as previously explained);
   │  └─ ros-noetic-urdf-sim-tutorial but there are no viable options
   │     ├─ ros-noetic-urdf-sim-tutorial 0.5.1 would require
   │     │  └─ ros-noetic-rqt-robot-steering but there are no viable options
   │     │     ├─ ros-noetic-rqt-robot-steering 0.5.12 would require
   │     │     │  └─ ros-noetic-python-qt-binding but there are no viable options
   │     │     │     ├─ ros-noetic-python-qt-binding 0.4.4 would require
   │     │     │     │  └─ qt-main >=5.15.8,<5.16.0a0  but there are no viable options
   │     │     │     │     ├─ qt-main 5.15.8 would require
   │     │     │     │     │  └─ openssl >=3.1.0,<4.0a0 , which conflicts with any installable versions previously reported;
   │     │     │     │     ├─ qt-main 5.15.8 would require
   │     │     │     │     │  └─ openssl >=3.2.1,<4.0a0 , which conflicts with any installable versions previously reported;
   │     │     │     │     ├─ qt-main 5.15.8 would require
   │     │     │     │     │  └─ openssl >=3.2.0,<4.0a0 , which conflicts with any installable versions previously reported;
   │     │     │     │     ├─ qt-main 5.15.8 would require
   │     │     │     │     │  └─ openssl >=3.0.8,<4.0a0 , which conflicts with any installable versions previously reported;
   │     │     │     │     ├─ qt-main 5.15.8 would require
   │     │     │     │     │  └─ openssl >=3.1.1,<4.0a0 , which conflicts with any installable versions previously reported;
[... 1 lines cut ...]
   │     │     │     │     │  └─ openssl >=3.1.3,<4.0a0 , which conflicts with any installable versions previously reported;
   │     │     │     │     └─ qt-main 5.15.8 would require
   │     │     │     │        └─ openssl >=3.1.2,<4.0a0 , which conflicts with any installable versions previously reported;
   │     │     │     └─ ros-noetic-python-qt-binding 0.4.4 would require
[... cut ...]
         │     │     │  └─ gazebo >=11.14.0,<12.0a0 , which conflicts with any installable versions previously reported;
         │     │     └─ ros-noetic-gazebo-dev 2.9.2 would require
         │     │        └─ gazebo >=11.12.0,<12.0a0 , which conflicts with any installable versions previously reported;
         │     └─ ros-noetic-gazebo-ros-pkgs 2.9.2 would require
         │        └─ ros-distro-mutex 0.4.* , which can be installed;
         └─ ros-noetic-simulators 1.5.0 would require
            └─ ros-distro-mutex 0.4.* , which can be installed.
```

traversaro, 2024-05-29:
I was able to reproduce the problem:

~~~
traversaro@IITBMP014LW012:~$ conda create -n testros -c conda-forge -c robostack-staging python==3.11.* ros-noetic-desktop-full
Channels:
 - conda-forge
 - robostack-staging
Platform: linux-64
Collecting package metadata (repodata.json): done
Solving environment: failed

LibMambaUnsatisfiableError: Encountered problems while solving:
  - package ros-noetic-gazebo-ros-pkgs-2.9.2-py311hb335429_17 requires ros-noetic-gazebo-dev, but none of the providers can be installed

Could not solve for environment specs
The following packages are incompatible
├─ python 3.11**  is installable and it requires
│  └─ python_abi 3.11.* *_cp311, which can be installed;
└─ ros-noetic-desktop-full is not installable because there are no viable options
   ├─ ros-noetic-desktop-full 1.5.0 would require
   │  ├─ ros-distro-mutex 0.5.* , which requires
   │  │  └─ gazebo 11 , which can be installed;
   │  └─ ros-noetic-simulators but there are no viable options
   │     ├─ ros-noetic-simulators 1.5.0 would require
   │     │  └─ ros-noetic-gazebo-ros-pkgs but there are no viable options
   │     │     ├─ ros-noetic-gazebo-ros-pkgs 2.9.2 would require
   │     │     │  └─ ros-noetic-gazebo-dev but there are no viable options
   │     │     │     ├─ ros-noetic-gazebo-dev 2.9.2 would require
   │     │     │     │  └─ gazebo >=11.14.0,<12.0a0 , which conflicts with any installable versions previously reported;
   │     │     │     └─ ros-noetic-gazebo-dev 2.9.2 would require
   │     │     │        └─ python_abi 3.9.* *_cp39, which conflicts with any installable versions previously reported;
   │     │     └─ ros-noetic-gazebo-ros-pkgs 2.9.2 would require
   │     │        └─ python_abi 3.9.* *_cp39, which conflicts with any installable versions previously reported;
   │     └─ ros-noetic-simulators 1.5.0 would require
   │        └─ python_abi 3.9.* *_cp39, which conflicts with any installable versions previously reported;
   └─ ros-noetic-desktop-full 1.5.0 would require
      └─ python_abi 3.9.* *_cp39, which conflicts with any installable versions previously reported.
~~~

traversaro, 2024-05-29:
That is strange, as `desktop-full` is in the list of build packages: https://github.com/RoboStack/ros-noetic/blob/7f743314408fbaea5701b8f09193e11136b9b899/vinca_linux_64.yaml#L106 .

traversaro, 2024-05-29:
I think the problem is the constraint of gazebo in https://github.com/RoboStack/ros-noetic/blob/7f743314408fbaea5701b8f09193e11136b9b899/additional_recipes/ros-distro-mutex/recipe.yaml#L34 , that is 11 instead of 11.* . Somehow now (and perhaps in the past this was difference) the 11 constraint gets translated to 11.0.0, and that is not compatible with the `gazebo >=11.14.0,<12.0a0` constraint of the other packages.

SJoJoK, 2024-05-29:
> I think the problem is the constraint of gazebo in
>
> https://github.com/RoboStack/ros-noetic/blob/7f743314408fbaea5701b8f09193e11136b9b899/additional_recipes/ros-distro-mutex/recipe.yaml#L34
>
> , that [...]

Yeah, but it's kinda weird that there will have no issue when specify python=3.9

traversaro, 2024-05-29:
This is good point. Indeed, by looking in the `index.json` ros-distro-mutex 0.4.0 (the one compatible with Python 3.9) you can read:

~~~
{
  "arch": "x86_64",
  "build": "noetic",
  "build_number": 3,
  "constrains": [
    "boost-cpp 1.78.*",
    "pcl 1.12.*",
    "gazebo 11.*",
    "ogre 1.10.12*",
    "libpqxx 6.*"
  ],
  "depends": [],
  "name": "ros-distro-mutex",
  "platform": "linux",
  "subdir": "linux-64",
  "timestamp": 1674167258383,
  "version": "0.4.0"
}
~~~

while the 0.5.0 version is:
~~~
{
  "arch": "x86_64",
  "build": "noetic",
  "build_number": 6,
  "constrains": [
    "libboost-devel 1.82",
    "gazebo 11",
    "ogre 1.10.12",
    "libpqxx 7.8"
  ],
  "depends": [],
  "name": "ros-distro-mutex",
  "platform": "linux",
  "subdir": "linux-64",
  "timestamp": 1713739594677,
  "version": "0.5.0"
}
~~~

Somehow at some point something changed in boa or similar that started rendering the `gazebo 11` constraint from `gazebo 11.*` to `gazebo 11` ? The problematic `ros-distro-mutex` 0.5.0 package was uploaded ~1 month ago.

traversaro, 2024-05-29:
Interestingly, prefix.dev contains a cache info of a 0.5.0 with build number 4 (that I guess we deleted from `robostack-staging`) that contained the `gazebo 11.*` constraint instead of `gazebo 11` that we have in the build 6 that is currently in the channel, see https://prefix.dev/channels/robostack-staging/packages/ros-distro-mutex .

traversaro, 2024-05-29:
fyi @wolfv @Tobias-Fischer

closed 2024-05-29 by Tobias-Fischer