RoboStack/ros-noetic #478: Package request: slam_gmapping

GokulEdakkePuram, 2024-06-10:
### Package name

slam_gmapping

### Operating System(s)

- [ ] linux-64
- [ ] linux-aarch64
- [X] osx-64
- [X] osx-arm64
- [ ] win-64

### Additional context (optional)

Package for OpenSlam's Gmapping. It provides laser-based SLAM and can create a 2-D occupancy grid map from laser and pose data.

### Links and references (optional)

ROS Documentation: http://wiki.ros.org/gmapping
GitHub Repo: https://github.com/ros-perception/slam_gmapping

Tobias-Fischer, 2024-06-13:
Hi @GokulEdakkePuram  - if you create pull requests for the packages you are looking for, I’ll be able to merge - unfortunately I currently don’t have the capacity to action issues.

Best, Tobi

Tobias-Fischer, 2024-07-21:
Fixed in #487

closed 2024-07-21 by Tobias-Fischer