RoboStack/ros-noetic #484: error when ource devel/setup.bash

camillaBobo, 2024-06-14:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I have a MacBook, and I've set up RoboStack and ROS Noetic. 

When I tried to run "source devel/setup.bash" in the catkin_ws directory, it showed an error: "devel/setup.bash:.:8: no such file or directory: /Users/user/catkin_ws/setup.sh". 
I tried to save environment variables into ros-noetic-catkin_activate.sh, following the instructions provided in this issue: https://github.com/RoboStack/ros-noetic/issues/148. However, I encountered a similar error when I tried to activate the environment using mamba.

Here are the steps I took:
>> mamba activate ros_env
>>cd catkin_ws
>>source  devel/setup.bash

Thank you in advance! 

### Installed packages

```shell
# packages in environment at /Users/user/miniforge3/envs/ros_env:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        46.2                     unix_0    conda-forge
aom                       3.9.0                h537b5a7_0    conda-forge
apr                       1.7.0                h4614cfb_6    conda-forge
argcomplete               3.3.0              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                he63ff86_2    conda-forge
atk-1.0                   2.38.0               hd03087b_2    conda-forge
bcrypt                    4.1.3           py311h98c6a39_0    conda-forge
boost                     1.82.0               hf3b2ce4_6    conda-forge
boost-cpp                 1.82.0               hca5e981_6    conda-forge
brotli                    1.1.0                hb547adb_1    conda-forge
brotli-bin                1.1.0                hb547adb_1    conda-forge
bzip2                     1.0.8                h93a5062_5    conda-forge
c-ares                    1.28.1               h93a5062_0    conda-forge
c-compiler                1.7.0                h6aa9301_1    conda-forge
ca-certificates           2024.6.2             hf0a4a13_0    conda-forge
cairo                     1.18.0               hd1e100b_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
cctools                   986                  h4faf515_0    conda-forge
cctools_osx-arm64         986                  h62378fb_0    conda-forge
certifi                   2024.6.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py311h4a08483_0    conda-forge
clang                     16.0.6          default_h095aff0_8    conda-forge
clang-16                  16.0.6          default_hb63da90_8    conda-forge
clang_impl_osx-arm64      16.0.6              hc421ffc_16    conda-forge
clang_osx-arm64           16.0.6              h54d7cd3_16    conda-forge
clangxx                   16.0.6          default_h095aff0_8    conda-forge
clangxx_impl_osx-arm64    16.0.6              hcd7bac0_16    conda-forge
clangxx_osx-arm64         16.0.6              h54d7cd3_16    conda-forge
cmake                     3.29.4               h2cf84f3_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py311h267d04e_1    conda-forge
colcon-core               0.16.1             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.3.0              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.4.0              pyhd8ed1ab_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.4.0              pyhd8ed1ab_0    conda-forge
[... 475 lines cut ...]
          conda version : 24.5.0
    conda-build version : not installed
         python version : 3.10.14.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=m2
                          __conda=24.5.0=0
                          __osx=14.5=0
                          __unix=0=0
       base environment : /Users/user/miniforge3  (writable)
      conda av data dir : /Users/user/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/user/miniforge3/pkgs
                          /Users/user/.conda/pkgs
       envs directories : /Users/user/miniforge3/envs
                          /Users/user/.conda/envs
               platform : osx-arm64
             user-agent : conda/24.5.0 requests/2.31.0 CPython/3.10.14 Darwin/23.5.0 OSX/14.5 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.8
                UID:GID : 502:20
             netrc file : None
           offline mode : False
```

traversaro, 2024-06-14:
What repos are contained in the `catkin_ws` ?

camillaBobo, 2024-06-14:
Inside the catkin_ws directory, there are three folders: build, devel, and src. 
The "devel/setup.bash" file wants to find "$_CATKIN_SETUP_DIR/setup.sh", but for some reason, the _CATKIN_SETUP_DIR is wrongly defined. I am not super familiar with bash programming, but this is how the _CATKIN_SETUP_DIR variable is defined in the setup.bash file as follows:
_CATKIN_SETUP_DIR=$(builtin cd "`dirname "${BASH_SOURCE[0]}"`" > /dev/null && pwd)

Tobias-Fischer, 2024-06-14:
Can you try sourcing setup.zsh instead?

camillaBobo, 2024-06-14:
sourcing setup.zsh works! Thank you!

closed 2024-06-14 by camillaBobo