RoboStack/ros-noetic #486: Update moveit can not use the old package .

kaola-zero, 2024-07-10:
### Package name

libgeometric_shapes.so.0.7.6

### Operating System(s)

- [X] linux-64
- [X] linux-aarch64
- [ ] osx-64
- [ ] osx-arm64
- [ ] win-64

### Additional context (optional)

Errors     << rm_demo:make /home/rm/demo_ws/logs/rm_demo/build.make.002.log
/usr/bin/ld: warning: libgeometric_shapes.so.0.7.6, needed by /opt/ros/noetic/lib/libmoveit_planning_scene_monitor.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_state.so: undefined reference to `shapes::constructMarkerFromShape(shapes::Shape const*, visualization_msgs::Marker_<std::allocator<void> >&, bool)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_state.so: undefined reference to `shapes::constructShapeFromMsg(shape_msgs::Mesh_<std::allocator<void> > const&)'
/usr/bin/ld: /opt/ros/noetic/lib//usrlibmoveit_robot_model.so/:bin /undefinedld reference:  towarning :` shapeslibgeometric_shapes.so.0.7.6:,: createMeshFromResourceneeded( stdby: :/__cxx11opt:/:rosbasic_string/<noeticchar/,lib /stdlibmoveit_planning_scene_monitor.so:,: char_traitsnot< charfound> ,( trystd :using :-allocatorrpath<char >or  >- rpathconst-&link,)
Eigen/:usr:/Matrixbin/<doubleld, :3 ,/ opt1/,ros /0noetic,/ lib3/,libmoveit_robot_state.so :1 >undefined  constreference& )to'
`/shapesusr:/:binconstructMarkerFromShape/(ldshapes:: :/Shapeopt /constros*/,noetic /visualization_msgslib:/:libmoveit_planning_scene.so: undefined reference Marker_to< std::`allocator<shapes::constructShapeFromText(stdvoid::istream&)'
/usr/bin/ld: /opt/ros/noetic>/lib />libmoveit_robot_state.so&:,  undefinedbool )reference'
to/ usr`/shapesbin:/:constructMsgFromShape(shapes::Shape const*, boost::variant<shape_msgs::SolidPrimitive_<std::allocator<void> >, shape_msgs::Mesh_<std::allocator<void> >, shape_msgs::Plane_<std::allocator<void> > >&)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_model.so: undefined ldreference:  to/ `opttypeinfo/ rosfor/ noeticshapes/:lib:/Meshlibmoveit_robot_state.so':
 /undefinedusr /referencebin /told:  `/shapesopt:/:rosconstructShapeFromMsg/(noeticshape_msgs/:lib:/Mesh_libmoveit_kinematic_constraints.so<:std :undefined: allocatorreference< voidto>  `>bodies :const:&createEmptyBodyFromShapeType)('shapes
:/:usrShapeType/ binconst/&ld):'
//optusr/ros//noeticbin//libld/:libmoveit_robot_model.so :/ optundefined/ rosreference/ noeticto/ lib`/shapeslibmoveit_planning_scene.so::: createMeshFromResourceundefined( stdreference:: __cxx11to:: basic_string`<shapeschar:,: saveAsTextstd(:shapes::char_traits:<Shapechar >const,* ,std :std::allocator:<ostreamchar&>) '>
 /constusr&/,bin /Eigenld::: Matrix/<optdouble/,ros /3noetic,/ lib1/,libmoveit_robot_model.so: 0,  3undefined,  reference1 >to  const`&)'
shapes/:usr:/Spherebin:/:ldSphere:( double/)opt'/
ros//usrnoetic//binlib//ldlibmoveit_planning_scene.so::  /undefinedopt /referenceros /tonoetic /`libshapes/:libmoveit_kinematic_constraints.so::constructShapeFromText (undefinedstd :reference: istreamto& )`'shapes
:/:usrMesh/:bin:/Meshld(:) '/
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_model.so: undefined reference to `shapes::Box::Box(double, double, double)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_model.so: undefined reference to `typeinfo for shapes::Shape'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_model.so: undefined reference to `shapes::computeShapeExtents(shapes::Shape const*)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_state.so: undefined reference to `shapes::Shape::padd(double)'
/optusr/bin/ld: /opt/ros/noetic/lib/libmoveit_planning_scene.so: undefined reference to `vtable for shapes::OcTree'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_state.so: undefined reference to `shapes::Shape::scale(double)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_model.so: undefined reference to `shapes::Cylinder::Cylinder(double, double)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_planning_scene.so: undefined reference to `shapes::OcTree::OcTree(std::shared_ptr<octomap::OcTree const> const&)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_planning_scene.so: undefined reference to `shapes::Shape::~Shape()'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_state.so: undefined reference to `shapes::constructShapeFromMsg(shape_msgs::Plane_<std::allocator<void> > const&)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_state.so: undefined reference to `shapes::constructShapeFromMsg(/ros/noetic/lib/libmoveit_robot_state.so: undefined reference to `shapes::constructMsgFromShape(shapes::Shape const*, boost::variant<shape_msgs::SolidPrimitive_<std::allocator<void> >, shape_msgs::Mesh_<std::allocator<void> >, shape_msgs::Plane_<std::allocator<void> > >&)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_model.so: undefined reference to `typeinfo for shapes::Mesh'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_kinematic_constraints.so: undefined reference to `bodies::createEmptyBodyFromShapeType(shapes::ShapeType const&)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_planning_scene.so: undefined reference to `shapes::saveAsText(shapes::Shape const*, std::ostream&)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_model.so: undefined reference to `shapes::Sphere::Sphere(double)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_kinematic_constraints.so: undefined reference to `shapes::Mesh::Mesh()'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_model.so: undefined reference to `shapes::Box::Box(double, double, double)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_model.so: undefined reference to `typeinfo for shapes::Shape'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_model.so: undefined reference to `shapes::computeShapeExtents(shapes::Shape const*)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_state.so: undefined reference to `shapes::Shape::padd(double)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_planning_scene.so: undefined reference to `vtable for shapes::OcTree'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_state.so: undefined reference to `shapes::Shape::scale(double)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_model.so: undefined reference to `shapes::Cylinder::Cylinder(double, double)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_planning_scene.so: undefined reference to `shapes::OcTree::OcTree(std::shared_ptr<octomap::OcTree const> const&)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_planning_scene.so: undefined reference to `shapes::Shape::~Shape()'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_state.so: undefined reference to `shapes::constructShapeFromMsg(shape_msgs::Plane_<std::allocator<void> > const&)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_state.so: undefined reference to `shapes::constructShapeFromMsg(shape_msgs::SolidPrimitive_<stdshape_msgs::SolidPrimitive_<std::allocator<void> > const&)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_collision_detection.so: undefined reference to `typeinfo for shapes::OcTree'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `shapes::Mesh::mergeVertices(double)'
collect2: error: ld returned 1 exit status
::allocator<void> > const&)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_collision_detection.so: undefined reference to `typeinfo for shapesmake[2]: *** [CMakeFiles/rm_eco65_pick_place_demo.dir/build.make:323：/home/rm/demo_ws/devel/.private/rm_demo/lib/rm_demo/rm_eco65_pick_place_demo] 错误 1
::OcTree'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_planning_scene_monitor.so: undefined reference to `shapes::Mesh::mergeVertices(double)'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/Makefile2:329：CMakeFiles/rm_eco65_pick_place_demo.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....
/usr/bin/ld: warning: libgeometric_shapes.so.0.7.6, needed by /opt/ros/noetic/lib/libmoveit_planning_scene_monitor.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_state.so: undefined reference to `shapes::constructMarkerFromShape(shapes::Shape const*, visualization_msgs::Marker_<std::allocator<void> >&, bool)'
[... 99 lines cut ...]
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_planning_scene.so: undefined reference to `shapes::OcTree::OcTree(std::shared_ptr<octomap::OcTree const> const&)'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_planning_scene.so: undefined reference to `shapes::Shape::~Shape()'
/usr/bin/ld: /opt/ros/noetic/lib/libmoveit_robot_state.so: undefined reference to `shapes::constructShapeFromMsg(shape_msgs::Plane_<std::allocator<void> > const&)'
[... cut ...]
....................................................................................................................................
Failed     << rm_demo:make           [ Exited with code 2 ]
Failed    <<< rm_demo                [ 24.5 seconds ]
[build] Summary: 0 of 1 packages succeeded.
[build]   Ignored:   None.
[build]   Warnings:  None.
[build]   Abandoned: None.
[build]   Failed:    1 packages failed.
[build] Runtime: 24.6 seconds total.

### Links and references (optional)

Source link: https://github.com/kaola-zero/rm_robot/tree/main/rm_demo

kaola-zero, 2024-07-12:
我解决了这个问题，需要将moveit有关的包进行重新安装

closed 2024-07-12 by kaola-zero

traversaro, 2024-07-12:
> 我解决了这个问题，需要将moveit有关的包进行重新安装

Just a google provided translation for anyone not fluent in your language:

> I solved this problem by reinstalling the moveit related packages.