RoboStack/ros-noetic #489: Install RoboStack against Python 3.10

USeebi, 2024-07-25:
### Package name

Colab

### Operating System(s)

- [ ] linux-64
- [ ] linux-aarch64
- [ ] osx-64
- [ ] osx-arm64
- [ ] win-64

### Additional context (optional)

I would like to run RoboStack on Google colab. I tried to have a Python 3.11 venv as well as to update colab python version but both have obstacles beyond my comprehensions.

I wonder is it possible to install an older version of RoboStack against Python 3.10?

Tobias-Fischer, 2024-07-25:
Unfortunately this is not feasible due to maintenance time. In theory it would be easy to do, but each additional Python version doubles the build time (+dealing with compatibility issues for multiple versions). See e.g. https://github.com/RoboStack/ros-humble/issues/97

Furthermore, on colab you would need not just any Python but installed via conda in particular, which as far as I understand is not easy either.

USeebi, 2024-07-25:
Thanks very much for your prompt reply!

I understand the burdens to maintain a cross-platform packages.  How about just the bare minimum say only up and running for turtlesim for proof of concepts to have RoboStack on colab.  Conda is not a problem as long as the Python version of colab is kept.  https://datascience.stackexchange.com/questions/75948/how-to-setup-and-run-conda-on-google-colab

Anyway is there any procedures to built/install RoboStack from sources could be shared?

Many thanks!

Tobias-Fischer, 2024-07-25:
Yes, building locally should be relatively straightforward, see https://robostack.github.io/Contributing.html

USeebi, 2024-07-26:
Thanks Tobias,

I tried the Python 3.11 version on a local installation with juypterlab and I got ModuleNotFoundError: No module named 'notebook.utils' from the following:

[image]

This is the same error I got with colab that is out of my comprehension. What did I miss?

Traceback (most recent call last)

Cell In[3], line 1
----> 1 import jupyros
      2 import rospy
      3 from jupyros import TurtleSim

File [~/miniconda3/envs/ros_noetic/lib/python3.11/site-packages/jupyros/__init__.py:33](http://192.168.0.122:8888/home/ucsd/miniconda3/envs/ros_noetic/lib/python3.11/site-packages/jupyros/__init__.py#line=32)
     28     from .ros2.turtle_sim import *
     30 else:
     31     # Default to ROS1
     32     # print(f'ROS {ros_distro} environment detected.')
---> 33     from .ros1.ipy import *
     34     from .ros1.pubsub import *
     35     from .ros1.ros_widgets import *

File [~/miniconda3/envs/ros_noetic/lib/python3.11/site-packages/jupyros/ros1/__init__.py:15](http://192.168.0.122:8888/home/ucsd/miniconda3/envs/ros_noetic/lib/python3.11/site-packages/jupyros/ros1/__init__.py#line=14)
     13 from .ros_widgets import *
     14 from .ros3d import *
---> 15 from .server_extension import *
     16 from .turtle_sim import *

File [~/miniconda3/envs/ros_noetic/lib/python3.11/site-packages/jupyros/ros1/server_extension.py:10](http://192.168.0.122:8888/home/ucsd/miniconda3/envs/ros_noetic/lib/python3.11/site-packages/jupyros/ros1/server_extension.py#line=9)
      1 #############################################################################
      2 # Copyright (c) Wolf Vollprecht, QuantStack                                 #
      3 #                                                                           #
   (...)
      6 # The full license is in the file LICENSE, distributed with this software.  #
      7 #############################################################################
      8 import os
---> 10 from notebook.utils import url_path_join
     11 from notebook.base.handlers import IPythonHandler
     13 import rospkg

ModuleNotFoundError: No module named 'notebook.utils'

Tobias-Fischer, 2024-07-26:
That seems unrelated to RoboStack - my best guess is some version incompatibility with a too new version of “notebook”; there are several reports on this on StackOverflow. 

/cc @hbcarlos @wolfv

Tobias-Fischer, 2024-07-27:
See https://github.com/RoboStack/jupyter-ros/issues/157

closed 2024-08-02 by USeebi