RoboStack/ros-noetic #492: Cannot source a `setup.bash` file from a private workspace

yunfanjiang, 2024-07-31:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Say someone shares with me their private workspace and ROS packages by only sending the `install` folder and several `setup.*sh` files (presumably following this [instruction](https://wiki.ros.org/catkin/commands/catkin_make)). To use them, I need to do `source install/setup.bash`. This works fine on a Ubuntu 20.04 with ROS Noetic installed with `sudo apt install ros-noetic-desktop-full`. Unfortunately, this doesn't work with ROS Noetic installed through robostack on a Ubuntu 22.04 machine. Are there any extra steps I need to perform to make it work with robostack ROS Noetic? Thanks.

### Installed packages

```shell
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
_sysroot_linux-64_curr_repodata_hack 3                   h69a702a_16    conda-forge
adwaita-icon-theme        46.2                     unix_0    conda-forge
alsa-lib                  1.2.12               h4ab18f5_0    conda-forge
aom                       3.9.1                hac33072_0    conda-forge
apr                       1.7.0                hd590300_6    conda-forge
argcomplete               3.4.0              pyhd8ed1ab_0    conda-forge
assimp                    5.3.1                hfb0e8fe_2    conda-forge
atk-1.0                   2.38.0               h04ea711_2    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
bcrypt                    4.2.0           py311hb3a8bbb_0    conda-forge
binutils                  2.40                 h4852527_7    conda-forge
binutils_impl_linux-64    2.40                 ha1999f0_7    conda-forge
binutils_linux-64         2.40                 hb3c18ed_0    conda-forge
boost                     1.82.0               h781c19f_6    conda-forge
boost-cpp                 1.82.0               h44aadfe_6    conda-forge
brotli                    1.1.0                hd590300_1    conda-forge
brotli-bin                1.1.0                hd590300_1    conda-forge
bzip2                     1.0.8                h4bc722e_7    conda-forge
c-ares                    1.32.3               h4bc722e_0    conda-forge
c-compiler                1.7.0                hd590300_1    conda-forge
ca-certificates           2024.7.4             hbcca054_0    conda-forge
cairo                     1.18.0               h3faef2a_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
certifi                   2024.7.4           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py311hb3a22ac_0    conda-forge
cmake                     3.30.1               hf8c4bd3_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py311h38be061_1    conda-forge
colcon-core               0.17.1             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.3.0              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.4.0              pyhd8ed1ab_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.4.0              pyhd8ed1ab_0    conda-forge
colcon-python-setup-py    0.2.7                      py_0    conda-forge
colcon-recursive-crawl    0.2.3              pyhd8ed1ab_0    conda-forge
colcon-ros                0.5.0              pyhd8ed1ab_0    conda-forge
colcon-test-result        0.3.8                      py_0    conda-forge
colcon-zsh                0.5.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
coloredlogs               15.0.1             pyhd8ed1ab_3    conda-forge
compilers                 1.7.0                ha770c72_1    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
contourpy                 1.2.1           py311h9547e67_0    conda-forge
coreutils                 9.5                  hd590300_0    conda-forge
coverage                  7.6.0           py311h61187de_0    conda-forge
cryptography              43.0.0          py311hc6616f6_0    conda-forge
[... 516 lines cut ...]
         python version : 3.10.14.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=zen3
                          __conda=24.7.1=0
                          __cuda=12.4=0
                          __glibc=2.35=0
                          __linux=6.5.0=0
                          __unix=0=0
       base environment : /home/yunfanj/miniforge3  (writable)
      conda av data dir : /home/yunfanj/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/linux-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/yunfanj/miniforge3/pkgs
                          /home/yunfanj/.conda/pkgs
       envs directories : /home/yunfanj/miniforge3/envs
                          /home/yunfanj/.conda/envs
               platform : linux-64
             user-agent : conda/24.7.1 requests/2.31.0 CPython/3.10.14 Linux/6.5.0-45-generic ubuntu/22.04.4 glibc/2.35 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.8
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2024-08-01:
Hi @yunfanjiang - could you please create a mininum working example demonstrating your problem, and also describe what exactly doesn't work?

yunfanjiang, 2024-08-01:
Thanks for the reply @Tobias-Fischer. Turns out it's the issue about something else. Closing this now.

closed 2024-08-01 by yunfanjiang