RoboStack/ros-noetic #496: unable to install repository dependencies for tiago_public_ws

FrancescoD98, 2024-10-10:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Hi, first of all, my congratulations for the work done. 

I've installed the environment to work on TIAGo robot which works with ROS1 and I'm trying to install the dependencies of the public_ws following their [official guide](http://wiki.ros.org/Robots/TIAGo/Tutorials/Installation/InstallUbuntuAndROS). To do so, inside the mamba environment, I've changed the source commands in the tutorial page with `source ~/miniconda3/envs/ros_env` and followed the tutorial until it calls the `rosdep install [...]` command to install the dependencies that gives the following error
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
pal_navigation_cfg_omni_base: No definition of [teb_local_planner] for OS [conda]
pal_gazebo_plugins: No definition of [gazebo] for OS [conda]
dynamic_introspection: No definition of [eigen] for OS [conda]
four_wheel_steering_controller: No definition of [urdf_geometry_parser] for OS [conda]
eband_local_planner: No definition of [tf2_eigen] for OS [conda]
pal_statistics: No definition of [boost] for OS [conda]
pal_navigation_cfg_tiago_dual: No definition of [octomap_server] for OS [conda]
pal_robotiq_controller_configuration_gazebo: No definition of [roboticsgroup_upatras_gazebo_plugins] for OS [conda]
pmb2_rgbd_sensors: No definition of [libuvc_camera] for OS [conda]
roboticsgroup_gazebo_plugins: No definition of [control_toolbox] for OS [conda]
simple_models_description: No definition of [realsense2_description] for OS [conda]
joint_limits_interface: No definition of [liburdfdom-dev] for OS [conda]
controller_manager: No definition of [realtime_tools] for OS [conda]
forward_command_controller: No definition of [realtime_tools] for OS [conda]
gazebo_dev: No definition of [libgazebo11-dev] for OS [conda]
pal_navigation_cfg_pmb2: No definition of [teb_local_planner] for OS [conda]
tiago_moveit_tutorial: No definition of [moveit_ros_planning_interface] for OS [conda]
demo_motions: No definition of [yaml-cpp] for OS [conda]
temperature_sensor_controller: No definition of [realtime_tools] for OS [conda]
pmb2_laser_sensors: No definition of [laser_filters] for OS [conda]
pal_navigation_cfg_pmb3: No definition of [teb_local_planner] for OS [conda]
rviz_plugin_covariance: No definition of [libogre-dev] for OS [conda]
range_sensor_layer: No definition of [costmap_2d] for OS [conda]
imu_sensor_controller: No definition of [realtime_tools] for OS [conda]
joint_trajectory_controller: No definition of [realtime_tools] for OS [conda]
tiago_rgbd_sensors: No definition of [openni2_launch] for OS [conda]
pal_navigation_cfg_ari: No definition of [teb_local_planner] for OS [conda]
pmb2_bringup: No definition of [joy] for OS [conda]
tiago_bringup: No definition of [usb_cam] for OS [conda]
tiago_laser_sensors: No definition of [ira_laser_tools] for OS [conda]
omni_base_bringup: No definition of [twist_mux] for OS [conda]
diff_drive_controller: No definition of [realtime_tools] for OS [conda]
pal_hardware_gazebo: No definition of [eigen] for OS [conda]
tiago_opencv_tutorial: No definition of [image_transport_plugins] for OS [conda]
pal_navigation_cfg_tiago: No definition of [octomap_server] for OS [conda]
gripper_action_controller: No definition of [realtime_tools] for OS [conda]
gazebo_ros: No definition of [tinyxml] for OS [conda]
omni_base_description: No definition of [hector_gazebo_plugins] for OS [conda]
joint_torque_sensor_state_controller: No definition of [realtime_tools] for OS [conda]
ros_control: No definition of [realtime_tools] for OS [conda]
gazebo_ros_control: No definition of [control_toolbox] for OS [conda]
tiago_description: No definition of [hector_gazebo_plugins] for OS [conda]
tf_lookup: No definition of [boost] for OS [conda]
mode_state_controller: No definition of [realtime_tools] for OS [conda]
force_torque_sensor_controller: No definition of [realtime_tools] for OS [conda]
pal_navigation_sm: No definition of [map_server] for OS [conda]
omni_base_laser_sensors: No definition of [ira_laser_tools] for OS [conda]
aruco: No definition of [eigen] for OS [conda]
effort_controllers: No definition of [realtime_tools] for OS [conda]
simple_models_gazebo: No definition of [boost] for OS [conda]
tiago_pick_demo: No definition of [moveit_commander] for OS [conda]
actuator_state_controller: No definition of [realtime_tools] for OS [conda]
velocity_controllers: No definition of [realtime_tools] for OS [conda]
play_motion: No definition of [moveit_ros_planning_interface] for OS [conda]
tiago_moveit_config: No definition of [moveit_simple_controller_manager] for OS [conda]
gazebo_plugins: No definition of [camera_info_manager] for OS [conda]
backward_ros: No definition of [libdw-dev] for OS [conda]
transmission_interface: No definition of [tinyxml] for OS [conda]
[... 588 lines cut ...]
    conda-build version : not installed
         python version : 3.12.6.final.0
                 solver : classic
       virtual packages : __archspec=1=skylake
                          __conda=24.7.1=0
                          __glibc=2.39=0
                          __linux=6.8.0=0
                          __unix=0=0
       base environment : /home/labrob/miniforge3  (writable)
      conda av data dir : /home/labrob/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/linux-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/labrob/miniforge3/pkgs
                          /home/labrob/.conda/pkgs
       envs directories : /home/labrob/miniforge3/envs
                          /home/labrob/.conda/envs
               platform : linux-64
             user-agent : conda/24.7.1 requests/2.32.3 CPython/3.12.6 Linux/6.8.0-41-generic ubuntu/24.04.1 glibc/2.39
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

traversaro, 2024-10-10:
Which options are you passing after `rosdep install` ?

FrancescoD98, 2024-10-11:
> Which options are you passing after `rosdep install` ?
I was following the [official PAL guide](http://wiki.ros.org/Robots/TIAGo/Tutorials/Installation/InstallUbuntuAndROS) to install the robot ws dependencies. All the commands have been taken from their guide. The `rosdep install` command is the following
```
rosdep install -y --from-paths src --ignore-src --rosdistro noetic --skip-keys "urdf_test omni_drive_controller orocos_kdl pal_filters libgazebo9-dev pal_usb_utils speed_limit_node camera_calibration_files pal_moveit_plugins pal_startup_msgs pal_local_joint_control pal_pcl_points_throttle_and_filter current_limit_controller hokuyo_node dynamixel_cpp pal_moveit_capabilities pal_pcl dynamic_footprint gravity_compensation_controller pal-orbbec-openni2 pal_loc_measure pal_map_manager ydlidar_ros_driver"
```

traversaro, 2024-10-11:
By executing this commands:

~~~
conda create -n tiagoenv -c conda-forge -c robostack-staging ros-noetic-desktop rosinstall
conda activate tiagoenv
mdkir tiago_public_ws
cd ~/tiago_public_ws
wget https://raw.githubusercontent.com/pal-robotics/tiago_tutorials/noetic-devel/tiago_public-noetic.rosinstall
rosinstall src tiago_public-noetic.rosinstall
rosdep init
rosdep update
rosdep install -y --from-paths src --ignore-src --rosdistro noetic --skip-keys "urdf_test omni_drive_controller orocos_kdl pal_filters libgazebo9-dev pal_usb_utils speed_limit_node camera_calibration_files pal_moveit_plugins pal_startup_msgs pal_local_joint_control pal_pcl_points_throttle_and_filter current_limit_controller hokuyo_node dynamixel_cpp pal_moveit_capabilities pal_pcl dynamic_footprint gravity_compensation_controller pal-orbbec-openni2 pal_loc_measure pal_map_manager ydlidar_ros_driver"
~~~

The failure is more reasonable:

~~~
backward_ros: No definition of [libdw-dev] for OS [conda]
~~~

I am not sure why you have that error instead. Can you start over from a clean environment and report exactly all the commands you executed (directly, not indirectly by saying "I followed http://wiki.ros.org/Robots/TIAGo/Tutorials/Installation/InstallUbuntuAndROS) so it may be easier to understand what is going on?

traversaro, 2024-10-11:
The missing `libdw-dev` key will be fixed by https://github.com/RoboStack/ros-noetic/pull/497 .

FrancescoD98, 2024-10-11:
at the beginning, and with a fresh installation, I followed more or less your same commands and I came up with your same error. I tried to overcome it by running the command
```
rosdep install -y --from-paths src --ignore-src --rosdistro noetic --skip-keys "urdf_test omni_drive_controller orocos_kdl pal_filters libgazebo9-dev pal_usb_utils speed_limit_node camera_calibration_files pal_moveit_plugins pal_startup_msgs pal_local_joint_control pal_pcl_points_throttle_and_filter current_limit_controller hokuyo_node dynamixel_cpp pal_moveit_capabilities pal_pcl dynamic_footprint gravity_compensation_controller pal-orbbec-openni2 pal_loc_measure pal_map_manager ydlidar_ros_driver"
```
from inside a docker container, in which ubuntu 20 was running, with `home` directory shared between system/conda and the container. At that point I tried to use again the command above inside the conda environment, to see if everything was effectively installed, and the error changed in what you see at the beginning of this issue.

In the end my final goal is to run Gazebo with TIAGo on newest ubuntu versions without passing through docker, since there are packages provided with conda that i would like to use.

FrancescoD98, 2024-10-11:
Sorry about that. After a fresh installation of miniforge I exectuted the following commands
```
mamba create -n ros_env python=3.11 -y
conda config --env --add channels robostack-staging
conda config --env --remove channels defaults
mamba install ros-noetic-desktop -y
mamba install compilers cmake pkg-config make ninja colcon-common-extensions catkin_tools rosdep rosinstall -y
mamba update --all

mkdir ~/tiago_public_ws
cd ~/tiago_public_ws
wget https://raw.githubusercontent.com/pal-robotics/tiago_tutorials/noetic-devel/tiago_public-noetic.rosinstall
rosinstall src ~/miniforge3/envs/ros_env tiago_public-noetic.rosinstall
rosdep init
rosdep update

rosdep install -y --from-paths src --ignore-src --rosdistro noetic --skip-keys "urdf_test omni_drive_controller orocos_kdl pal_filters libgazebo9-dev pal_usb_utils speed_limit_node camera_calibration_files pal_moveit_plugins pal_startup_msgs pal_local_joint_control pal_pcl_points_throttle_and_filter current_limit_controller hokuyo_node dynamixel_cpp pal_moveit_capabilities pal_pcl dynamic_footprint gravity_compensation_controller pal-orbbec-openni2 pal_loc_measure pal_map_manager ydlidar_ros_driver"
```
and the last command reports the following error
```
rosdep install -y --from-paths src --ignore-src --rosdistro noetic --skip-keys "urdf_test omni_drive_controller orocos_kdl pal_filters libgazebo9-dev pal_usb_utils speed_limit_node camera_calibration_files pal_moveit_plugins pal_startup_msgs pal_local_joint_control pal_pcl_points_throttle_and_filter current_limit_controller hokuyo_node dynamixel_cpp pal_moveit_capabilities pal_pcl dynamic_footprint gravity_compensation_controller pal-orbbec-openni2 pal_loc_measure pal_map_manager ydlidar_ros_driver"
executing command [mamba install -c robostack-staging -c conda-forge -y ros-noetic-costmap-2d ros-noetic-realtime-tools ros-noetic-pcl-ros ros-noetic-pcl-conversions ros-noetic-polled-camera ros-noetic-camera-info-manager ros-noetic-map-server ros-noetic-amcl ros-noetic-fake-localization ros-noetic-move-base ros-noetic-spatio-temporal-voxel-layer ros-noetic-global-planner ros-noetic-teb-local-planner ros-noetic-octomap-server ros-noetic-moveit-ros-move-group ros-noetic-moveit-planners-ompl ros-noetic-pilz-industrial-motion-planner ros-noetic-moveit-planners-chomp ros-noetic-moveit-chomp-optimizer-adapter ros-noetic-moveit-ros-visualization ros-noetic-moveit-simple-controller-manager ros-noetic-people-msgs ros-noetic-control-toolbox ros-noetic-sick-tim ros-noetic-depthimage-to-laserscan ros-noetic-pointcloud-to-laserscan ros-noetic-laser-filters ros-noetic-joy ros-noetic-openni2-launch ros-noetic-libuvc-camera ros-noetic-usb-cam gazebo ros-noetic-moveit-ros-planning-interface ros-noetic-hector-gazebo-plugins ros-noetic-twist-mux ros-noetic-roboticsgroup-upatras-gazebo-plugins ros-noetic-realsense2-description ros-noetic-image-transport-plugins elfutils ros-noetic-four-wheel-steering-msgs ros-noetic-urdf-geometry-parser ros-noetic-ira-laser-tools ros-noetic-base-local-planner ros-noetic-nav-core ros-noetic-tf2-eigen ros-noetic-slam-toolbox ros-noetic-ddynamic-reconfigure ros-noetic-moveit-msgs ros-noetic-moveit-commander]

Looking for: ['ros-noetic-costmap-2d', 'ros-noetic-realtime-tools', 'ros-noetic-pcl-ros', 'ros-noetic-pcl-conversions', 'ros-noetic-polled-camera', 'ros-noetic-camera-info-manager', 'ros-noetic-map-server', 'ros-noetic-amcl', 'ros-noetic-fake-localization', 'ros-noetic-move-base', 'ros-noetic-spatio-temporal-voxel-layer', 'ros-noetic-global-planner', 'ros-noetic-teb-local-planner', 'ros-noetic-octomap-server', 'ros-noetic-moveit-ros-move-group', 'ros-noetic-moveit-planners-ompl', 'ros-noetic-pilz-industrial-motion-planner', 'ros-noetic-moveit-planners-chomp', 'ros-noetic-moveit-chomp-optimizer-adapter', 'ros-noetic-moveit-ros-visualization', 'ros-noetic-moveit-simple-controller-manager', 'ros-noetic-people-msgs', 'ros-noetic-control-toolbox', 'ros-noetic-sick-tim', 'ros-noetic-depthimage-to-laserscan', 'ros-noetic-pointcloud-to-laserscan', 'ros-noetic-laser-filters', 'ros-noetic-joy', 'ros-noetic-openni2-launch', 'ros-noetic-libuvc-camera', 'ros-noetic-usb-cam', 'gazebo', 'ros-noetic-moveit-ros-planning-interface', 'ros-noetic-hector-gazebo-plugins', 'ros-noetic-twist-mux', 'ros-noetic-roboticsgroup-upatras-gazebo-plugins', 'ros-noetic-realsense2-description', 'ros-noetic-image-transport-plugins', 'elfutils', 'ros-noetic-four-wheel-steering-msgs', 'ros-noetic-urdf-geometry-parser', 'ros-noetic-ira-laser-tools', 'ros-noetic-base-local-planner', 'ros-noetic-nav-core', 'ros-noetic-tf2-eigen', 'ros-noetic-slam-toolbox', 'ros-noetic-ddynamic-reconfigure', 'ros-noetic-moveit-msgs', 'ros-noetic-moveit-commander']

conda-forge/linux-64                                        Using cache
conda-forge/noarch                                          Using cache
robostack-staging/linux-64                                    No change
robostack-staging/noarch                                      No change

Pinned packages:
  - python 3.11.*

warning  libmamba Added empty dependency for problem type SOLVER_RULE_UPDATE
Could not solve for environment specs
The following packages are incompatible
├─ __cuda is requested and can be installed;
├─ harfbuzz is installable with the potential options
│  ├─ harfbuzz [8.4.0|8.5.0|9.0.0] would require
│  │  └─ libglib [>=2.80.0,<3.0a0 |>=2.80.2,<3.0a0 ] with the potential options
│  │     ├─ libglib [2.80.0|2.80.1|2.80.2] would require
[... cut ...] oetic-moveit-ros-planning-interface ros-noetic-hector-gazebo-plugins ros-noetic-twist-mux ros-noetic-roboticsgroup-upatras-gazebo-plugins ros-noetic-realsense2-description ros-noetic-image-transport-plugins elfutils ros-noetic-four-wheel-steering-msgs ros-noetic-urdf-geometry-parser ros-noetic-ira-laser-tools ros-noetic-base-local-planner ros-noetic-nav-core ros-noetic-tf2-eigen ros-noetic-slam-toolbox ros-noetic-ddynamic-reconfigure ros-noetic-moveit-msgs ros-noetic-moveit-commander] failed
```

Tobias-Fischer, 2025-01-10:
Should be fixed with the rebuild - feel free to reopen if not.

Note there are some packages not supplied by RoboStack currently, feel free to open a PR if you're interested @FrancescoD98
```
error    libmamba Could not solve for environment specs
    The following packages are incompatible
    ├─ ros-noetic-four-wheel-steering-msgs =* * does not exist (perhaps a typo or a missing channel);
    ├─ ros-noetic-ira-laser-tools =* * does not exist (perhaps a typo or a missing channel);
    ├─ ros-noetic-libuvc-camera =* * does not exist (perhaps a typo or a missing channel);
    ├─ ros-noetic-moveit-chomp-optimizer-adapter =* * does not exist (perhaps a typo or a missing channel);
    ├─ ros-noetic-roboticsgroup-upatras-gazebo-plugins =* * does not exist (perhaps a typo or a missing channel);
    ├─ ros-noetic-sick-tim =* * does not exist (perhaps a typo or a missing channel);
    ├─ ros-noetic-spatio-temporal-voxel-layer =* * does not exist (perhaps a typo or a missing channel);
    └─ ros-noetic-urdf-geometry-parser =* * does not exist (perhaps a typo or a missing channel).
```

closed 2025-01-10 by Tobias-Fischer