RoboStack/ros-noetic #503: Could not import "pyqt" bindings of qt_gui_cpp library

stevalkr, 2025-02-07:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

Same as https://github.com/RoboStack/ros-noetic/issues/237

### Installed packages

```shell
# packages in environment at /opt/homebrew/Caskroom/miniconda/base/envs/ros:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        47.0                     unix_0    conda-forge
aiohappyeyeballs          2.4.4              pyhd8ed1ab_1    conda-forge
aiohttp                   3.11.12          py39hefdd603_0    conda-forge
aiosignal                 1.3.2              pyhd8ed1ab_0    conda-forge
aom                       3.5.0                h7ea286d_0    conda-forge
apr                       1.7.0                h4614cfb_6    conda-forge
argcomplete               3.5.2              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                h276577b_0    conda-forge
async-timeout             5.0.1              pyhd8ed1ab_1    conda-forge
atk-1.0                   2.38.0               hd03087b_2    conda-forge
attrs                     25.1.0             pyh71513ae_0    conda-forge
autoconf                  2.71            pl5321hcd07c0c_1    conda-forge
automake                  1.17            pl5321hce30654_0    conda-forge
bcrypt                    4.2.1            py39hc40b5db_0    conda-forge
blosc                     1.21.5               hc338f07_0    conda-forge
boost                     1.78.0           py39h819cc4c_5    conda-forge
boost-cpp                 1.78.0               hf1d6563_2    conda-forge
brotli                    1.0.9                h1a8c8d9_9    conda-forge
brotli-bin                1.0.9                h1a8c8d9_9    conda-forge
brunsli                   0.1                  h9f76cd9_0    conda-forge
bzip2                     1.0.8                h99b78c6_7    conda-forge
c-ares                    1.34.4               h5505292_0    conda-forge
c-blosc2                  2.12.0               ha57e6be_0    conda-forge
c-compiler                1.7.0                h6aa9301_1    conda-forge
ca-certificates           2025.1.31            hf0a4a13_0    conda-forge
cairo                     1.16.0            h73a0509_1014    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_1    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_1    conda-forge
cctools                   973.0.1             h4faf515_13    conda-forge
cctools_osx-arm64         973.0.1             h7a0f565_13    conda-forge
certifi                   2024.12.14         pyhd8ed1ab_0    conda-forge
cffi                      1.17.1           py39h7f933ea_0    conda-forge
cfitsio                   4.2.0                h2f961c4_0    conda-forge
charls                    2.3.4                hbdafb3b_0    conda-forge
clang                     16.0.1               h1e5164c_0    conda-forge
clang-16                  16.0.1          default_he7141eb_0    conda-forge
clang_osx-arm64           16.0.1               hdccda30_1    conda-forge
clangxx                   16.0.1          default_ha0f6327_0    conda-forge
clangxx_osx-arm64         16.0.1               h0c1d50e_1    conda-forge
cmake                     3.26.3               hf234bd0_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_1    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_1    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_1    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0            py39h2804cbe_2    conda-forge
colcon-core               0.18.2             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.3.0              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.4.0              pyhd8ed1ab_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.3.0              pyhd8ed1ab_1    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.4.0              pyhd8ed1ab_0    conda-forge
[... 517 lines cut ...]
          conda version : 24.5.0
    conda-build version : not installed
         python version : 3.11.8.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=m1
                          __conda=24.5.0=0
                          __osx=15.2=0
                          __unix=0=0
       base environment : /opt/homebrew/Caskroom/miniconda/base  (writable)
      conda av data dir : /opt/homebrew/Caskroom/miniconda/base/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /opt/homebrew/Caskroom/miniconda/base/pkgs
                          /Users/walker/.conda/pkgs
       envs directories : /opt/homebrew/Caskroom/miniconda/base/envs
                          /Users/walker/.conda/envs
               platform : osx-arm64
             user-agent : conda/24.5.0 requests/2.31.0 CPython/3.11.8 Darwin/24.2.0 OSX/15.2 solver/libmamba conda-libmamba-solver/24.1.0 libmambapy/1.5.6
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2025-02-07:
You are using very old packages - can you please follow our instructions and install the latest packages?

Tobias-Fischer, 2025-02-07:
Using Python 3.11

closed 2025-02-08 by stevalkr