RoboStack/ros-noetic #504: RViz and RQt in ros-noetic don't work as expected on Ubuntu24.04

Ericsii, 2025-02-18:
RViz and RQt in ros-noetic don't work as expected.

- The buttons' icons for closing the window are not rendered.
- The sub-window cannot be dragged out of the main window.

[image]

[image]

https://github.com/user-attachments/assets/edce1cfc-9e9b-4d98-b46f-89d761259563

My pixi.toml

```toml
[project]
authors = ["XXX <XXX@XX.com>"]
channels = ["robostack-staging", "conda-forge"]
description = "RATE environment"
name = "rate_ws"
platforms = ["linux-64"]
version = "0.1.0"

[tasks]
build = { cmd = "catkin build --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_EXPORT_COMPILE_COMMANDS=ON"}
clean = { cmd = "rm -rf build devel logs" }

[dependencies]
compilers = ">=1.9.0,<2"
cmake = ">=3.31.5,<4"
pkg-config = ">=0.29.2,<0.30"
make = ">=4.4.1,<5"
ninja = ">=1.12.1,<2"
colcon-common-extensions = ">=0.3.0,<0.4"
catkin_tools = ">=0.9.4,<0.10"
rosdep = ">=0.25.1,<0.26"
ros-noetic-desktop = ">=1.5.0,<2"
libcaer = ">=3.3.15,<4"
ceres-solver = "1.14.0.*"
ros-noetic-camera-info-manager = ">=1.12.0,<2"
ros-noetic-cv-bridge = ">=1.16.2,<2"
```

Tobias-Fischer, 2025-02-20:
Can you try to `pixi intall qt6-gtk-platformtheme`?

Ericsii, 2025-02-22:
> Can you try to `pixi intall qt6-gtk-platformtheme`?

After I tried this. The close button icon in rviz is rendering but the window still cannot drag out and has no effect on rqt