RoboStack/ros-noetic #511: conda ros still works with the normal ros c++ node ?

linzha0, 2025-03-04:
Hi, I am working on Ros with PyTorch in Ubuntu 20.04. I just found this package when I was trying to find the solution. Thanks for your help.

If I am running a ros python node in the conda env (a ros node predicted some variable using specific version of pytroch and pre-trained model), is this node still talking to the normal ros c++ node?

Tobias-Fischer, 2025-03-04:
Yes that's correct

closed 2025-03-04 by Tobias-Fischer

linzha0, 2025-03-20:
I think I may not ask the question right, If I have robotstack-ros-noetic installed in my conda env, the node running inside can talk to the node running at regular ros environment?

Tobias-Fischer, 2025-03-20:
Yes, this is correct, they can communicate with each other.