RoboStack/ros-noetic #512: rqt_bag doesn't work.

JiahangWu, 2025-03-06:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

Hi,

I try to use '''rqt_bag''', but I have a problem.

```
PluginManager._load_plugin() could not load plugin "rqt_bag/Bag":
Traceback (most recent call last):
  File "/Users/ryan/.local/share/mamba/envs/ros_env/lib/python3.11/site-packages/qt_gui/plugin_handler.py", line 102, in load
    self._load()
  File "/Users/ryan/.local/share/mamba/envs/ros_env/lib/python3.11/site-packages/qt_gui/plugin_handler_direct.py", line 55, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/ryan/.local/share/mamba/envs/ros_env/lib/python3.11/site-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/ryan/.local/share/mamba/envs/ros_env/lib/python3.11/site-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/ryan/.local/share/mamba/envs/ros_env/lib/python3.11/site-packages/rqt_gui_py/ros_py_plugin_provider.py", line 61, in load
    return super(RosPyPluginProvider, self).load(plugin_id, plugin_context)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/ryan/.local/share/mamba/envs/ros_env/lib/python3.11/site-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/ryan/.local/share/mamba/envs/ros_env/lib/python3.11/site-packages/rqt_gui/ros_plugin_provider.py", line 106, in load
    return class_ref(plugin_context)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/ryan/.local/share/mamba/envs/ros_env/lib/python3.11/site-packages/rqt_bag/bag.py", line 57, in __init__
    self._widget = BagWidget(context, args.clock)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/ryan/.local/share/mamba/envs/ros_env/lib/python3.11/site-packages/rqt_bag/bag_widget.py", line 77, in __init__
    self._timeline = BagTimeline(context, publish_clock)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/ryan/.local/share/mamba/envs/ros_env/lib/python3.11/site-packages/rqt_bag/bag_timeline.py", line 111, in __init__
    self._timeline_frame = TimelineFrame(self)
                           ^^^^^^^^^^^^^^^^^^^
  File "/Users/ryan/.local/share/mamba/envs/ros_env/lib/python3.11/site-packages/rqt_bag/timeline_frame.py", line 129, in __init__
    self._topic_font.setPointSize(self._topic_font_size)
TypeError: setPointSize(self, a0: int): argument 1 has unexpected type 'float'
```

### Installed packages

```shell
DEPRECATION: --no-python-version-warning is deprecated. pip 25.1 will enforce this behaviour change. A possible replacement is to remove the flag as it's a no-op. Discussion can be found at https://github.com/pypa/pip/issues/13154
List of packages in environment: "/Users/ryan/.local/share/mamba/envs/ros_env"

  Name                                       Version       Build                      Channel
─────────────────────────────────────────────────────────────────────────────────────────────────────────
  adwaita-icon-theme                         47.0          unix_0                     conda-forge
  aom                                        3.9.1         h7bae524_0                 conda-forge
  apr                                        1.7.0         h4614cfb_6                 conda-forge
  argcomplete                                3.5.2         pyhd8ed1ab_0               conda-forge
  assimp                                     5.4.3         ha9c0b8d_0                 conda-forge
  atk-1.0                                    2.38.0        hd03087b_2                 conda-forge
  autoconf                                   2.71          pl5321hcd07c0c_1           conda-forge
  bcrypt                                     4.2.1         py311h3ff9189_0            conda-forge
  brotli                                     1.1.0         hd74edd7_2                 conda-forge
  brotli-bin                                 1.1.0         hd74edd7_2                 conda-forge
  bzip2                                      1.0.8         h80987f9_6
  c-ares                                     1.34.4        h5505292_0                 conda-forge
  c-compiler                                 1.9.0         hdf49b6b_0                 conda-forge
  ca-certificates                            2024.12.31    hca03da5_0
  cairo                                      1.18.2        h6a3b0d2_1                 conda-forge
[... 501 lines cut ...]
### Environment info

```shell
libmamba version : 2.0.5
          mamba version : 2.0.5
           curl version : libcurl/8.11.1 OpenSSL/3.4.1 (SecureTransport) zlib/1.3.1 zstd/1.5.6 libssh2/1.11.1 nghttp2/1.64.0
     libarchive version : libarchive 3.7.7 zlib/1.3.1 liblzma/5.6.3 bz2lib/1.0.8 liblz4/1.10.0 libzstd/1.5.6
       envs directories : /Users/ryan/.local/share/mamba/envs
          package cache : /Users/ryan/.local/share/mamba/pkgs
                          /Users/ryan/.mamba/pkgs
            environment : ros_env (active)
           env location : /Users/ryan/.local/share/mamba/envs/ros_env
      user config files : /Users/ryan/.mambarc
 populated config files : /Users/ryan/.local/share/mamba/envs/ros_env/.condarc
                          /Users/ryan/.condarc
       virtual packages : __unix=0=0
                          __osx=15.3.1=0
                          __archspec=1=arm64
               channels : https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
       base environment : /Users/ryan/.local/share/mamba
               platform : osx-arm64
```

Tobias-Fischer, 2025-03-06:
Actually, see this PR for required changes:
https://github.com/ros-visualization/rqt_bag/pull/131

JiahangWu, 2025-03-06:
> In /Users/ryan/.local/share/mamba/envs/ros_env/lib/python3.11/site-packages/rqt_bag/timeline_frame.py try replacing
> 
> ```
> self._topic_font.setPointSize(self._topic_font_size)
> ```
> 
> with
> 
> ```
> self._topic_font.setPointSizeF(self._topic_font_size)
> ```

I modified it in the source code as you mentioned.

[image]

Then, I reactivated Env and tried again, but the error showed that I didn't change it successfully. 
```
  File "/Users/ryan/.local/share/mamba/envs/ros_env/lib/python3.11/site-packages/rqt_bag/timeline_frame.py", line 140, in __init__
    self._time_font.setPointSize(self._time_font_size)
TypeError: setPointSize(self, a0: int): argument 1 has unexpected type 'float'
```
What should I do after I modify it?

Tobias-Fischer, 2025-03-06:
There are multiple changes required in this file. Please refer to the pull request I mentioned above for a full list of changes.

Tobias-Fischer, 2025-03-06:
Can you try to `micromamba install pyside2`?

JiahangWu, 2025-03-07:
Thank you. The error is fixed. But there are still no icons.

JiahangWu, 2025-05-25:
Yes. Sry replying late.

I have no idea how to add this, could you please give me more details?

traversaro, 2025-05-25:
Hi @JiahangWu, I do not know the details of the command you are launching, but I guess it is a Python executable, the suggestion of @Tobias-Fischer is to add the snippet linked to some of the modules it loads, or directly to the Python executable code, and check if this fixes the problem.