RoboStack/ros-noetic #515: tinyxml2 10.1.0 ABI break follow up

traversaro, 2025-03-11:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

The main problem for users (reported in https://github.com/RoboStack/robostack.github.io/issues/80) that install `ros-*-desktop` was fixed by https://github.com/conda-forge/conda-forge-repodata-patches-feedstock/pull/972 . However, we do not have repodata patches in robostack channels, so as soon as the conda-forge dependencies of ROS will be rebuilt with tinyxml2 10.1.0, we will have back the problem. Furthermore, the problem is still there for packages that do not depend on any conda-forge dependency that depend on tinyxml2, see for example:

~~~
traversaro@IITBMP014LW012:~$ micromamba create -n testros -c conda-forge -c robostack-noetic ros-noetic-pluginlib-tutori
als
conda-forge/linux-64                                        Using cache
conda-forge/noarch                                          Using cache

Transaction

  Prefix: /home/traversaro/miniforge3/envs/testros

  Updating specs:

   - ros-noetic-pluginlib-tutorials

  Package                                 Version  Build                  Channel                Size
───────────────────────────────────────────────────────────────────────────────────────────────────────
  Install:
───────────────────────────────────────────────────────────────────────────────────────────────────────

  + _libgcc_mutex                             0.1  conda_forge            conda-forge          Cached
  + _openmp_mutex                             4.5  2_gnu                  conda-forge          Cached
  + apr                                     1.7.0  hd590300_6             conda-forge          Cached
  + bzip2                                   1.0.8  h4bc722e_7             conda-forge          Cached
  + c-ares                                 1.34.4  hb9d3cd8_0             conda-forge          Cached
  + ca-certificates                     2025.1.31  hbcca054_0             conda-forge          Cached
  + catkin_pkg                              1.0.0  pyhd8ed1ab_1           conda-forge          Cached
  + cmake                                  3.31.6  h74e3db0_0             conda-forge          Cached
  + console_bridge                          1.0.2  h924138e_1             conda-forge          Cached
  + distro                                  1.9.0  pyhd8ed1ab_1           conda-forge          Cached
  + docutils                               0.21.2  pyhd8ed1ab_1           conda-forge          Cached
  + empy                                    3.3.4  pyh9f0ad1d_1           conda-forge          Cached
  + expat                                   2.6.4  h5888daf_0             conda-forge          Cached
  + fmt                                    11.1.4  h07f6e7f_0             conda-forge          Cached
  + gmock                                  1.15.2  ha770c72_0             conda-forge          Cached
  + gtest                                  1.15.2  h434a139_0             conda-forge          Cached
  + icu                                      75.1  he02047a_0             conda-forge          Cached
  + keyutils                                1.6.1  h166bdaf_0             conda-forge          Cached
  + krb5                                   1.21.3  h659f571_0             conda-forge          Cached
  + ld_impl_linux-64                         2.43  h712a8e2_4             conda-forge          Cached
  + libapr                                  1.7.0  hd590300_6             conda-forge          Cached
  + libapriconv                             1.2.2  hd590300_6             conda-forge          Cached
  + libaprutil                              1.6.1  h40f5838_6             conda-forge          Cached
  + libblas                                 3.9.0  31_h59b9bed_openblas   conda-forge          Cached
  + libboost                               1.86.0  h6c02f8c_3             conda-forge          Cached
  + libboost-devel                         1.86.0  h1a2810e_3             conda-forge          Cached
  + libboost-headers                       1.86.0  ha770c72_3             conda-forge          Cached
  + libcblas                                3.9.0  31_he106b2a_openblas   conda-forge          Cached
  + libcurl                                8.12.1  h332b0f4_0             conda-forge          Cached
  + libedit                          3.1.20250104  pl5321h7949ede_0       conda-forge          Cached
  + libev                                    4.33  hd590300_2             conda-forge          Cached
  + libexpat                                2.6.4  h5888daf_0             conda-forge          Cached
  + libffi                                  3.4.6  h2dba641_0             conda-forge          Cached
  + libgcc                                 14.2.0  h767d61c_2             conda-forge          Cached
  + libgcc-ng                              14.2.0  h69a702a_2             conda-forge          Cached
  + libgfortran                            14.2.0  h69a702a_2             conda-forge          Cached
  + libgfortran5                           14.2.0  hf1ad2bd_2             conda-forge          Cached
  + libgomp                                14.2.0  h767d61c_2             conda-forge          Cached
  + libiconv                                 1.18  h4ce23a2_1             conda-forge          Cached
  + liblapack                               3.9.0  31_h7ac8fdf_openblas   conda-forge          Cached
  + liblzma                                 5.6.4  hb9d3cd8_0             conda-forge          Cached
  + libnghttp2                             1.64.0  h161d5f1_0             conda-forge          Cached
  + libnsl                                  2.0.1  hd590300_0             conda-forge          Cached
[... 60 lines cut ...]
  + yaml                                    0.2.5  h7f98852_2             conda-forge          Cached
  + zstd                                    1.5.7  hb8e6e7a_1             conda-forge          Cached

  Summary:

  Install: 105 packages

  Total download: 131kB

───────────────────────────────────────────────────────────────────────────────────────────────────────

Confirm changes: [Y/n]
~~~

### Installed packages

```shell
.
```

### Environment info

```shell
.
```

traversaro, 2025-03-11:
https://github.com/RoboStack/ros-noetic/pull/514 seems to have solved the problem:

~~~
traversaro@IITBMP014LW012:~$ micromamba create -n testros -c conda-forge -c robostack-noetic ros-noetic-pluginlib-tutorials
robostack-noetic/noarch                             ??.?MB @  ??.?MB/s  0.6s
robostack-noetic/linux-64                           84.2kB @  91.6kB/s  0.8s
conda-forge/noarch                                  19.4MB @ 527.5kB/s 37.1s
conda-forge/linux-64                                42.5MB @ 705.8kB/s 1m:..

Transaction

  Prefix: /home/traversaro/miniforge3/envs/testros

  Updating specs:

   - ros-noetic-pluginlib-tutorials

  Package                                 Version  Build                  Channel                Size
───────────────────────────────────────────────────────────────────────────────────────────────────────
  Install:
───────────────────────────────────────────────────────────────────────────────────────────────────────

  + _libgcc_mutex                             0.1  conda_forge            conda-forge          Cached
  + _openmp_mutex                             4.5  2_gnu                  conda-forge          Cached
  + apr                                     1.7.0  hd590300_6             conda-forge          Cached
  + bzip2                                   1.0.8  h4bc722e_7             conda-forge          Cached
  + c-ares                                 1.34.4  hb9d3cd8_0             conda-forge          Cached
  + ca-certificates                     2025.1.31  hbcca054_0             conda-forge          Cached
  + catkin_pkg                              1.0.0  pyhd8ed1ab_1           conda-forge          Cached
  + cmake                                  3.31.6  h74e3db0_0             conda-forge          Cached
  + console_bridge                          1.0.2  h924138e_1             conda-forge          Cached
  + distro                                  1.9.0  pyhd8ed1ab_1           conda-forge          Cached
  + docutils                               0.21.2  pyhd8ed1ab_1           conda-forge          Cached
  + empy                                    3.3.4  pyh9f0ad1d_1           conda-forge          Cached
  + expat                                   2.6.4  h5888daf_0             conda-forge          Cached
  + fmt                                    11.1.4  h07f6e7f_0             conda-forge          Cached
  + gmock                                  1.15.2  ha770c72_0             conda-forge          Cached
  + gtest                                  1.15.2  h434a139_0             conda-forge          Cached
  + icu                                      75.1  he02047a_0             conda-forge          Cached
  + keyutils                                1.6.1  h166bdaf_0             conda-forge          Cached
  + krb5                                   1.21.3  h659f571_0             conda-forge          Cached
  + ld_impl_linux-64                         2.43  h712a8e2_4             conda-forge          Cached
  + libapr                                  1.7.0  hd590300_6             conda-forge          Cached
  + libapriconv                             1.2.2  hd590300_6             conda-forge          Cached
  + libaprutil                              1.6.1  h40f5838_6             conda-forge          Cached
  + libblas                                 3.9.0  31_h59b9bed_openblas   conda-forge          Cached
  + libboost                               1.86.0  h6c02f8c_3             conda-forge          Cached
  + libboost-devel                         1.86.0  h1a2810e_3             conda-forge          Cached
  + libboost-headers                       1.86.0  ha770c72_3             conda-forge          Cached
  + libcblas                                3.9.0  31_he106b2a_openblas   conda-forge          Cached
  + libcurl                                8.12.1  h332b0f4_0             conda-forge          Cached
  + libedit                          3.1.20250104  pl5321h7949ede_0       conda-forge          Cached
  + libev                                    4.33  hd590300_2             conda-forge          Cached
  + libexpat                                2.6.4  h5888daf_0             conda-forge          Cached
  + libffi                                  3.4.6  h2dba641_0             conda-forge          Cached
  + libgcc                                 14.2.0  h767d61c_2             conda-forge          Cached
  + libgcc-ng                              14.2.0  h69a702a_2             conda-forge          Cached
  + libgfortran                            14.2.0  h69a702a_2             conda-forge          Cached
  + libgfortran5                           14.2.0  hf1ad2bd_2             conda-forge          Cached
  + libgomp                                14.2.0  h767d61c_2             conda-forge          Cached
  + libiconv                                 1.18  h4ce23a2_1             conda-forge          Cached
  + liblapack                               3.9.0  31_h7ac8fdf_openblas   conda-forge          Cached
  + liblzma                                 5.6.4  hb9d3cd8_0             conda-forge          Cached
  + libnghttp2                             1.64.0  h161d5f1_0             conda-forge          Cached
  + libnsl                                  2.0.1  hd590300_0             conda-forge          Cached
  + libopenblas                            0.3.29  pthreads_h94d23a6_0    conda-forge          Cached
[... cut ...]
I also tested with pixi and conda, and it seems to be working fine. Clearly this is not a perfect solution, as for example `micromamba create -n testros -c conda-forge -c robostack-noetic tinyxml2 ros-noetic-pluginlib-tutorials` still installs the broken combination of ros packages and tinyxml2 . However, unless we remove packages (that would break existing pixi lockfiles) or we have some kind of repodata patches or way of marking packages broken (that as far as I know it is not possible on non-conda-forge packages) that is the best solution I can think while we wait for the following rebuild.

Tobias-Fischer, 2025-04-08:
Can we close here @traversaro?

traversaro, 2025-04-08:
Yes, this was definitely fixed by https://github.com/conda-forge/admin-requests/pull/1477, and by the ongoing on the conda-forge migration to tinyxml2 to 11 .

closed 2025-04-08 by traversaro