RoboStack/ros-noetic #516: Rviz crash

entatic, 2025-03-11:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

Rviz crashes immediately after start. In Console.app it shows:

 "CODE SIGNING: cs_invalid_page(0x120048000): p=86856[rviz] final status 0x23000200, denying page sending SIGKILL".

### Installed packages

```shell
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        47.0                     unix_0    conda-forge
aom                       3.9.1                h7bae524_0    conda-forge
apr                       1.7.0                h4614cfb_6    conda-forge
argcomplete               3.6.0              pyhd8ed1ab_0    conda-forge
assimp                    5.4.3                ha9c0b8d_0    conda-forge
atk-1.0                   2.38.0               hd03087b_2    conda-forge
bcrypt                    4.3.0           py311h3ff9189_0    conda-forge
brotli                    1.1.0                hd74edd7_2    conda-forge
brotli-bin                1.1.0                hd74edd7_2    conda-forge
bzip2                     1.0.8                h99b78c6_7    conda-forge
c-ares                    1.34.4               h5505292_0    conda-forge
c-compiler                1.9.0                hdf49b6b_0    conda-forge
ca-certificates           2025.1.31            hf0a4a13_0    conda-forge
cairo                     1.18.4               h6a3b0d2_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_1    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_1    conda-forge
cctools                   1010.6               hb4fb6a3_3    conda-forge
cctools_osx-arm64         1010.6               h3b4f5d3_3    conda-forge
cffi                      1.17.1          py311h3a79f62_0    conda-forge
clang                     18.1.8          default_h474c9e2_7    conda-forge
clang-18                  18.1.8          default_hf90f093_7    conda-forge
clang_impl_osx-arm64      18.1.8              h2ae9ea5_23    conda-forge
clang_osx-arm64           18.1.8              h07b0088_23    conda-forge
clangxx                   18.1.8          default_h1ffe849_7    conda-forge
clangxx_impl_osx-arm64    18.1.8              h555f467_23    conda-forge
clangxx_osx-arm64         18.1.8              h07b0088_23    conda-forge
cmake                     3.31.6               ha25475f_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_1    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_1    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_1    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py311h267d04e_2    conda-forge
colcon-core               0.18.2             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.3.0              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.4.0              pyhd8ed1ab_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.3.0              pyhd8ed1ab_1    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.4.0              pyhd8ed1ab_0    conda-forge
colcon-python-setup-py    0.2.9              pyhff2d567_0    conda-forge
colcon-recursive-crawl    0.2.3              pyhd8ed1ab_0    conda-forge
colcon-ros                0.5.0              pyhd8ed1ab_0    conda-forge
colcon-test-result        0.3.8                      py_0    conda-forge
colcon-zsh                0.5.0              pyhd8ed1ab_1    conda-forge
colorama                  0.4.6              pyhd8ed1ab_1    conda-forge
coloredlogs               15.0.1             pyhd8ed1ab_4    conda-forge
compiler-rt               18.1.8               h856b3c1_1    conda-forge
compiler-rt_osx-arm64     18.1.8               h832e737_1    conda-forge
compilers                 1.9.0                hce30654_0    conda-forge
console_bridge            1.0.2                h3e96240_1    conda-forge
contourpy                 1.3.1           py311h210dab8_0    conda-forge
[... 460 lines cut ...]
          conda version : 24.11.3
    conda-build version : not installed
         python version : 3.12.9.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=m1
                          __conda=24.11.3=0
                          __osx=14.6.1=0
                          __unix=0=0
       base environment : /Users/mansoor/miniforge3  (writable)
      conda av data dir : /Users/mansoor/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-noetic/osx-arm64
                          https://conda.anaconda.org/robostack-noetic/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/mansoor/miniforge3/pkgs
                          /Users/mansoor/.conda/pkgs
       envs directories : /Users/mansoor/miniforge3/envs
                          /Users/mansoor/.conda/envs
               platform : osx-arm64
             user-agent : conda/24.11.3 requests/2.32.3 CPython/3.12.9 Darwin/23.6.0 OSX/14.6.1 solver/libmamba conda-libmamba-solver/24.9.0 libmambapy/1.5.12
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

JiahangWu, 2025-03-21:
I also met this problem. Is there any solution?

JiahangWu, 2025-03-21:
> dmesg | grep "CODE SIGNING

Is that right?
```
╰─$ rviz dmesg | grep "CODE SIGNING"                                                              137 ↵
[1]    47209 killed     rviz dmesg |
       47210 exit 1     grep --color=auto --exclude-dir={.bzr,CVS,.git,.hg,.svn,.idea,.tox,.venv,venv
```

traversaro, 2025-03-21:
Sorry, I was not clear. First reproduce the crash, and then (after the crash happened) run "dmesg | grep "CODE SIGNING"

JiahangWu, 2025-03-21:
```
$ sudo dmesg | grep "CODE SIGNING"                                                                1 ↵
[689969.014054]: Library (/Users/ryan/.local/share/mamba/envs/esvo/lib/librviz_default_plugin.dylib) allowed in Process (/Users/ryan/.local/share/mamba/envs/esvo/bin/rviz) via dev tool environment (/Applications/iTerm.app/Contents/MacOS/iTerm2)Library (/Users/ryan/.local/share/mamba/envs/esvo/lib/librviz_default_plugin.dylib) allowed in Process (/Users/ryan/.local/share/mamba/envs/esvo/bin/rviz) via dev tool environment (/Applications/iTerm.app/Contents/MacOS/iTerm2)Library (/Users/ryan/.local/share/mamba/envs/esvo/lib/libinteractive_markers.dylib) allowed in Process (/Users/ryan/.local/share/mamba/envs/esvo/bin/rviz) via dev tool environment (/Applications/iTerm.app/Contents/MacOS/iTerm2)Library (/Users/ryan/.local/share/mamba/envs/esvo/lib/libinteractive_markers.dylib) allowed in Process (/Users/ryan/.local/share/mamba/envs/esvo/bin/rviz) via dev tool environment (/Applications/iTerm.app/Contents/MacOS/iTerm2)Library (/Users/ryan/.local/share/mamba/envs/esvo/lib/liblaser_geometry.dylib) allowed in Process (/Users/ryan/.local/share/mamba/envs/esvo/bin/rviz) via dev tool environment (/Applications/iTerm.app/Contents/MacOS/iTerm2)Library (/Users/ryan/.local/share/mamba/envs/esvo/lib/liblaser_geometry.dylib) allowed in Process (/Users/ryan/.local/share/mamba/envs/esvo/bin/rviz) via dev tool environment (/Applications/iTerm.app/Contents/MacOS/iTerm2)Library (/Users/ryan/.local/share/mamba/envs/esvo/lib/liburdf.dylib) allowed in Process (/Users/ryan/.local/share/mamba/envs/esvo/bin/rviz) via dev tool environment (/Applications/iTerm.app/Contents/MacOS/iTerm2)Library (/Users/ryan/.local/share/mamba/envs/esvo/lib/liburdf.dylib) allowed in Process (/Users/ryan/.local/share/mamba/envs/esvo/bin/rviz) via dev tool environment (/Applications/iTerm.app/Contents/MacOS/iTerm2)Library (/Users/ryan/.local/share/mamba/envs/esvo/lib/libtf.dylib) allowed in Process (/Users/ryan/.local/share/mamba/envs/esvo/bin/rviz) via dev tool environment (/Applications/iTerm.app/Contents/MacOS/iTerm2)Library (/Users/ryan/.local/share/mamba/envs/esvo/lib/libtf.dylib) allowed in Process (/Users/ryan/.local/share/mamba/envs/esvo/bin/rviz) via dev tool environment (/Applications/iTerm.app/Contents/MacOS/iTerm2)Library (/Users/ryan/.local/share/mamba/envs/esvo/lib/libtinyxml.dylib) allowed in Process (/Users/ryan/.local/share/mamba/envs/esvo/bin/rviz) via dev tool environment (/Applications/iTerm.app/Contents/MacOS/iTerm2)Library (/Users/ryan/.local/share/mamba/envs/esvo/lib/libtinyxml.dylib) allowed in Process (/Users/ryan/.local/share/mamba/envs/esvo/bin/rviz) via dev tool environment (/Applications/iTerm.app/Contents/MacOS/iTerm2)Library (/Users/ryan/.local/share/mamba/envs/esvo/lib/librosconsole_bridge.dylib) allowed in Process (/Users/ryan/.local/share/mamba/envs/esvo/bin/rviz) via dev tool environment (/Applications/iTerm.app/Contents/MacOS/iTerm2)Library (/Users/ryan/.local/share/mamba/envs/esvo/lib/librosconsole_bridge.dylib) allowed in Process (/Users/ryan/.local/share/mamba/envs/esvo/bin/rviz) via dev tool environment (/Applications/iTerm.app/Contents/MacOS/iTerm2)CODE SIGNING: cs_invalid_page(0x114f64000): p=54459[rviz] final status 0x23000200, denying page sending SIGKILL
[689969.016572]: CODE SIGNING: process 54459[rviz]: rejecting invalid page at address 0x114f64000 from offset 0x0 in file "/Users/ryan/.local/share/mamba/envs/esvo/lib/liburdfdom_model.4.0.dylib" (cs_mtime:1741340741.0 == mtime:1741340741.0) (signed:1 validated:1 tainted:1 nx:0 wpmapped:0 dirty:0 depth:0)

```

traversaro, 2025-03-21:
Thanks! Then the issue was created by https://github.com/RoboStack/ros-noetic/pull/514 . I will properly fix this tomorrow, in the meanwhile installing rviz=1.14.25=*_9 should work as a workaround.

JiahangWu, 2025-03-21:
Aha, Thank you! Btw, what do you mean rviz=1.14.25=*_9?

traversaro, 2025-03-22:
Just run mamba install ros-noetic-rviz=1.14.25=*_9 (or the variant for any package manager you may use).

traversaro, 2025-03-22:
@JiahangWu @entatic thanks a lot for reporting the issue. The latest rviz package available in `osx-arm64/ros-noetic-rviz-1.14.25-np126py311h4203010_23.conda`  (see https://anaconda.org/robostack-noetic/ros-noetic-rviz/files) should have fixed the issue, can you try to do a fresh install (i.e. without lock file) and check if the issue is solved? Thanks!

JiahangWu, 2025-03-22:
It works! Thank you!

traversaro, 2025-03-22:
Great! I will close, if anyone still has the problem feel free to open a new issue.

closed 2025-03-22 by traversaro