RoboStack/ros-noetic #518: catkin build cannot find yaml-cpp

JiahangWu, 2025-03-19:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

Hi,

When I use ```catkin build```  my project with ```yaml-cpp```, it always cannot find ```yaml-cpp```. But I am sure I have installed ```yaml-cpp``` in my mamba environment. 

If I don't use ```catkin build```, but only use ```cmake```, the project can be compiled successfully.

My project is a simple call ```YAML::Node rightCamCalibInfo = YAML::LoadFile("xxxx.yaml");```
The error messages are:
```
╰─$ catkin build beginner_tutorials                                                                                                              1 ↵
------------------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /Users/ryan/.local/share/mamba/envs/esvo
Workspace:                   /Users/ryan/Code/test_ros/catkin_ws
------------------------------------------------------------------------
Build Space:        [exists] /Users/ryan/Code/test_ros/catkin_ws/build
Devel Space:        [exists] /Users/ryan/Code/test_ros/catkin_ws/devel
Install Space:      [unused] /Users/ryan/Code/test_ros/catkin_ws/install
Log Space:          [exists] /Users/ryan/Code/test_ros/catkin_ws/logs
Source Space:       [exists] /Users/ryan/Code/test_ros/catkin_ws/src
DESTDIR:            [unused] None
------------------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
------------------------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
------------------------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------------------------
[build] Found 1 packages in 0.0 seconds.
[build] Package table is up to date.
Starting  >>> beginner_tutorials
____________________________________________________________________________________________________________________________________________________
Warnings   << beginner_tutorials:check /Users/ryan/Code/test_ros/catkin_ws/logs/beginner_tutorials/build.check.000.log
CMake Warning (dev) at /Users/ryan/.local/share/mamba/envs/esvo/share/catkin/cmake/python.cmake:4 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  /Users/ryan/.local/share/mamba/envs/esvo/share/catkin/cmake/all.cmake:164 (include)
  /Users/ryan/.local/share/mamba/envs/esvo/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:17 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

 yaml-cpp NOT FOUND!
cd /Users/ryan/Code/test_ros/catkin_ws/build/beginner_tutorials; catkin build --get-env beginner_tutorials | catkin env -si  /Users/ryan/.local/share/mamba/envs/esvo/bin/make cmake_check_build_system; cd -

....................................................................................................................................................
____________________________________________________________________________________________________________________________________________________
Errors     << beginner_tutorials:make /Users/ryan/Code/test_ros/catkin_ws/logs/beginner_tutorials/build.make.001.log
Undefined symbols for architecture arm64:
  "YAML::LoadFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&)", referenced from:
      _main in test.cpp.o
ld: symbol(s) not found for architecture arm64
arm64-apple-darwin20.0: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [CMakeFiles/test_yaml.dir/build.make:171: test_yaml] Error 1
[... 586 lines cut ...]
          mamba version : 2.0.7
           curl version : libcurl/8.12.1 OpenSSL/3.4.1 (SecureTransport) zlib/1.3.1 zstd/1.5.6 libssh2/1.11.1 nghttp2/1.64.0
     libarchive version : libarchive 3.7.7 zlib/1.3.1 liblzma/5.6.3 bz2lib/1.0.8 liblz4/1.10.0 libzstd/1.5.6
       envs directories : /Users/ryan/.local/share/mamba/envs
          package cache : /Users/ryan/.local/share/mamba/pkgs
                          /Users/ryan/.mamba/pkgs
            environment : esvo (active)
           env location : /Users/ryan/.local/share/mamba/envs/esvo
      user config files : /Users/ryan/.mambarc
 populated config files : /Users/ryan/.local/share/mamba/envs/esvo/.condarc
                          /Users/ryan/.condarc
       virtual packages : __unix=0=0
                          __osx=15.3.2=0
                          __archspec=1=arm64
               channels : https://conda.anaconda.org/robostack-noetic/noarch
                          https://conda.anaconda.org/robostack-noetic/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/main/osx-arm64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/r/osx-arm64
       base environment : /Users/ryan/.local/share/mamba
               platform : osx-arm64
```

JiahangWu, 2025-03-19:
Here is a simple example to reproduce this issue.

```test.cpp```

```
#include <opencv2/opencv.hpp>
#include <iostream>
#include <yaml-cpp/yaml.h>

int main(int argc, char** argv) {
    
	std::string imagePath = "/Users/ryan/Code/Data/MVSEC/indoor_flying3/indoor_flying3_gt/depthframes/frame_0000000000.png";

    cv::Mat image = cv::imread(imagePath, cv::IMREAD_COLOR);

    if (image.empty()) {
        std::cerr << "Error: Could not open or find the image!" << std::endl;
        return -1;
    }

    cv::imshow("Display Image", image);
    cv::waitKey(0); // Wait for a key press indefinitely
    
    
    YAML::Node rightCamCalibInfo = YAML::LoadFile("xxxx.yaml");
    
    return 0;
}
```

```Cmakelists.txt```
```
cmake_minimum_required(VERSION 3.10)
project(beginner_tutorials)

set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -stdlib=libc++ -Wall -Wextra")

## Find catkin macros and libraries
## if COMPONENTS list like find_package(catkin REQUIRED COMPONENTS xyz)
## is used, also find other catkin packages
find_package(catkin REQUIRED COMPONENTS
  roscpp
  rospy
  std_msgs
)

find_package(OpenCV REQUIRED)
find_package(Eigen3 REQUIRED)
find_package(glog REQUIRED)
find_package(yaml-cpp REQUIRED)

if(YAML_CPP_FOUND)
    message("yaml-cpp FOUND!")
else()
  message(" yaml-cpp NOT FOUND!")
endif()

include_directories(
  include
  ${catkin_INCLUDE_DIRS} 
)

add_executable(test_yaml src/test.cpp)
target_link_libraries(test_yaml
    ${catkin_LIBRARIES}
    # ${OpenCV_LIBRARIES}
    ${OpenCV_LIBS}
    ${YAML_CPP_LIBRARY})

```

```package.xml```

```
<?xml version="1.0"?>
<package format="2">
  <name>beginner_tutorials</name>
  <version>0.0.0</version>
  <description>The beginner_tutorials package</description>

  <!-- One maintainer tag required, multiple allowed, one person per tag -->
  <!-- Example:  -->
  <!-- <maintainer email="jane.doe@example.com">Jane Doe</maintainer> -->
  <maintainer email="ryan@todo.todo">ryan</maintainer>

  <!-- One license tag required, multiple allowed, one license per tag -->
  <!-- Commonly used license strings: -->
  <!--   BSD, MIT, Boost Software License, GPLv2, GPLv3, LGPLv2.1, LGPLv3 -->
  <license>TODO</license>

  <buildtool_depend>catkin</buildtool_depend>
  <buildtool_depend>catkin_simple</buildtool_depend>
  
  <build_depend>roscpp</build_depend>
  <build_depend>rospy</build_depend>
  <build_depend>std_msgs</build_depend>
  <build_export_depend>roscpp</build_export_depend>
  <build_export_depend>rospy</build_export_depend>
  <build_export_depend>std_msgs</build_export_depend>
  <exec_depend>roscpp</exec_depend>
  <exec_depend>rospy</exec_depend>
  <exec_depend>std_msgs</exec_depend>
  <build_depend>dvs_msgs</build_depend>
  <depend>yaml-cpp</depend>

  <!-- The export tag contains other, unspecified, tags -->
  <export>
    <!-- Other tools can request additional information be placed here -->

  </export>
</package>
```

Tobias-Fischer, 2025-03-19:
You need to link “yaml-cpp::yaml-cpp”

closed 2025-03-19 by Tobias-Fischer