RoboStack/ros-noetic #521: All post-PR builds are failing late March 2025

traversaro, 2025-03-22:
~~~
usage: basename string [suffix]
       basename [-a] [-s suffix] string [...]
/Users/runner/work/_temp/3a26111f-3062-482b-b77b-568da8688fe2.sh: line 4: .scripts/build_unix.sh: No such file or directory
~~~

traversaro, 2025-03-22:
@Tobias-Fischer @nmarticorena this seems a change related to https://github.com/RoboStack/vinca/pull/73 . Is there a change we need to propagate to ros-* robotack repos after https://github.com/RoboStack/vinca/pull/73 ?

traversaro, 2025-03-22:
I added a temporary workaround in https://github.com/RoboStack/ros-noetic/pull/522 .

Tobias-Fischer, 2025-03-23:
@nmarticorena are you happy to take care of this? We need to move to build_unix.sh just like in humble.

Tobias-Fischer, 2025-04-08:
Has been fixed in #523

closed 2025-04-08 by Tobias-Fischer