RoboStack/ros-noetic #525: ROS Python cv_bridge bug

JiahangWu, 2025-03-28:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

Hi,

When I write python code ```cv_img = bridge.imgmsg_to_cv2(msg, desired_encoding="passthrough")```, there is an error: 
```
Exception has occurred: ImportError
dlopen(/Users/ryan/.local/share/mamba/envs/esvo/lib/python3.11/site-packages/cv_bridge/boost/cv_bridge_boost.so, 0x0002): symbol not found in flat namespace '__ZN5boost6python6detail13current_scopeE'
  File "/Users/ryan/Code/depth_analysis/rosbag2h5img.py", line 22, in convert
    cv_img = bridge.imgmsg_to_cv2(msg, desired_encoding="passthrough")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/ryan/Code/depth_analysis/rosbag2h5img.py", line 45, in <module>
    convert(bag_path, topic, output_h5)
ImportError: dlopen(/Users/ryan/.local/share/mamba/envs/esvo/lib/python3.11/site-packages/cv_bridge/boost/cv_bridge_boost.so, 0x0002): symbol not found in flat namespace '__ZN5boost6python6detail13current_scopeE'
```

Does it mean that cv_bridge doesn't support osx-arm64? Do you have any idea how to solve it?

### Installed packages

```shell
List of packages in environment: "/Users/ryan/.local/share/mamba/envs/esvo"

  Name                                       Version       Build                      Channel
────────────────────────────────────────────────────────────────────────────────────────────────────────
  adwaita-icon-theme                         48.0          unix_0                     conda-forge
  aiohappyeyeballs                           2.6.1         pyhd8ed1ab_0               conda-forge
  aiohttp                                    3.11.13       py311h4921393_0            conda-forge
  aiosignal                                  1.3.2         pyhd8ed1ab_0               conda-forge
  aom                                        3.9.1         h7bae524_0                 conda-forge
  apr                                        1.7.0         h4614cfb_6                 conda-forge
  argcomplete                                3.6.0         pyhd8ed1ab_0               conda-forge
  assimp                                     5.4.3         ha9c0b8d_0                 conda-forge
  atk-1.0                                    2.38.0        hd03087b_2                 conda-forge
  attrs                                      25.3.0        pyh71513ae_0               conda-forge
  autoconf                                   2.71          pl5321hcd07c0c_1           conda-forge
  bcrypt                                     4.3.0         py311h3ff9189_0            conda-forge
  bimvee                                     1.0.23        pypi_0                     pypi
  blosc                                      1.21.6        h5499902_0                 conda-forge
  brotli                                     1.1.0         hd74edd7_2                 conda-forge
  brotli-bin                                 1.1.0         hd74edd7_2                 conda-forge
  bullet-cpp                                 3.25          py311h57524c7_3            conda-forge
  bzip2                                      1.0.8         h99b78c6_7                 conda-forge
  c-ares                                     1.34.4        h5505292_0                 conda-forge
  c-compiler                                 1.9.0         hdf49b6b_0                 conda-forge
  ca-certificates                            2025.1.31     hf0a4a13_0                 conda-forge
  cached-property                            1.5.2         hd8ed1ab_1                 conda-forge
  cached_property                            1.5.2         pyha770c72_1               conda-forge
  cairo                                      1.18.4        h6a3b0d2_0                 conda-forge
  catkin_pkg                                 1.0.0         pyhd8ed1ab_1               conda-forge
  catkin_tools                               0.9.4         pyhd8ed1ab_1               conda-forge
  cctools                                    1010.6        hb4fb6a3_3                 conda-forge
  cctools_osx-arm64                          1010.6        h3b4f5d3_3                 conda-forge
  ceres-solver                               2.1.0         hfaa3f39_2                 conda-forge
  cffi                                       1.17.1        py311h3a79f62_0            conda-forge
  clang                                      18.1.8        default_h474c9e2_7         conda-forge
  clang-18                                   18.1.8        default_hf90f093_7         conda-forge
  clang_impl_osx-arm64                       18.1.8        h2ae9ea5_23                conda-forge
  clang_osx-arm64                            18.1.8        h07b0088_23                conda-forge
  clangxx                                    18.1.8        default_h1ffe849_7         conda-forge
  clangxx_impl_osx-arm64                     18.1.8        h555f467_23                conda-forge
  clangxx_osx-arm64                          18.1.8        h07b0088_23                conda-forge
  cmake                                      3.31.6        ha25475f_0                 conda-forge
  colcon-argcomplete                         0.3.3         pyhd8ed1ab_1               conda-forge
  colcon-bash                                0.5.0         pyhd8ed1ab_1               conda-forge
  colcon-cd                                  0.1.1         pyhd8ed1ab_1               conda-forge
[... 533 lines cut ...]
          mamba version : 2.0.7
           curl version : libcurl/8.12.1 OpenSSL/3.4.1 (SecureTransport) zlib/1.3.1 zstd/1.5.6 libssh2/1.11.1 nghttp2/1.64.0
     libarchive version : libarchive 3.7.7 zlib/1.3.1 liblzma/5.6.3 bz2lib/1.0.8 liblz4/1.10.0 libzstd/1.5.6
       envs directories : /Users/ryan/.local/share/mamba/envs
          package cache : /Users/ryan/.local/share/mamba/pkgs
                          /Users/ryan/.mamba/pkgs
            environment : esvo (active)
           env location : /Users/ryan/.local/share/mamba/envs/esvo
      user config files : /Users/ryan/.mambarc
 populated config files : /Users/ryan/.local/share/mamba/envs/esvo/.condarc
                          /Users/ryan/.condarc
       virtual packages : __unix=0=0
                          __osx=15.3.2=0
                          __archspec=1=arm64
               channels : https://conda.anaconda.org/robostack-noetic/noarch
                          https://conda.anaconda.org/robostack-noetic/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/main/osx-arm64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/r/osx-arm64
       base environment : /Users/ryan/.local/share/mamba
               platform : osx-arm64
```

traversaro, 2025-03-28:
Thanks for raising the issue! Can you report the output of:

~~~
otool -L /Users/ryan/.local/share/mamba/envs/esvo/lib/python3.11/site-packages/cv_bridge/boost/cv_bridge_boost.so
~~~

traversaro, 2025-03-28:
(by the way, I see from the profile that you are in IIT in Genoa, if you are able to reproduce and you are in CRIS feel free to pass by my office, it is at the first floor near to the printer.

JiahangWu, 2025-03-28:
```
/Users/ryan/.local/share/mamba/envs/esvo/lib/python3.11/site-packages/cv_bridge/boost/cv_bridge_boost.so:
	@rpath/cv_bridge_boost.so (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libcv_bridge.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libopencv_core.410.dylib (compatibility version 410.0.0, current version 4.10.0)
	@rpath/libc++.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1345.120.2)
```

JiahangWu, 2025-03-28:
> (by the way, I see from the profile that you are in IIT in Genoa, if you are able to reproduce and you are in CRIS feel free to pass by my office, it is at the first floor near to the printer).

Haha, give me half an hour. I am in a meeting. 😂

traversaro, 2025-03-28:
I am also in a meeting until 15:00, so we can align after, thanks!

traversaro, 2025-04-03:
Fix proposed in https://github.com/RoboStack/ros-noetic/pull/526 .

closed 2025-04-03 by Tobias-Fischer