RoboStack/ros-noetic #532: Upgrade to MacOS Sequoia 15.5: rosbag play stalls

jbmouret, 2025-05-22:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

Hello

Thank you for this very useful set of packages!

I was happily using ROS-noetic on my mac (M4) until yesterday. This morning, I upgrade to Sequoia 15.5 (I guess I was before on 15.4). ROS still works (I can publish and read topics)... but not rosbag play.

```shell
rosbag play  data/rosbags/pick_cube_and_put_in_bowl/20250424_00003.bag --clock
[ INFO] [1747903571.703974000]: Opening data/rosbags/pick_cube_and_put_in_bowl/20250424_00003.bag

Waiting 0.2 seconds after advertising topics... done.

Hit space to toggle paused, or 's' to step.
 [RUNNING]  Bag Time: 1745485029.199244   Duration: 0.000000 / 13.191962        
```
→ it never plays. The duration stays at 0.0000. And no message is ever published. 

I tried to see where it is stuck:

```
(lldb) process attach --pid 8587
Process 8587 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGSTOP
    frame #0: 0x00000001a011caa4 libsystem_kernel.dylib`__write_nocancel + 8
libsystem_kernel.dylib`__write_nocancel:
->  0x1a011caa4 <+8>:  b.lo   0x1a011cac4    ; <+40>
    0x1a011caa8 <+12>: pacibsp 
    0x1a011caac <+16>: stp    x29, x30, [sp, #-0x10]!
    0x1a011cab0 <+20>: mov    x29, sp
Target 0: (play) stopped.
Executable module set to "/opt/homebrew/Cellar/micromamba/2.1.1/envs/ros_noetic/lib/rosbag/play".
Architecture set to: arm64-apple-macosx-.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGSTOP
  * frame #0: 0x00000001a011caa4 libsystem_kernel.dylib`__write_nocancel + 8
    frame #1: 0x00000001a001bd78 libsystem_c.dylib`__swrite + 24
    frame #2: 0x000000019fffcccc libsystem_c.dylib`_swrite + 108
    frame #3: 0x000000019fffad6c libsystem_c.dylib`__sflush + 232
    frame #4: 0x000000019fffac0c libsystem_c.dylib`fflush + 124
    frame #5: 0x0000000102e41d14 librosbag.dylib`rosbag::Player::printTime() + 512
    frame #6: 0x0000000102e41330 librosbag.dylib`rosbag::Player::doPublish(rosbag::MessageInstance const&) + 1676
    frame #7: 0x0000000102e3ed64 librosbag.dylib`rosbag::Player::publish() + 4820
    frame #8: 0x0000000102d56fcc play`main + 168
    frame #9: 0x000000019fdbab98 dyld`start + 6076
```

→ I would not expect anything to be blocked at a simple write (especially for printTime()??).

Same thing happens if I play in root.

What else can I test? I can try a full reinstall but it does not seem there is any update that I am missing.

### Installed packages

```shell
Name                                       Version       Build                         Channel         
───────────────────────────────────────────────────────────────────────────────────────────────────────────
  adwaita-icon-theme                         48.0          unix_0                        conda-forge     
  ampl-asl                                   1.0.0         h286801f_2                    conda-forge     
  aom                                        3.9.1         h7bae524_0                    conda-forge     
  apr                                        1.7.0         h4614cfb_6                    conda-forge     
  argcomplete                                3.6.2         pyhd8ed1ab_0                  conda-forge     
  assimp                                     5.4.3         ha9c0b8d_0                    conda-forge     
  atk-1.0                                    2.38.0        hd03087b_2                    conda-forge     
[... 574 lines cut ...]
  zziplib                                    0.13.69       h57f5043_2                    conda-forge
```

### Environment info

```shell
libmamba version : 2.1.1
          mamba version : 2.1.1
           curl version : libcurl/8.7.1 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.12 nghttp2/1.64.0
     libarchive version : libarchive 3.7.9 zlib/1.2.12 liblzma/5.8.1 bz2lib/1.0.8 liblz4/1.10.0 libzstd/1.5.7
       envs directories : /opt/homebrew/Cellar/micromamba/2.1.1/envs
          package cache : /opt/homebrew/Cellar/micromamba/2.1.1/pkgs
                          /Users/jmouret/.mamba/pkgs
            environment : ros_noetic (active)
           env location : /opt/homebrew/Cellar/micromamba/2.1.1/envs/ros_noetic
      user config files : /Users/jmouret/.mambarc
 populated config files : 
       virtual packages : __unix=0=0
                          __osx=15.5=0
                          __archspec=1=arm64
               channels : https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
       base environment : /opt/homebrew/Cellar/micromamba/2.1.1
               platform : osx-arm64
```

jbmouret, 2025-05-30:
Thank you. Unfortunately, I get the same behavior (rosbag not playing).

traversaro, 2025-05-30:
Hello @jbmouret ! When the program stops with `SIGSTOP`, is there anything useful printed in `dmesg` ? In https://github.com/stack-of-tasks/pinocchio/issues/2619#issuecomment-2736381333 we investigated a similar issue (but probably unrelated) for which we understood the root cause by inspecting the root cause via `dmesg`.

DEAR18, 2025-07-08:
I encountered the same problem on macOS Sequoia 15.5, but I couldn't capture any information using `dmesg | grep urdfdom`. I'm not sure if I performed the operation correctly. Here are the steps I followed in one terminal:
1. rosbag play xx.bag (then stucked)
2. Ctrl C
3. sudo dmesg | grep urdfdom

DEAR18, 2025-07-08:
> That is correct, but I would report the full dmesg output, as the problem may not be related to urdfdom

Here is a complete dmesg output. I'm not familier with it and tried to ask GPT, and it couldn’t identify the issue. 
[dmesg.log](https://github.com/user-attachments/files/21131489/dmesg.log)