RoboStack/ros-noetic #533: Unable to resolve dependency when trying to install ROS Noetic + MoveIt + libboost-devel on MacOS

gonzalocasas, 2025-06-02:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

I'm trying a clean install of ROS1 Noetic with MoveIt including `libboost-devel`, but it fails to find a solution.

To reproduce: 
```bash
$ mamba create -n ros1 -c robostack ros-noetic-desktop-full ros-noetic-moveit libboost-devel python
```

The underlying reason for trying to include `libboost-devel` on env creation is that on a first test, it was failing during `rosdep` after adding the universal robots repo to my catkin ws.

Notice that I cannot create the environment, so, below is not the list of installed packages, but the error message when trying to create it.

### Installed packages

```shell
❯ mamba create -n ros1 -c robostack ros-noetic-desktop-full ros-noetic-moveit libboost-devel python

Looking for: ['ros-noetic-desktop-full', 'ros-noetic-moveit', 'libboost-devel', 'python']

conda-forge/osx-arm64                                         No change
robostack/osx-arm64                                           No change
robostack/noarch                                              No change
conda-forge/noarch                                  20.6MB @  47.5MB/s  0.4s
Could not solve for environment specs
The following packages are incompatible
├─ libboost-devel is installable with the potential options
│  ├─ libboost-devel 1.82.0 would require
│  │  └─ boost-cpp 1.82.0* , which can be installed;
│  ├─ libboost-devel 1.83.0 would require
│  │  └─ boost-cpp 1.83.0* , which can be installed;
│  ├─ libboost-devel 1.84.0 would require
│  │  └─ boost-cpp 1.84.0* , which can be installed;
│  ├─ libboost-devel 1.85.0 would require
│  │  └─ boost-cpp 1.85.0* , which can be installed;
│  └─ libboost-devel [1.86.0|1.87.0|1.88.0] would require
│     └─ boost-cpp <0.0a0 , which can be installed;
├─ ros-noetic-desktop-full is installable and it requires
│  └─ ros-distro-mutex 0.3.* , which requires
│     └─ boost-cpp 1.74.* , which can be installed;
└─ ros-noetic-moveit is not installable because it requires
   └─ ros-noetic-moveit-core, which requires
      └─ ros-noetic-roslib but there are no viable options
         ├─ ros-noetic-roslib 1.15.8 would require
         │  └─ ros-noetic-rospack but there are no viable options
         │     ├─ ros-noetic-rospack 2.6.2 would require
         │     │  └─ boost-cpp, which conflicts with any installable versions previously reported;
         │     ├─ ros-noetic-rospack 2.6.2 would require
         │     │  └─ ros-distro-mutex 0.1.* , which conflicts with any installable versions previously reported;
         │     └─ ros-noetic-rospack 2.6.2 would require
         │        └─ ros-distro-mutex 0.2.* , which conflicts with any installable versions previously reported;
         ├─ ros-noetic-roslib 1.15.7 would require
         │  └─ ros-distro-mutex 0.1.* , which conflicts with any installable versions previously reported;
         └─ ros-noetic-roslib 1.15.8 would require
            └─ ros-distro-mutex 0.2.* , which conflicts with any installable versions previously reported.
eth/ros/moveit_ws took 4s
```

### Environment info

```shell
❯ mamba info

          mamba version : 1.5.9
     active environment : None
            shell level : 0
       user config file : /Users/gcasas/.condarc
 populated config files : /Users/gcasas/miniforge3/.condarc
                          /Users/gcasas/.condarc
          conda version : 24.7.1
    conda-build version : not installed
         python version : 3.12.6.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=m1
                          __conda=24.7.1=0
                          __osx=15.3.2=0
                          __unix=0=0
       base environment : /Users/gcasas/miniforge3  (writable)
      conda av data dir : /Users/gcasas/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack/osx-arm64
                          https://conda.anaconda.org/robostack/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/gcasas/miniforge3/pkgs
                          /Users/gcasas/.conda/pkgs
       envs directories : /Users/gcasas/miniforge3/envs
                          /Users/gcasas/.conda/envs
               platform : osx-arm64
             user-agent : conda/24.7.1 requests/2.32.3 CPython/3.12.6 Darwin/24.3.0 OSX/15.3.2 solver/libmamba conda-libmamba-solver/24.7.0 libmambapy/1.5.9
                UID:GID : 503:20
             netrc file : None
           offline mode : False
```

wolfv, 2025-06-02:
`libboost-devel` was introduced later and is not compatible with the boost version we used to build noetic. If you just require `boost-cpp` (which is anyways already installed) you should have a working `boost` (a little bit older, at 1.74).

traversaro, 2025-06-02:
You should use as a channel robostack-noetic, not robostack for the latest packages.

closed 2025-06-02 by Tobias-Fischer

gonzalocasas, 2025-06-02:
That indeed worked, thanks!