RoboStack/ros-noetic #534: RViz and ROS-Noetic-Compressed-Image-Transport Not Responding on Windows 10

tsungHannn, 2025-06-16:
# Description
I am using **Windows 10** and have followed the official installation instructions to set up **ROS1 Noetic** using Mamba. After installation, I was able to successfully start roscore (though it took slightly longer than expected). However, when testing RViz, I encountered an issue: after the RViz logo appears, the program becomes unresponsive after approximately five seconds. The RViz logo disappears, and the terminal used to launch RViz returns to the command prompt.

I checked the log and confirmed that an error occurred, but I couldn't pinpoint the cause.
 **(Apologies, the log contains some Chinese text. The relevant error message translates to: ConnectionResetError: [WinError 10054] An existing connection was forcibly closed by the remote host.)**

Additionally, I have observed that the same issue occurs when attempting to use ros-noetic-compressed-image-transport.

I have also verified that the issue is not related to the **Windows firewall**. I disabled the firewall completely, but the same connection reset error still occurs under the same conditions.

Here is the log after I execute roscore and rviz:
```
[rosmaster.main][INFO] 2025-05-19 14:38:48,758: initialization complete, waiting for shutdown
[rosmaster.main][INFO] 2025-05-19 14:38:48,758: Starting ROS Master Node
[xmlrpc][INFO] 2025-05-19 14:38:48,760: XML-RPC server binding to 0.0.0.0:11311
[xmlrpc][INFO] 2025-05-19 14:38:48,761: Started XML-RPC server [http://DESKTOP-DFDKCE5:11311/]
[rosmaster.master][INFO] 2025-05-19 14:38:48,762: Master initialized: port[11311], uri[http://DESKTOP-DFDKCE5:11311/]
[xmlrpc][INFO] 2025-05-19 14:38:48,763: xml rpc node: starting XML-RPC server
[rosmaster.master][INFO] 2025-05-19 14:39:06,801: +PARAM [/run_id] by /roslaunch
[rosmaster.master][INFO] 2025-05-19 14:39:06,802: +PARAM [/roslaunch/uris/host_desktop_dfdkce5__50838] by /roslaunch
[rosmaster.master][INFO] 2025-05-19 14:39:13,043: +SERVICE [/rosout/get_loggers] /rosout http://DESKTOP-DFDKCE5:51097/
[rosmaster.master][INFO] 2025-05-19 14:39:13,043: +SERVICE [/rosout/set_logger_level] /rosout http://DESKTOP-DFDKCE5:51097/
[rosmaster.master][INFO] 2025-05-19 14:39:13,044: +PUB [/rosout_agg] /rosout http://DESKTOP-DFDKCE5:51097/
[rosmaster.master][INFO] 2025-05-19 14:39:13,046: +SUB [/rosout] /rosout http://DESKTOP-DFDKCE5:51097/
[rosmaster.master][INFO] 2025-05-19 14:39:25,272: +PARAM [/rosversion] by /roslaunch
[rosmaster.master][INFO] 2025-05-19 14:39:25,273: +PARAM [/rosdistro] by /roslaunch
[rosmaster.master][INFO] 2025-05-19 14:39:41,784: +PUB [/rosout] /rviz_1747636781781100800 http://DESKTOP-DFDKCE5:51208/
[rosmaster.master][INFO] 2025-05-19 14:39:41,784: +SERVICE [/rviz_1747636781781100800/get_loggers] /rviz_1747636781781100800 http://DESKTOP-DFDKCE5:51208/
[rosmaster.master][INFO] 2025-05-19 14:39:41,785: +SERVICE [/rviz_1747636781781100800/set_logger_level] /rviz_1747636781781100800 http://DESKTOP-DFDKCE5:51208/
[rosmaster.master][INFO] 2025-05-19 14:39:41,785: +CACHEDPARAM [/rosout_disable_topics_generation] by /rviz_1747636781781100800
[rosmaster.master][INFO] 2025-05-19 14:39:41,858: publisherUpdate[/rosout] -> http://DESKTOP-DFDKCE5:51097/ ['http://DESKTOP-DFDKCE5:51208/']
[rosmaster.master][INFO] 2025-05-19 14:39:43,378: +SUB [/tf] /rviz_1747636781781100800 http://DESKTOP-DFDKCE5:51208/
[rosmaster.master][INFO] 2025-05-19 14:39:43,379: +SUB [/tf_static] /rviz_1747636781781100800 http://DESKTOP-DFDKCE5:51208/
[rosmaster.master][INFO] 2025-05-19 14:39:47,985: publisherUpdate[/rosout] -> http://DESKTOP-DFDKCE5:51097/ ['http://DESKTOP-DFDKCE5:51208/']: sec=6.13, result=[1, '', 0]
[xmlrpc][ERROR] 2025-05-19 14:39:48,597: Traceback (most recent call last):
  File "C:\Users\John\.local\share\mamba\envs\ros1\Lib\socketserver.py", line 691, in process_request_thread
    self.finish_request(request, client_address)
  File "C:\Users\John\.local\share\mamba\envs\ros1\Lib\socketserver.py", line 361, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "C:\Users\John\.local\share\mamba\envs\ros1\Lib\socketserver.py", line 755, in __init__
    self.handle()
  File "C:\Users\John\.local\share\mamba\envs\ros1\Lib\http\server.py", line 438, in handle
    self.handle_one_request()
  File "C:\Users\John\.local\share\mamba\envs\ros1\Lib\http\server.py", line 404, in handle_one_request
    self.raw_requestline = self.rfile.readline(65537)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\John\.local\share\mamba\envs\ros1\Lib\socket.py", line 718, in readinto
    return self._sock.recv_into(b)
           ^^^^^^^^^^^^^^^^^^^^^^^
ConnectionResetError: [WinError 10054] 遠端主機已強制關閉一個現存的連線。
```

Here is the ros-noetic-compressed-image-transport error log after I run the command: `rosrun image_transport republish compressed in:=/aravis_cam/image_color raw out:=/aravis_cam/image_color`.
It seems to be the same problem as RVIZ.
```
[rosmaster.master][INFO] 2025-05-19 14:55:55,389: +PUB [/rosout] /image_republisher_1747637755385144900 http://DESKTOP-DFDKCE5:53853/
[rosmaster.master][INFO] 2025-05-19 14:55:55,390: +SERVICE [/image_republisher_1747637755385144900/get_loggers] /image_republisher_1747637755385144900 http://DESKTOP-DFDKCE5:53853/
[rosmaster.master][INFO] 2025-05-19 14:55:55,390: +SERVICE [/image_republisher_1747637755385144900/set_logger_level] /image_republisher_1747637755385144900 http://DESKTOP-DFDKCE5:53853/
[rosmaster.master][INFO] 2025-05-19 14:55:55,413: publisherUpdate[/rosout] -> http://DESKTOP-DFDKCE5:51097/ ['http://DESKTOP-DFDKCE5:51208/', 'http://DESKTOP-DFDKCE5:53853/']
[rosmaster.master][INFO] 2025-05-19 14:55:55,840: +PUB [/aravis_cam/image_color] /image_republisher_1747637755385144900 http://DESKTOP-DFDKCE5:53853/
[xmlrpc][ERROR] 2025-05-19 14:55:57,733: Traceback (most recent call last):
  File "C:\Users\John\.local\share\mamba\envs\ros1\Lib\socketserver.py", line 691, in process_request_thread
    self.finish_request(request, client_address)
  File "C:\Users\John\.local\share\mamba\envs\ros1\Lib\socketserver.py", line 361, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "C:\Users\John\.local\share\mamba\envs\ros1\Lib\socketserver.py", line 755, in __init__
    self.handle()
  File "C:\Users\John\.local\share\mamba\envs\ros1\Lib\http\server.py", line 438, in handle
    self.handle_one_request()
  File "C:\Users\John\.local\share\mamba\envs\ros1\Lib\http\server.py", line 404, in handle_one_request
[... 504 lines cut ...]
                          C:\Users\John\.condarc
       virtual packages : __win=10.0.19045=0
                          __archspec=1=x86_64
                          __cuda=12.6=0
               channels : https://conda.anaconda.org/robostack-noetic/noarch
                          https://conda.anaconda.org/robostack-noetic/win-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/conda-forge/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
       base environment : C:\Users\John\.local\share\mamba
               platform : win-64
```

Is there any additional information I should provide to help diagnose this issue? Thank you!

Tobias-Fischer, 2025-06-16:
Can you please create a fresh environment without any `pypi` packages just to confirm that this is not causing issues?

Tobias-Fischer, 2025-06-16:
Also checking that you run `roscore` and `rviz` in separate terminals?