RoboStack/ros-noetic #538: rviz canvas doesn't update when window size changes

lauralindzey, 2025-07-16:
First, thank you so much for this tool! One of my projects is stuck on ROS1, and I've quickly been able to get enough of the stack up and running to get work done without having to actually run an unsupported OS =)

It seems like rviz doesn't properly redraw when the window size changes -- the display stretches/shrinks to fit in the window, rather than rendering more/less of the scene and keeping the aspect ratio correct.

Is this a configuration error on my end? Or is it a known limitation of running on MacOS? (I'm on 15.4.1)

Example screencaps:

[image]

[image]

traversaro, 2025-07-16:
Depending on the tool you are using, can you share your conda/mamba/micromamba/pixi info and list?

Tobias-Fischer, 2025-07-16:
There are a few osx-specific hacks in the patches: https://github.com/RoboStack/ros-noetic/blob/main/patch/ros-noetic-rviz.osx.patch (also note the platform-independent patches: https://github.com/RoboStack/ros-noetic/blob/main/patch/ros-noetic-rviz.patch)

These have been around for a loooong time, and I wonder 1) if they are still needed (I added them as rviz crashed when adding multiple image panels or something like that), and 2) if they are related to your observations.

Would you feel comfortable to build rviz from scratch and investigate this?

Tobias-Fischer, 2025-07-16:
Especially this might be of interest - we should check if this actually runs:
https://github.com/RoboStack/ros-noetic/blob/da62e07c03d7432313283dd0640d4e74f744ac4a/patch/ros-noetic-rviz.osx.patch#L211

lauralindzey, 2025-07-16:
> Depending on the tool you are using, can you share your conda/mamba/micromamba/pixi info and list?

pixi info:
```
System
------------
       Pixi version: 0.49.0
           Platform: osx-arm64
   Virtual packages: __unix=0=0
                   : __osx=15.4.1=0
                   : __archspec=1=m1
          Cache dir: /Users/lindzey/Library/Caches/rattler/cache
       Auth storage: /Users/lindzey/.rattler/credentials.json
   Config locations: No config files found

Global
------------
            Bin dir: /Users/lindzey/.pixi/bin
    Environment dir: /Users/lindzey/.pixi/envs
       Manifest dir: /Users/lindzey/.pixi/manifests/pixi-global.toml

Workspace
------------
               Name: robostack
      Manifest file: /Users/lindzey/Documents/RSAmanip/robostack/pixi.toml
       Last updated: 15-07-2025 19:51:30

Environments
------------
        Environment: default
           Features: default
           Channels: https://prefix.dev/conda-forge
   Dependency count: 12
       Dependencies: python, compilers, cmake, pkg-config, make, ninja, rosdep, colcon-common-extensions, vcstool, scipy, pip, textual
   Target platforms: osx-64, osx-arm64, linux-aarch64, linux-64, win-64
    Prefix location: /Users/lindzey/Documents/RSAmanip/robostack/.pixi/envs/default

        Environment: noetic
           Features: noetic, build, default
           Channels: https://prefix.dev/robostack-noetic, https://prefix.dev/conda-forge
   Dependency count: 17
       Dependencies: ros-noetic-desktop, catkin_tools, ros-noetic-image-geometry, ros-noetic-moveit-msgs, ros-noetic-uuid-msgs, python, compilers, cmake, pkg-config, make, ninja, rosdep, colcon-common-extensions, vcstool, scipy, pip, textual
   Target platforms: linux-64, linux-aarch64, win-64, osx-arm64, osx-64
    Prefix location: /Users/lindzey/Documents/RSAmanip/robostack/.pixi/envs/noetic
              Tasks: build

        Environment: humble
           Features: humble, build, default
           Channels: https://prefix.dev/robostack-humble, https://prefix.dev/conda-forge
   Dependency count: 13
       Dependencies: ros-humble-desktop, python, compilers, cmake, pkg-config, make, ninja, rosdep, colcon-common-extensions, vcstool, scipy, pip, textual
   Target platforms: win-64, linux-64, osx-arm64, osx-64, linux-aarch64
    Prefix location: /Users/lindzey/Documents/RSAmanip/robostack/.pixi/envs/humble
              Tasks: build

        Environment: jazzy
           Features: jazzy, build, default
           Channels: https://prefix.dev/robostack-jazzy, https://prefix.dev/conda-forge
   Dependency count: 13
       Dependencies: ros-jazzy-desktop, python, compilers, cmake, pkg-config, make, ninja, rosdep, colcon-common-extensions, vcstool, scipy, pip, textual
   Target platforms: win-64, osx-arm64, osx-64, linux-64, linux-aarch64
    Prefix location: /Users/lindzey/Documents/RSAmanip/robostack/.pixi/envs/jazzy
              Tasks: build

        Environment: kilted
           Features: kilted, build, default
           Channels: https://prefix.dev/robostack-kilted, https://prefix.dev/conda-forge
   Dependency count: 13
       Dependencies: ros-kilted-desktop, python, compilers, cmake, pkg-config, make, ninja, rosdep, colcon-common-extensions, vcstool, scipy, pip, textual
   Target platforms: linux-64, osx-64, win-64, osx-arm64, linux-aarch64
[... 503 lines cut ...]
unixodbc                                   2.3.12        h0e2417a_0                 248 KiB    conda  https://prefix.dev/conda-forge/
urdfdom                                    4.0.1         h090268e_2                 103.2 KiB  conda  https://prefix.dev/conda-forge/
urdfdom_headers                            1.1.2         h7b3277c_0                 18.8 KiB   conda  https://prefix.dev/conda-forge/
vcstool                                    0.3.0         pyhd8ed1ab_0               32.6 KiB   conda  https://prefix.dev/conda-forge/
wheel                                      0.45.1        pyhd8ed1ab_1               61.5 KiB   conda  https://prefix.dev/conda-forge/
x264                                       1!164.3095    h57fd34a_2                 700.2 KiB  conda  https://prefix.dev/conda-forge/
x265                                       3.5           hbc6ce65_3                 1.7 MiB    conda  https://prefix.dev/conda-forge/
xorg-libice                                1.1.2         h5505292_0                 47.3 KiB   conda  https://prefix.dev/conda-forge/
xorg-libsm                                 1.2.6         h5505292_0                 23.8 KiB   conda  https://prefix.dev/conda-forge/
xorg-libx11                                1.8.12        h6a5fb8c_0                 744.1 KiB  conda  https://prefix.dev/conda-forge/
xorg-libxau                                1.0.12        h5505292_0                 13.3 KiB   conda  https://prefix.dev/conda-forge/
xorg-libxaw                                1.0.16        hd74edd7_0                 241.8 KiB  conda  https://prefix.dev/conda-forge/
xorg-libxdmcp                              1.1.5         hd74edd7_0                 18.1 KiB   conda  https://prefix.dev/conda-forge/
xorg-libxext                               1.3.6         hd74edd7_0                 40.9 KiB   conda  https://prefix.dev/conda-forge/
xorg-libxmu                                1.2.1         hd74edd7_1                 62.1 KiB   conda  https://prefix.dev/conda-forge/
[... cut ...]
yaml-cpp                                   0.8.0         ha1acc90_0                 133 KiB    conda  https://prefix.dev/conda-forge/
zlib                                       1.3.1         h8359307_2                 75.8 KiB   conda  https://prefix.dev/conda-forge/
zstd                                       1.5.7         h6491c7d_2                 390.6 KiB  conda  https://prefix.dev/conda-forge/
zziplib                                    0.13.69       h57f5043_2                 97.1 KiB   conda  https://prefix.dev/conda-forge/
```