RoboStack/ros-noetic #541: rosbash: $) is not a valid variable in fish

stevalkr, 2025-08-16:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

```
~> source $ROS_PACKAGE_PATH/rosbash/rosfish                                                                                                                                                                                                                                                                                 (noetic)
~/.devkit/pixis/noetic/.pixi/envs/default/share/rosbash/rosfish (line 391): $) is not a valid variable in fish.
            set opts (_rosfind -L ".*/\.\(?!pixi(/|$)\)[^/].*" -print0 | tr '\000' '\n' | sed -e "s/.*\/\(.*\)/\1\//g")
                                                    ^
from sourcing file ~/.devkit/pixis/noetic/.pixi/envs/default/share/rosbash/rosfish
source: Error while reading file '/Users/walker/.devkit/pixis/noetic/.pixi/envs/default/share/rosbash/rosfish'
```

The regex `.*/\.\(?!pixi(/|$)\)[^/].*` is from `patch/ros-noetic-rosbash.patch`

### Installed packages

```shell
Too long, uploaded to https://envs.sh/2h6.txt
```

### Environment info

```shell
~> pixi info                                                                                                                                                                    (noetic)
System
------------
       Pixi version: 0.50.2
           Platform: osx-arm64
   Virtual packages: __unix=0=0
                   : __osx=15.6=0
                   : __archspec=1=m1
          Cache dir: /Users/walker/Library/Caches/rattler/cache
       Auth storage: /Users/walker/.rattler/credentials.json
   Config locations: No config files found

Global
------------
            Bin dir: /Users/walker/.pixi/bin
    Environment dir: /Users/walker/.pixi/envs
       Manifest dir: /Users/walker/.pixi/manifests/pixi-global.toml

Workspace
------------
               Name: noetic
            Version: 0.1.0
      Manifest file: /Users/walker/.devkit/pixis/noetic/pixi.toml
       Last updated: 14-08-2025 01:52:48

Environments
------------
        Environment: default
           Features: default
           Channels: https://fast.prefix.dev/conda-forge, https://prefix.dev/robostack-noetic
   Dependency count: 14
       Dependencies: make, ninja, cmake, python, pyright, compilers, pkg-config, numpy, scipy, scikit-learn, rosdep, catkin_tools, ros-noetic-desktop, ros-noetic-image-transport-plugins
  PyPI Dependencies: torch, torchvision
   Target platforms: linux-64, osx-arm64
    Prefix location: /Users/walker/.devkit/pixis/noetic/.pixi/envs/default
```

Tobias-Fischer, 2025-08-16:
Do you want to open a pull request with a fix?

stevalkr, 2025-08-16:
I'd love to help, but I don't really understand what this regex does and would need some time to learn 😅

Tobias-Fischer, 2025-08-17:
Cc @daenny @traversaro