RoboStack/ros-noetic #550: rqt_bag does not start on Mac

dirkpitt2050, 2025-09-21:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

rqt_bag fails to start with the following stack trace. The rqt command works, it's just rqt_bag that doesn't.

```
PluginManager._load_plugin() could not load plugin "rqt_bag/Bag":
Traceback (most recent call last):
  File "/Users/aaa/src/github/openvins/.pixi/envs/noetic/lib/python3.11/site-packages/qt_gui/plugin_handler.py", line 102, in load
    self._load()
  File "/Users/aaa/src/github/openvins/.pixi/envs/noetic/lib/python3.11/site-packages/qt_gui/plugin_handler_direct.py", line 55, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/aaa/src/github/openvins/.pixi/envs/noetic/lib/python3.11/site-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/aaa/src/github/openvins/.pixi/envs/noetic/lib/python3.11/site-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/aaa/src/github/openvins/.pixi/envs/noetic/lib/python3.11/site-packages/rqt_gui_py/ros_py_plugin_provider.py", line 61, in load
    return super(RosPyPluginProvider, self).load(plugin_id, plugin_context)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/aaa/src/github/openvins/.pixi/envs/noetic/lib/python3.11/site-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/aaa/src/github/openvins/.pixi/envs/noetic/lib/python3.11/site-packages/rqt_gui/ros_plugin_provider.py", line 106, in load
    return class_ref(plugin_context)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/aaa/src/github/openvins/.pixi/envs/noetic/lib/python3.11/site-packages/rqt_bag/bag.py", line 57, in __init__
    self._widget = BagWidget(context, args.clock)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/aaa/src/github/openvins/.pixi/envs/noetic/lib/python3.11/site-packages/rqt_bag/bag_widget.py", line 77, in __init__
    self._timeline = BagTimeline(context, publish_clock)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/aaa/src/github/openvins/.pixi/envs/noetic/lib/python3.11/site-packages/rqt_bag/bag_timeline.py", line 111, in __init__
    self._timeline_frame = TimelineFrame(self)
                           ^^^^^^^^^^^^^^^^^^^
  File "/Users/aaa/src/github/openvins/.pixi/envs/noetic/lib/python3.11/site-packages/rqt_bag/timeline_frame.py", line 129, in __init__
    self._topic_font.setPointSize(self._topic_font_size)
TypeError: setPointSize(self, a0: int): argument 1 has unexpected type 'float'
```

### Installed packages

```shell
# packages in environment at /Users/aaa/src/github/openvins/.pixi/envs/noetic:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        48.1                     unix_1    conda-forge
aiohappyeyeballs          2.6.1              pyhd8ed1ab_0    conda-forge
aiohttp                   3.12.15         py311hfbe4617_0    conda-forge
aiosignal                 1.4.0              pyhd8ed1ab_0    conda-forge
ampl-asl                  1.0.0                h240833e_2    conda-forge
aom                       3.9.1                hf036a51_0    conda-forge
apr                       1.7.0                hf5a09cc_6    conda-forge
assimp                    5.4.3                hf9fe5ce_0    conda-forge
atk-1.0                   2.38.0               h4bec284_2    conda-forge
attrs                     25.3.0             pyh71513ae_0    conda-forge
aws-c-auth                0.8.6                h321fff7_4    conda-forge
aws-c-cal                 0.8.7                hfaf822f_0    conda-forge
aws-c-common              0.12.0               h6e16a3a_0    conda-forge
aws-c-compression         0.3.1                hb1ee187_2    conda-forge
aws-c-event-stream        0.5.4                hf9b3e9c_2    conda-forge
aws-c-http                0.9.4                h29be59e_4    conda-forge
aws-c-io                  0.17.0               h786d7a7_6    conda-forge
aws-c-mqtt                0.12.2               h6a909e1_2    conda-forge
aws-c-s3                  0.7.13               h2313cb2_2    conda-forge
[... 643 lines cut ...]
       user config file : /Users/aaa/.condarc
 populated config files : /Users/aaa/.condarc
          conda version : 23.3.1
    conda-build version : not installed
         python version : 3.10.10.final.0
       virtual packages : __archspec=1=x86_64
                          __osx=14.7.6=0
                          __unix=0=0
       base environment : /Users/aaa/miniconda3  (writable)
      conda av data dir : /Users/aaa/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/aaa/miniconda3/pkgs
                          /Users/aaa/.conda/pkgs
       envs directories : /Users/aaa/miniconda3/envs
                          /Users/aaa/.conda/envs
               platform : osx-64
             user-agent : conda/23.3.1 requests/2.29.0 CPython/3.10.10 Darwin/23.6.0 OSX/14.7.6
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2025-09-22:
We need to backport https://github.com/ros-visualization/rqt_bag/pull/109/files to ros-noetic. Feel free to open a pull request :)

dirkpitt2050, 2025-09-22:
Yes, sorry this issue also seems to be a duplicate of #512. Closing.

closed 2025-09-22 by dirkpitt2050