RoboStack/ros-noetic #551: cannot import name 'ImageQt' from 'PIL.ImageQt'

dirkpitt2050, 2025-09-22:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

After patching rqt_bag using this (https://github.com/ros-visualization/rqt_bag/pull/131/files), rqt_bag runs but the following warning appears in the terminal on startup. Also no images appear when I select view thumbnails or try to view raw image data in the bag. (it is not a corrupt bag - images were visible in a non-robostack install on Ubuntu)
```
RosPluginProvider.load(rqt_bag_plugins/BagImagePlugin) exception raised in __builtin__.__import__(rqt_bag_plugins.image_plugin, [ImagePlugin]):
Traceback (most recent call last):
  File "/Users/aaa/src/github/openvins/.pixi/envs/noetic/lib/python3.11/site-packages/rqt_gui/ros_plugin_provider.py", line 79, in load
    module = __builtin__.__import__(
             ^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/aaa/src/github/openvins/.pixi/envs/noetic/lib/python3.11/site-packages/rqt_bag_plugins/image_plugin.py", line 35, in <module>
    from .image_timeline_renderer import ImageTimelineRenderer
  File "/Users/aaa/src/github/openvins/.pixi/envs/noetic/lib/python3.11/site-packages/rqt_bag_plugins/image_timeline_renderer.py", line 45, in <module>
    from PIL.ImageQt import ImageQt
ImportError: cannot import name 'ImageQt' from 'PIL.ImageQt' (/Users/aaa/src/github/openvins/.pixi/envs/noetic/lib/python3.11/site-packages/PIL/ImageQt.py)
```

### Installed packages

```shell
# packages in environment at /Users/aaa/src/github/openvins/.pixi/envs/noetic:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        48.1                     unix_1    conda-forge
aiohappyeyeballs          2.6.1              pyhd8ed1ab_0    conda-forge
aiohttp                   3.12.15         py311hfbe4617_0    conda-forge
aiosignal                 1.4.0              pyhd8ed1ab_0    conda-forge
ampl-asl                  1.0.0                h240833e_2    conda-forge
aom                       3.9.1                hf036a51_0    conda-forge
apr                       1.7.0                hf5a09cc_6    conda-forge
assimp                    5.4.3                hf9fe5ce_0    conda-forge
atk-1.0                   2.38.0               h4bec284_2    conda-forge
attrs                     25.3.0             pyh71513ae_0    conda-forge
aws-c-auth                0.8.6                h321fff7_4    conda-forge
aws-c-cal                 0.8.7                hfaf822f_0    conda-forge
aws-c-common              0.12.0               h6e16a3a_0    conda-forge
aws-c-compression         0.3.1                hb1ee187_2    conda-forge
aws-c-event-stream        0.5.4                hf9b3e9c_2    conda-forge
aws-c-http                0.9.4                h29be59e_4    conda-forge
aws-c-io                  0.17.0               h786d7a7_6    conda-forge
aws-c-mqtt                0.12.2               h6a909e1_2    conda-forge
aws-c-s3                  0.7.13               h2313cb2_2    conda-forge
aws-c-sdkutils            0.2.3                hb1ee187_2    conda-forge
aws-checksums             0.2.3                hb1ee187_2    conda-forge
aws-crt-cpp               0.31.0               hc7e8f17_4    conda-forge
aws-sdk-cpp               1.11.510             ha0394b9_3    conda-forge
azure-core-cpp            1.14.0               h9a36307_0    conda-forge
azure-identity-cpp        1.10.0               ha4e2ba9_0    conda-forge
azure-storage-blobs-cpp   12.13.0              h3d2f5f1_1    conda-forge
azure-storage-common-cpp  12.8.0               h1ccc5ac_1    conda-forge
bcrypt                    4.3.0           py311hd3d88a1_2    conda-forge
blosc                     1.21.6               hd145fbb_1    conda-forge
brotli                    1.1.0                h1c43f85_4    conda-forge
brotli-bin                1.1.0                h1c43f85_4    conda-forge
bullet-cpp                3.25                 hcf53e2f_4    conda-forge
bzip2                     1.0.8                h500dc9f_8    conda-forge
c-ares                    1.34.5               hf13058a_0    conda-forge
c-compiler                1.11.0               h7a00415_0    conda-forge
ca-certificates           2025.8.3             hbd8a1cb_0    conda-forge
cairo                     1.18.4               h950ec3b_0    conda-forge
capnproto                 1.0.2                h1c0ecac_3    conda-forge
catkin_pkg                1.1.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.5              pyh29332c3_0    conda-forge
cctools                   1024.3               h67a6458_1    conda-forge
cctools_osx-64            1024.3               haa85c18_1    conda-forge
cffi                      1.17.1          py311he66fa18_1    conda-forge
cfitsio                   4.5.0                h8c0165a_0    conda-forge
[... 618 lines cut ...]
       user config file : /Users/aaa/.condarc
 populated config files : /Users/aaa/.condarc
          conda version : 23.3.1
    conda-build version : not installed
         python version : 3.10.10.final.0
       virtual packages : __archspec=1=x86_64
                          __osx=14.7.6=0
                          __unix=0=0
       base environment : /Users/aaa/miniconda3  (writable)
      conda av data dir : /Users/aaa/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/aaa/miniconda3/pkgs
                          /Users/aaa/.conda/pkgs
       envs directories : /Users/aaa/miniconda3/envs
                          /Users/aaa/.conda/envs
               platform : osx-64
             user-agent : conda/23.3.1 requests/2.29.0 CPython/3.10.10 Darwin/23.6.0 OSX/14.7.6
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

dirkpitt2050, 2025-09-22:
If I look inside PIL/ImageQt.py, it is checking for PyQt6 or PySide6 but I believe the pixi environment does not contain either of these (only PyQt5)...

Tobias-Fischer, 2026-03-10:
Duplicate of https://github.com/RoboStack/ros-noetic/issues/512

closed 2026-03-10 by Tobias-Fischer